gitbls/sdm #19: sdm-cportal outside of sdm

mcguirepr89, 2021-12-10:
Hi, Benn -- I hope this finds you well

I have been tinkering with `sdm-cportal` and have made a few adjustements, namely I have edited the links and HTTPServer to use port 9090 instead of 80 (since I already have a web-server bound to port 80, 8080, and 8888). 

With those adjustments, _most_ of the captive portal works.
I am able to "Start the captive portal" and fill in the form successfully.
'testinputs' returns success.

However, when I attempt the next step, I find that `sdm-cportal` depends on an `sdm` configuration file, `/etc/sdm/local-1piboot.conf`

I wanted to bring this to your attention since I know it was a goal of `sdm-cportal` to be usable outside of `sdm`, though it seems there is a small dependency issue.

Looking at that file, however, everything is commented out, so I'm going to see what happens if I just `touch` that file and try again.

Below is the Python traceback for the issue
```
----------------------------------------
Exception occurred during processing of request from ('10.1.1.246', 60992)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/./sdm/sdm-cportal", line 231, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/pi/./sdm/sdm-cportal", line 364, in do_GET
    self._setstop(pd, False)
  File "/home/pi/./sdm/sdm-cportal", line 268, in _setstop
    writel10n(pd, self.pd.keymap, self.pd.locale, self.pd.timezone)
  File "/home/pi/./sdm/sdm-cportal", line 225, in writel10n
    with open("/etc/{}/local-1piboot.conf".format(pd.facname), 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/etc/sdm/local-1piboot.conf'
----------------------------------------
```

mcguirepr89, 2021-12-10:
Update:
```
sudo mkdir /etc/sdm
sudo touch /etc/sdm/local-1piboot.conf
```
fixed the issue

gitbls, 2021-12-10:
Ugh. Will have a look at it after I get done chasing what appears to be a surprise error message when using --shrink on Bullseye.

In the meantime, your workaround is good.

mcguirepr89, 2021-12-10:
Good luck with you hunting!! Bullseye is still seeing significant changes daily -- mtfbwy

gitbls, 2021-12-10:
Just looked at the code.

It's structured such that you either have to specify an `--l10nhandler` script, or cportal will try to write to the sdm file hierarchy. 

If you don't care about the values for L10N, you can use `--l10nhandler /dev/null`, as the code neither checks nor reports errors from the `--l10nhandler` script.

gitbls, 2021-12-16:
Fixed and tested.

closed 2021-12-16 by gitbls