gitbls/sdm #22: Usage hints

cjnaz, 2021-12-15:
sdm looks very interesting.  I have a few customizations I do/need that I'm not clear on how to best handle them with sdm

1. I mount a network samba share (via a line if fstab) with the login info in a credentials file in /root with 600 protection.
2. I modify the pi user .bashrc for sourcing aliases from a network share (mounted in the fstab). 
3. I create a mount point (/mnt/RAMDRIVE) for the boot time config of a RAM-based tmpfs partition listed in fstab.
4. I set up samba to share said RAM drive on the network
5. I need to start some specific custom services by adding .service files and enabling them (run at boot).

My base server is a Centos 7 box.  It looks like I'll have to set up sdm on one of my Pi devices.

Thanks for your work on this project, and for addressing my questions.

gitbls, 2021-12-15:
Thanks for your interest in sdm. I do similar things in my use of sdm. Everything you want to do should be done in a custom phase script. You can easily start with the example on the github and extend it. Here's some commentary on each of your points:

1. Build a file with the lines that you want to add to /etc/fstab. Use the `--fstab` switch, pointing to the file. That will automatically append the lines to /etc/fstab during the first boot of the system (does it there since in some cases some network-related configuration might not be done until the first boot of the system). Alternatively, you can append the lines to /etc/fstab in your custom phase script. Can be done in either phase 0 or Phase 1. Before I added --fstab I copied the file into the image during phase 0 and appended it to /etc/fstab. I'd copy the credentials file into /mnt/sdm/root in Phase 0, and set the protection, etc. 
2. Again, you can do this in your Custom Phase Script Phase 0 (where you have access to the local system to copy files), or in Phase1 (running "inside" the image. You have flexibility to do it either way. Note that if you want to change the owner of a file, it's best to wait until Phase 1, where the /etc/passwd and /etc/group of the IMG are used. I use a separate user from 'pi'. I handle it by a) creating the home directory and copying my login files into it in my custom Phase 0, and then b) chowning and setting protection on files in custom Phase 1. You can do the same types of things with the pi user, of course.
3. You can do this the same way as 1) above. Put both (all?) the lines you want to append to /etc/fstab into a single file and handle per 1) above.
4. I haven't installed samba manually for a while, but when I added `--poptions samba` I'm pretty sure that I did it because the samba package install prompted, which is pretty annoying in the context of sdm. The code that installs samba for `--poptions samba`  uses debconf to avoid the prompts. Unless there's a reason not to, you can build your smb.conf outside of sdm in your file system somewhere, and copy it into the image in your Custom Phase Phase 0 (for instance, copy it to /root, or maybe /etc/sdm/local-assets (which I just made up, and think I'll add to the sdm docs). Then in your Custom Phase script post-install phase, move the file from whereever you stashed it into /etc/samba.
5. I do this by copying the .service files into /mnt/sdm/etc/systemd/system during Phase 0, and then use the appropriate systemctl commands (e.g., `systemctl enable myservice > /dev/null 2>&1`)  during Phase 1 to enable them.

Hope this helps!

LMK if you need clarifications on any of the above, or have "the next round of questions", or (gasp!) run into a bug.

gitbls, 2021-12-15:
A couple of other things:

- As you start working on your Custom Phase Script, get in the habit of using `logtoboth "string to log"` so that you can see what's going on in your script. Also, after your customization completes, use `sdm --explore IMG` to go into the image and check everything out (/etc/sdm/history, files exist where they should, with correct file owner and protection, etc).
- I've not tried sdm on Centos. You could try it on Centos by installing sdm manually, and installing systemd-container, binfmt-support, and qemu-user-static, but Centos 7 is pretty old, so unless those packages have been updated, you're probably out of luck.

gitbls, 2021-12-15:
One last thing. I added a copy of my personal Custom Phase Script to the wiki [here](https://github.com/gitbls/sdm/wiki/Example-Custom-Phase-script).

gitbls, 2021-12-18:
I haven't used sdm extensively on a 0W but I expect it will be a bit slow, primarily due to the 0W CPU speed.

You can definitely use sdm on a network mounted share. I use it against files on an NFS share all the time. I haven't done Samba, but I would expect that to work as well. LMK if you're using Samba and run into problems.

I think it's simply an awesome tool. Not because I wrote it. If someone else had done this, I'd definitely be using it, because it saves me SO much time in getting a new system set up. It actually makes tinkering fun, because all the dull boring configuration stuff is done...automatically. It also greatly helps with my impatience. I can wait for an SD Card to burn, but that's about it ;)

Looking forward to hearing how you get on with it

cjnaz, 2021-12-18:
Off topic, but check this out...  https://github.com/cjnaz/lanmonitor
I get to know instantly when something is awry...  backups out of date, missing hosts, services down, etc, across any node on my LAN.

gitbls, 2021-12-18:
Looks interesting and well-done!

gitbls, 2021-12-19:
Just had a thought...if you're going to be using sdm against files on a network share, here's a little script that does the copy on the remote node. This avoids a bunch of redundant network copying. I use this often when I need to reset the image I'm working on to a "vanilla" IMG. Also if you don't have password-less ssh see my rxapp project. This bit depends on password-less ssh for sanity maintenance ;)
```
#!/bin/bash
# assumes that the network share is mounted on /ssd on both the 'workstation' and the network server
rmt="rmthost"  #The name of the host with the network share
rel="2021-10-30"
arch="arm64"
img="$rel-raspios-bullseye-$arch-lite.img"
src="/ssd/vanilla/${rel}/${arch}"
[ ! -f $src/$img ] && echo "? File '$src/$img' not found" && exit

echo -e "$(date +"%Y-%m-%d %H:%M:%S") Starting remote service"
# Start the remote ssh session and execute the copy over there
(ssh -t $rmt 2>/dev/null <<EOF
cd $PWD
rm -f $img
#echo 1 | tee /proc/sys/vm/drop_caches > /dev/null
echo -e "\$(date +"%Y-%m-%d %H:%M:%S") Start image copy from $src/$img"
cp $src/$img .
echo -e "\n\$(date +"%Y-%m-%d %H:%M:%S") Image copy completed\n"
#echo 1 | tee /proc/sys/vm/drop_caches > /dev/null
exit
EOF
) &
```

gitbls, 2022-01-18:
Closing this since no action in several weeks. Please re-open or start a new issue if warranted.

closed 2022-01-18 by gitbls