gitbls/sdm #23: Port for arch linux?

arnav7633, 2021-12-23:
Hey! I saw that the installation script uses apt sadly which isn't available on arch (The host pc) so my question is do you have a port for arch Linux? or do we need to modify the install script ourselves?

gitbls, 2021-12-23:
Are those the only 2 choices?

What about the third choice: I'll update sdm to support Arch, you'll do the research and get me the answers to the questions i need in order to support it.

arnav7633, 2021-12-24:
Sure! I am down for that

gitbls, 2021-12-24:
OK, I haven't looked at arch at all yet.. Here's an incomplete list that popped when I flipped through the sdm sources.

- Is there an IMG for the Pi with 2 partitions (or can one be created)? Not having an IMG, or having more than 2 partitions would be a problem
- Is systemd supported? 
- Assuming systemd is supported, does it use /etc/systemd/...
- Is systemd-nspawn (package systemd-container) available? Need this if you want to run sdm on arch, as opposed to running, for instance, on RasPiOS and operating on Arch images
- How are software packages installed (what tool)?
- Does config.txt have the same format and settings as RasPiOS?
- Is raspi-config available? If not, how are the settings done that are done in raspi-config?
- How is GUI-less network configuration done  (dhcpcd? dhclient? Network Manager, etc)
- How would sdm determine L10N information in the system running sdm? On RasPios: Gets timezone from /etc/localtime; keyboard map from /etc/default/keyboard; and locale from /etc/default/locale. Also grabs wifi country from /etc/wpa_supplicant/wpa_supplicant.conf

arnav7633, 2021-12-24:
1. Not sure what you mean
2. Yep
3. Yes
[image]
4. Yes 
[image]
5. Pacman
6. config.txt?
7. Yes
https://aur.archlinux.org/packages/raspi-config-git/
8. We use a command line utility included in arch called iwd/iwctl or the wpa_supplicant conf files
9. There are symbolic links for the timezone
[image]
Keyboard - https://man.archlinux.org/man/vconsole.conf.5
Wfi - https://wiki.archlinux.org/title/Network_configuration/Wireless
[image]
"can" So we need the user to set the country in one of the following ways listed in the "Regulatory domain" area of the guide

gitbls, 2021-12-24:
Well, #1 is a big issue. The RasPiOS distro is delivered in an IMG, which is a disk image. It includes the partition tables and 2 partitions (1 FAT, and 1 EXT4). Becuase it's a disk image, you can simply dd it to an SD card and boot. It can also be mounted into the running system using losetup/mount, so that sdm can operate on it.

What format do the Arch bits come in, and what are the steps to get them onto an SD card and boot it?

arnav7633, 2021-12-24:
Arch is also delivered as an iso which has a few partitions. Like root and boot. I am away from my pc rn but afaik you can flash a USB(never tried a sd card) with arch using dd

arnav7633, 2021-12-24:
I will get back to you with more info tommorning morning

gitbls, 2021-12-24:
sdm can't work with an ISO, AFAIK they are not writeable. The more arch is different from RasPiOS, the more difficult it would be to add support for arch.  Remember, one of the goals of sdm is to have a common image with all your customizations in it, and then burn that to a bazillion SSDs/SD Cards. If the only way that sdm can customize your image is on USB stick or SD card, it will still be useful, but won't be nearly as good as the RasPiOS model.

arnav7633, 2021-12-24:
Ok so I used to develop a arch based distro. Basically you can create a file structure and we can write that out to a iso and make a disk image. If you would be interested I can go into detail

gitbls, 2021-12-24:
TBH I'm not terribly interested in becoming an Arch expert, as I'm quite happy with RasPiOS. Have used plenty of other distros in the past, and they're pretty much all the same, except for differences in management and package managers. 

I'm interested in Arch to see what it would take to support it with sdm, as it seems fairly popular on the Pi. 

An ISO is NOT the same as a disk image. a disk image can be mounted and written to, and extended, etc. I'm pretty sure you can't do that with an ISO.

arnav7633, 2021-12-24:
Ah btw, what I meant originally was the ability to build raspios images from a arch system. And yeah arch offers a mire low level control compared to debian imo

arnav7633, 2021-12-24:
https://youtu.be/FQ6PEPodnQg
This was the video I referenced to

arnav7633, 2021-12-24:
Another approach could be to make a diskimage out of a installed arch system
https://wiki.archlinux.org/title/Disk_cloning

gitbls, 2021-12-24:
Re the video: I don't see what building your own distro has to do with sdm.

re disk cloning: The beauty of sdm is that you start with a "virgin" RasPiOS (in the supported case) image, so there's no possibility of having introduced any obvious or hidden problems. I'm not a big fan of cloning disks for subsequent use. Fine for backups, but not to create a disk for another system. 

Based on the info you've provided me thus far, it looks like supporting arch with sdm could be a good-sized lift. I'm not really interested in doing major surgery on sdm, but if you want to port it to arch, you're certainly free to go do that. The sources are all here, so have at it.

I'm still willing to be convinced otherwise, but it will take more than what you've provided so far to win me over.

arnav7633, 2021-12-25:
Ok so what i meant was i can bootup a a freshly downloaded iso and make a disk image out of that which is completely new and free of problems which can be later used with sdm

arnav7633, 2021-12-25:
"Re the video: I don't see what building your own distro has to do with sdm."
What i wanted to show is we could do that with a freshly installed iso and copy over all the base stuff then when we want to make a iso with sdm we just copy our custom created vanilla disk image and then do our magic on it

arnav7633, 2021-12-25:
If i can make a diskimage of arch (vanila) that would greatly reduce your workload correct?

arnav7633, 2021-12-25:
I found this interesting repo
https://github.com/archdroid-org/alarm-image-generator

gitbls, 2021-12-25:
I've been thinking about multi-distro support for a while, and now have a plan for overhauling Phase 1, which is where most of the system changes are done. I will get that done in the next wave of updates to sdm, and then adding additional distros will be much easier and should result in far less code spaghetti mess.

But there will still need to be an editable IMG file as I described above, as RasPiOS and some other distros are delivered for the Pi. ISOs won't cut it. If there's no disk IMG for arch, I'm far less interested in supporting it, as it will require changes in areas of sdm that are now distro-independent.

I'll revisit arch once I've finished the distro refactoring. Perhaps you can figure out how to get an arch IMG created in an automated way while I'm working on the refactoring. You should be able to test that IMG with `sdm --explore`

arnav7633, 2021-12-25:
Well. I cant run some on my machine because I run arch and it doesn't have apt so some exits due to its absence

gitbls, 2021-12-25:
I'm not sure what "some" you're referring to, or what point you're trying to make. sdm does not run on arch today, nor does it operate on arch ISOs or (apparently non-existent) arch IMG files.

Please clarify the point you're trying to make. Thx

arnav7633, 2021-12-26:
Sorry my bad. Autocorrect I meant sdm..

gitbls, 2021-12-26:
so s/some/sdm/ in your comment, your point is still unclear. We knew at the outset of this Issue that it didn't work on arch. IOW no new information or am I missing something?

arnav7633, 2021-12-26:
"You should be able to test that IMG with sdm --explore"
I can't use/install sdm on my pc because I am an arch user which means I am unable to try that if i make a test disk image

gitbls, 2021-12-26:
I wouldn't have suggested it if I wasn't confident that it would work. `--explore` only uses systemd-nspawn, so there's a very good chance that `--explore` will work on arch.

Try installing it per [https://github.com/gitbls/sdm/wiki/Detailed-Installation-Guide](https://github.com/gitbls/sdm/wiki/Detailed-Installation-Guide) and instead of using apt to install the 3 additional packages, use whatever the pacman command is to install them.

arnav7633, 2021-12-28:
We can't use tar files right?
[image]

arnav7633, 2021-12-28:
[image]
What does this mean?

gitbls, 2021-12-28:
sdm only operates on IMG files, so no TAR files.

The only time I've seen the nspawn failing like this is documented [here](https://github.com/gitbls/sdm/discussions/10). What distro (version and bits (32/64)) are you running on, and what distro (version, bits) is in the IMG?

arnav7633, 2021-12-28:
I am running a 64 bit arch install and the image is of a armv7 arch

gitbls, 2021-12-28:
Does your 64-bit arch install have qemu-user-static and binfmt-support installed?

arnav7633, 2021-12-28:
I missed that currently i am building it :flushed:

arnav7633, 2021-12-29:
[image]
I was chrooted in and can confirm the image was accessible normally idk why sdm fails

arnav7633, 2021-12-29:
BTW arch linux has a special util package called `arch-chroot` which spawns a shell into the disk image
[image]

gitbls, 2021-12-29:
Please post the results of `sudo /usr/local/sdm/sdm --ppart ~/custom-pi.img`

arnav7633, 2021-12-29:
[image]
`* Human-readable partition list
Model:  (file)
Disk /home/arnav/custom-pi.img: 2147483648B
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start      End          Size         Type     File system  Flags
 1      1048576B   99614719B    98566144B    primary  fat32        lba
 2      99614720B  2147483647B  2047868928B  primary  ext4

* Machine-readable partition list
BYT;
/home/arnav/custom-pi.img:2147483648B:file:512:512:msdos::;
1:1048576B:99614719B:98566144B:fat32::lba;
2:99614720B:2147483647B:2047868928B:ext4::;`

gitbls, 2021-12-29:
no clue why you're getting those errors. Here's a script that does the bare minimum: creates a loop device, mounts the 2 partitions, and then systemd-nspawn into the image. 

```
#!/bin/bash                                                                                                                                                               

/usr/local/sdm/sdm --ppart $1
mkdir -p /mnt/sdm/boot
loopdev=$(losetup --show -P -f $1)
mount -v ${loopdev}p2 /mnt/sdm
mount -v ${loopdev}p1 /mnt/sdm/boot
systemd-nspawn --directory /mnt/sdm < $(tty)
umount /mnt/sdm/{boot,}
losetup -d $loopdev
```
If this doesn't work there's something seriously wrong with systemd-nspawn on arch.

gitbls, 2021-12-29:
Rereading this thread again, what do you mean by "I was chrooted in and can confirm the image was accessible normally idk why sdm fails". Were you in a chroot when you ran sdm? If so, what is the result if you run sdm when NOT in a chroot?

arnav7633, 2021-12-30:
Basically what I wanted to show was that the disk image is in a correct format and usable.

arnav7633, 2021-12-30:
> no clue why you're getting those errors. Here's a script that does the bare minimum: creates a loop device, mounts the 2 partitions, and then systemd-nspawn into the image.
> 
> ```
> #!/bin/bash                                                                                                                                                               
> 
> /usr/local/sdm/sdm --ppart $1
> mkdir -p /mnt/sdm
> loopdev=$(losetup --show -P -f $1)
> mount -v ${loopdev}p2 /mnt/sdm
> mount -v ${loopdev}p1 /mnt/sdm/boot
> systemd-nspawn --directory /mnt/sdm < $(tty)
> umount /mnt/sdm/{boot,}
> losetup -d $loopdev
> ```
> 
> If this doesn't work there's something seriously wrong with systemd-nspawn on arch.

Ok that worked and spawned me into the disk image

gitbls, 2021-12-30:
I asked you a specific question about whether you were chrooted or not, and you did not respond. I can't tell what state your system was in, and you didn't clarify, so please do the following without doing any chroot, etc. and let's see the output from that:

- Reboot your arch system
- `sudo rm -rf /mnt/sdm/boot`  #in case it's there
- `sudo /usr/local/sdm/sdm --explore custom-pi.img`

arnav7633, 2021-12-30:
I thought my 2nd previous message justified it. But yeah I wasn't chrooted in when I tried exploring jt

arnav7633, 2021-12-30:
Ok so i fixed it. It was a permission issue. APOLOGIES!

gitbls, 2021-12-30:
That's good to hear, since the code in the test script (which worked) came directly from sdm.

So, now that we've verified that `sdm --explore` works, you can start experimenting with sdm to determine what switches do and don't work. That's quite necessary to scope the work required to support arch.

Looking forward to hearing the results from your testing!

arnav7633, 2021-12-30:
> what switches do and don't work.

Sorry switches? You mean sdm function?

arnav7633, 2021-12-30:
Feeling stupid asking this but i suppose NetworkManager uses wpa_supplicant but doesn't create any configs?

gitbls, 2021-12-30:
> Feeling stupid asking this but i suppose NetworkManager uses wpa_supplicant but doesn't create any configs?

Sorry, don't know very much about NM. I've used it a couple of times, but it felt very heavyweight for the Pi. Obviously useful for mobile devices like laptops, which would connect to a variety of different networks.

gitbls, 2022-02-02:
Closing this topic due to lack of participation from @arnav7633 . Please feel free to open the topic for further discussion if/when you're ready to continue.

closed 2022-02-02 by gitbls

arnav7633, 2022-02-12:
Welp. I took a break. 
Arch arm ships as a tar ball @gitbls

arnav7633, 2022-02-14:
Cool, although i would have to create a version with another language
I am not very familiar with bash

gitbls, 2022-02-14:
I thought you were looking for a port of the existing code to to fully support Arch. If you go down that path, you'd be using bash. Done with the goal of **_minimal_** changes only to support Arch, it could then be integrated back into the core sdm. That would be cool, technically, and something I would be interested in doing, because then some things that sdm does can be factored out by which OS it's operating on, making it easier to support additional OS in the future.

OTOH, if you are considering changing languages, then it's not really a **_port_**, is it?  It's more like something completely new, possibly doing some/most/all of what sdm does today,  At best it will be _culturally compatible_, because that's what happens with these things.

To be clear, I don't have a vested interest in which way you choose to go. If you do the minimal changes approach, that ultimately will require me to do a fair amount of work (see above). I'm good with that, but at the same time not eagerly awaiting it.

If you do a complete reimplementation, I'll be interested in seeing what features you add beyond what's already there.

One last thought...it never hurts to learn another programming language. 👍