gitbls/sdm #24: pip3 installs

royord, 2022-01-03:
I have just found this and have been browsing the documentation. In particular I'm interested in installing python packages. Is there a method for "pip3 install..."?

gitbls, 2022-01-03:
There is no built-in support for pip3 install in sdm at the current time. Obviously it could be added, but I am a bit concerned about chasing pip3, then at some point pip4, etc., especially since you can easily use pip3 install in a custom phase script:

- Copy /usr/local/sdm/sdm-customphase to somewhere. I suggest establishing a directory where you keep all your sdm assets (files). I have established an NFS share, /rpi, where I keep Pi-related assets, so that I can access them from any of the Pi systems on my LAN. You could also use /home/pi/sdm or any other directory, it's totally up to you.
- Edit your copy of sdm-customphase and add the "pip3 install" commands in the post-install section. I suggest using the post-install phase in case you discover dependencies that need to be installed during Phase 1. If there aren't any dependencies, you can equivalently put the pip3 install commands in the Phase 1 section.
- Add `--cscript /path/to/your/custom-phase-script` to the `sdm --customize` command line

That's it! 

If you have further questions on this, or would like to weigh in on adding an integrated `--pip3` command switch, happy to have further discussion in this issue.

gitbls, 2022-01-11:
Closing this issue due to no further replies. Please re-open or start a new issue if you want to discuss further.

closed 2022-01-11 by gitbls

royord, 2022-02-01:
I did appreciate the information and am working to customize my image as much as possible and follow this methodology. This is a great tool and is going to save me many hours.

gitbls, 2022-02-01:
Great! LMK if you get stuck on anything. Like most tools, the more you use it, the easier it gets to make it do what you want.