gitbls/sdm #25: Is it possible to have sdm-cportal run on every boot, and not just first boot?

everyplace, 2022-03-20:
I’m building an image wherein I’d like the user to be able to go through this flow if they’ve moved to an unfamiliar network. Basically if a familiar network is available, great. Otherwise, launch into the sdm-cportal experience. Is this possible via one of the scripts at the moment, or could you point me in the direction of what I might investigate to accomplish this?

gitbls, 2022-03-20:
Are you trying to use sdm-cportal with sdm, or standalone?

everyplace, 2022-03-20:
Preferably with. You can see the sketch for my end result in [quadratura](https://github.com/CooperUnion/quadratura), but the idea is to manage the creation of this image through sdm. I'm officially starting on this for my students tomorrow night, so still in the discovery phase.

gitbls, 2022-03-21:
I took a quick look at your notes. You're on the right path in using a systemd service to run sdm-cportal, that is exactly the path I was thinking as I mulled this over last night.

Irrelevant but I feel compelled to point out that sdm-cportal is NOT based on using a HotSpot with hostapd. sdm-cportal does its thing without using hostapd.

All that said, the service to run sdm-cportal is actually fairly simple, and should do the job. 

I'll have a look into this because I think it would be a useful addition to sdm (and sdm-cportal in particular), and will get back to you here in the next day or two.

everyplace, 2022-03-21:
Thanks for the great response. I’m glad I’m not totally not on the wrong track re: following the path of configuring systemd services. 

On your not about not using hostapd, I’m not too worried about that. In my target scenario for my students, they’re only ever accessing these devices via the cportal to get them onto wifi, and never as a proper bridge or router from another connection to wifi (as the target platform is just the Pi Zero 2). From my readings, this lets me believe that the lack of hostapd is not a big deal for my use-case.

I’ll proceed in building my image using sdm with sdm-cportal, and will assume that either a) you’ll be able to add this in the future, or b) I’ll be able to hack my functionality in place on top of it. Either way, thank you again for your feedback, and let me know if there’s anything I can to do help / test / provide more details.

gitbls, 2022-03-21:
I'd encourage you to try sdm-cportal on a standalone pi to ensure that it basically meets your needs, or can be updated to do so. Best to test it with a keyboard/monitor attached, since it mucks with the network. Here are a couple of considerations:

- If the Pi's only have a single WiFi in them the phone (assuming that's what students are using to configure the Pi) drops off the Pi WiFi network when sdm-cportal tests whether the supplied SSID/credentials are good. It's not a big deal, but it's not a seamless experience. 
- The presented Web UI is really basic. It can be modified, so maybe not a big deal

You can try it by running (interactively): `sudo /usr/local/sdm/sdm-cportal` 

If you try it in this mode, I'd value your feedback on it, as it's pretty basic at this point. I'm up for improving it if there's a "customer"

everyplace, 2022-03-30:
I filed a new issue #26, as I was attempting to test this, but was unable to complete the burn process.

gitbls, 2022-04-16:
Closing this due to no further input from OP. If you want to reopen for further discussion, please feel free to do so.

closed 2022-04-16 by gitbls