gitbls/sdm #26: Sdm burn fails to modify after writing new image

everyplace, 2022-03-30:
So I’ve been experimenting with sdm per my previous issue filed, and have yet to be able to test the output. While the image is burned successfully, the post-burn scripts that do things like edit hostnames fail with an error about read-only disk partitions. I’ll follow-up with the actual error text, but just wanted to summarize first to see if there’s something obvious I should be doing.

everyplace, 2022-03-30:
I put the sample commands I’ve been using in my repo in [sdm.md](https://github.com/CooperUnion/quadratura/blob/main/sdm.md)

everyplace, 2022-03-30:
One other note: I’m running these commands on an up-to-date Pi4 running the 64-bit Raspberry OS release. The microSD card is connected via an adapter that works fine to burn an image via rpi-imager, so I’m assuming it’s not the system or hardware as the source of the problem.

gitbls, 2022-03-30:
The commands look fine. Please provide the actual error output...it's difficult to guess what might be going on without that. Thx!

everyplace, 2022-03-30:
Hmm... just recorded [this video](https://www.loom.com/share/38fce8e39e9f4d53a5563560e16f4993), and of course it worked without errors. Let me verify that the card burned properly, and I'll close this issue.

gitbls, 2022-03-30:
No, there is definitely an error. You shouldn't be getting the "partition /dev/sdc is in use" message.  I see that /dev/sdc is mounted BEFORE you start the burn. How did that happen? It really needs to be unmounted. please sort out how it got mounted, and why, and unmount it before you start the burn, and things will work much better. 

Also, see https://github.com/gitbls/sdm/wiki/Troubleshooting in case you need to clean up dangling mounts or loop devices created by sdm.

Once you've cleared up all this cruft, try the burn again.

gitbls, 2022-03-30:
Sigh. The disk is mounted due to the Full Desktop automounter. This, coupled with the fact that sdm has a failed check for "is the disk mounted" results in what you're seeing.

For now, you should be able to `sudo umount /dev/sdc1 ; sudo umount /dev/sdc1` and then do the burn. 

Fixing sdm to correctly check for disks being mounted, and if so, will ask if you really want to write there. Shouldn't take too long, once I sort out what the behavior should be if any of the disk's partitions are mounted.

gitbls, 2022-03-30:
V5.5 checked in corrects this. The only files changed are sdm (bump version number) and sdm-cmdsubs. Please try this and let me know how it goes. It _should_ inform you that the partitions are mounted and what to do before redoing the burn command.

everyplace, 2022-04-04:
I updated to v5.5 today, and was able to successfully burn. Your modification did correctly catch that the two mounts were active, and needed to be unmounted, so thank you!

closed 2022-04-04 by everyplace