gitbls/sdm #27: How to --bootscripts work?

everyplace, 2022-04-04:
I'd like to install docker, and pull down a few images, prior to the system being available for the first time. Per specific [pi instructions](https://dev.to/elalemanyo/how-to-install-docker-and-docker-compose-on-raspberry-pi-1mo), the most reliable way of getting it up and running is with `curl -sSL https://get.docker.com | sh`, but when I attempt to run this at various different phases, it fails. This type of setup also doesn't seem to clearly fit into the "apps" install scripts.

I suspect the right time to run this and the subsequent image fetching is after first-boot, but the custom-phase-script interface doesn't seem to allow for this customization. Reading through the docs, I'm guessing using a `--bootscripts` setting is the right procedure, but it doesn't seem to be spelled out. Roughly:

1. Create a standard image using whatever config you'd like
2. After logging into it at the end of the script in `nspawn`, manually create and edit a script in `/usr/local/sdm/1piboot` called `0-docker.sh` per the [--bootscripts](https://github.com/gitbls/sdm/blob/8638bbe13a46a8135121917cf25a6f5f2939164f/sdm#L458) flag. 
3. Done!

What's not clear to me is if the `--bootscripts` flag can allow me to pass in a file during burn, which it will subsequently run, or if I pass in an argument of a name of a script that already exists, so that it will run it properly. Any chance you could clarify please?

gitbls, 2022-04-04:
First, installing docker. There are (at least) two ways to install docker. docker is part of the Debian Bullseye distro and can be installed with `sudo apt install docker.io`.

Or, you can use the `curl -sSL https://get.docker.com | sh` method. 

If you use docker.io, you can simply add docker.io to your list of apps to be installed.

If you want to use the curl method, you can put that in a custom phase script in the post-install phase so that it is installed after all of your stuff from apps and xapps.

I don't know what the difference is between these two methods in terms of what docker bits are installed, and what docker features are available (or not). Here's what I do know:
- Using `apt install docker.io` consumes about 370MB of storage in the IMG
- Using the curl method consumes about 500MB of storage in the IMG, plus it adds an additional apt repository in /etc/apt/sources.list.d

You can't use docker to install or run any containers in sdm, since the required docker services aren't running, but you CAN do it once the system has booted, either using a bootscript, or interactively after the system is up and running.

Rather than using the boot scripts as documented, I suggest you do the following:
- Create the script you want to run at boot in /etc/sdm/0piboot/0*-*.sh. Replace the two asterisks as you like. For instance, 010-getdockercontainer.sh
- That's it! When sdm-firstboot runs during the first system boot, it runs all 0*-*.sh files in that directory with no need to use `--bootscripts`. In other words, these scripts are always run at first boot.
- Yes, need to improve the bootscript documentation. 

Here's the code stanza I added to my custom phase script post-install phase:
 ```
logtoboth "  $pfx Install Docker"
# Install docker
curl -sSL https://get.docker.com | sh
# Set up script in /etc/sdm/0piboot to install a docker image
cat > /etc/sdm/0piboot/010-get-docker-image.sh <<EOF
#!/bin/bash
docker pull hello-world
exit 0
EOF
chmod 755 /etc/sdm/0piboot/010-get-docker-image.sh
 ```
This creates the bash script as I described above, properly protected so that it can be executed.

everyplace, 2022-04-04:
Wow, a post-install script to GENERATE the first-boot script! What a cool idea.

gitbls, 2022-04-04:
You should read through the code in sdm-phase1. It does a lot of that!

everyplace, 2022-04-04:
Ok, fun edge case: this post-install step ONLY works if the computer has a `wpa_supplicant.conf` file setup. Otherwise, the post-install script exits unsuccessfully, as the device doesn’t have an internet connection. So this brings me back to the need to re-enable the `sdm-cportal` at a time after first-boot, as I could very easily setup the wifi config to work where I’m imaging the devices, but would need to trigger the portal for the time _after_ that.

gitbls, 2022-04-04:
I'm confused. Can you please describe the scenario a bit more? I thought I understood, but you called it a post-install script, so now I'm wondering if I got it right.

But, yes, if you want to install software of any sort from the internet, you need to have an internet connection.

gitbls, 2022-04-16:
Closing this due to no further input from OP. If you want to reopen for further discussion, please feel free to do so.

closed 2022-04-16 by gitbls