gitbls/sdm #28: Documentation Clarification

pc-magas, 2022-06-06:
Upon https://github.com/gitbls/sdm/wiki/Command-Details I have read the following:

> sudo /usr/local/sdm/sdm --burnfile customized-for-myhostname.img --host myhostname raspios-image.img
> Burns the IMG file to the specified SD Card Image and sets the hostname. The customized IMG file must be burned to an SD Card to be used.

Does that mean that will generate an output image file from `customized-for-myhostname.img` into `raspios-image.img` Into which sd card will be burned? I fail to understand it.

gitbls, 2022-06-06:
No, your undersatding is not correct. Assume you are building a customized Pi system for Bill, but you can't send him a physical SD Card, for whatever reason (speed, cost, etc).

In this case, you would use `sdm --burnfile bills-IMG.img --hostname billshost my-customized.IMG` and then deliver bills-IMG.img to him over the network somehow. Bill burns that IMG to an SD Card using `dd` or rpi-imager, etc, installs it into his Pi and he's ready to go.

Does this clarify the use of `--burnfile` for you?

pc-magas, 2022-06-07:
Thanks it helps a lot.

closed 2022-06-07 by pc-magas

gitbls, 2022-06-10:
You might want to have a look at sdm-gburn, which I just added. It burns customized SD cards driven by descriptions in a datafile. For instance, each SD card can be customized for wifi, L10N, and username/password. Further unique customizations can be added easily via additional bash-ing.