gitbls/sdm #30: Using sdm to set a static IP

ShovelJockey, 2022-08-21:
Hi we spoke on the raspi forums about using sdm for setting a static IP among other parameters for a raspberry pi image.

I am aiming to use an image setup with the needed libraries (pi-gen or creating an image from an SD install) to be able to burn to about 20 SD cards with each image having a shared wifi network/password with ssid broadcast off, and each having a different:
-unique hostname
-unique username/password
-static IP

This is with the aim of automating the setup needed to create a connected network for these 20 pis. Any help with using sdm for this goal would be very much appreciated!

gitbls, 2022-08-21:
As i mentioned in the forum, sdm-gburn already does _most_ of what you want. It is, however, lacking the ability to set a static IP address.

Following our forum exchange I took a closer look at it, and should have something for you soon. It's not that complex, but definitely something I can add much more quickly than you.

Once completed, you'll be able to have lines in the data file for sdm-gburn with (punctuation and exact wording may change, but current thinking):
```
bobuser password=bobspassword,dhcpcd=[dev:eth0;ipaddress:192.168.x.x;router:192.168.x.y;dns:1.1.1.1]
```
Plus all the other parameters I mentioned in the forum. This will append lines to the end of /etc/dhcpcd.conf on the SSD/SD with:
```
interface eth0
    static ip_address=192.168.x.x
    static routers=192.168.x.y
    static domain_name_servers=1.1.1.1
```
Obviously, the values for ipaddress, router, and dns are not constrained in any way by sdm-gburn. In fact, sdm-gburn will do NO validity checking on any of these items, so will be up to you to ensure their correctness.

`dev` must be specified and must be a valid network device (although as mentioned, sdm-gburn will not validate it). If any of `ipaddress`, `router`, or `dns` are not specified, the corresponding line will not be written to dhcpcd.conf, and the resulting behavior may not be as desired.

ShovelJockey, 2022-08-22:
Wow thanks so much! Trying able to write to the dhcpcd.conf was pretty much were I got stuck with pi-gen and this is a much more reliable solution to the other triggered scripts I looked at using.

gitbls, 2022-08-22:
Of course, the proof is in the code (that you don't have yet). While I'm finishing up testing/documentation, is there anything I can help you with in using sdm so that you'll be ready to rock and roll when the updated sdm-gburn is done?

What customizing do you want to do to the RasPiOS IMG before you burn your fleet of SSD/SDs?

ShovelJockey, 2022-08-22:
Well I am running into problems trying to burn a test image to an sd just to see if I am understanding how to use sdm.

The image is burning successfully to the sd and boots into the first time setup ok but hasn't got the hostname, username or userpass that i tried to set. I suspect this is an issue with my usage or setup of sdm.
heres what I ran:

$ sdm --burn /dev/sdb --hostname Jamiehost --user Jamieuser --password-user testpass --expand-root /home/jamie/Downloads/test.img

heres the terminal output:

[sudo] password for jamie: 
* Burn '/home/jamie/Downloads/test.img' to '/dev/sdb'...
dd if=/home/jamie/Downloads/test.img of=/dev/sdb status=progress bs=16M iflag=direct
4324327424 bytes (4.3 GB, 4.0 GiB) copied, 118 s, 36.7 MB/s
257+1 records in
257+1 records out
4324327424 bytes (4.3 GB, 4.0 GiB) copied, 274.12 s, 15.8 MB/s
> Expand Root: Expand partition from 3.8GB (4.0GiB) to 29.5GB (31.6GiB)
Checking that no-one is using this disk right now ... OK

Disk /dev/sdb: 29.72 GiB, 31914983424 bytes, 62333952 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x4a422963

Old situation:

Device     Boot  Start     End Sectors  Size Id Type
/dev/sdb1         8192  532479  524288  256M  c W95 FAT32 (LBA)
/dev/sdb2       532480 8445951 7913472  3.8G 83 Linux

/dev/sdb2: 
New situation:
Disklabel type: dos
Disk identifier: 0x4a422963

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sdb1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sdb2       532480 62333951 61801472 29.5G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Synching disks.
> Expand Root: Set new disk ID 'b5e68578' on '/dev/sdb'
Disk identifier changed from 0x4a422963 to 0xb5e68578.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Synching disks.
mount: /dev/sdb2 mounted on /mnt/sdm.
mount: /dev/sdb1 mounted on /mnt/sdm/boot.
> Expand Root: Set new Disk ID 'b5e68578' in /etc/fstab and /boot/cmdline.txt
% IMG '/home/jamie/Downloads/test.img' is not sdm-enhanced
  Logs and configuration updates will not be written
* Storage '/dev/sdb' is ready
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted

I am running Ubuntu 22.04.1

gitbls, 2022-08-22:
To be clear, when you say you're running Ubuntu, is that the system on which you're using sdm, or are you trying to burn an ubuntu IMG?

The username is set when you customize the IMG. What is the command that you used to customize it? According to the console log, you didn't customize it at all, so apparently you didn't set a username/password. 

The general flow is: 

- customize the image using sdm --customize. This sets the username, password, installs apps if you want any, and tweaks the configuration so it's just the way you want it
- Burn the customized IMG to an SSD/SD, where the hostname will be set.

Please clarify how what you did (or didn't do) aligns with the above.

ShovelJockey, 2022-08-22:
Ah sorry my bad, I am trying to juggle several things and totally missed that I needed to customize the image first for user/pass and not try to apply them with --burn, would trying to set those with --burn likely be the reason that a hostname wasn't set?

Ubuntu is the system I am using sdm on,  currently I am using the standard raspi os img to test functionality until the exact setup for the libraries etc has been decided on.

I will try again now and maybe read docs a little more carefully next time...

gitbls, 2022-08-22:
Have a look a the quick start guide on the main GitHub page. It shows how you can set the username, localization stuff, etc. quickly and easily.

You CAN add the username/password when the IMG is burned, that's what sdm-burn does. BUT, it's a bit more complex since you need to build some scripts, etc. That's what sdm-gburn does.

I don't understand why the host name wasn't set, will have a look at that when I get back to my desk in a couple of hours.

ShovelJockey, 2022-08-22:
Using --customize then --burn i have been able to set the host name, user/pass and localisation and boot without issue its a much faster process than pi-gen as I hoped it would be.

gitbls, 2022-08-22:
Great to hear 👍 And, congratulations, you found a bug! 🏆 Should be able to set the hostname even on an un-customized image. Need to slosh some code around. 

Also, you may want to add `--svcdisable userconfig` it's not needed when you set have sdm do everything. I haven't end-to-end tested without doing this for a while, but will shortly as part of finishing up sdm-gburn.

ShovelJockey, 2022-08-22:
Ok cool I wanted to check is it ok to use sdm for commercial projects? As I was hoping to use this code for parts of my work.

Would --svcdisable userconfig disable the first time boot setup as well?

gitbls, 2022-08-22:
Yes, fine to use for commercial projects. See the file LICENSE for details.

WRT userconfig, are you building on the Lite or "with desktop" version?

gitbls, 2022-08-22:
Looking at the first boot time setup stuff, and making it work correctly both for Lite and Desktop versions. Seems my first go-around on that was a touch short of correct.

ShovelJockey, 2022-08-23:
Currently just using the default full image from the site for testing but in the end will use a lite img I think might create with pi-gen or make an image of a currently setup pi sd install.

And yeah thats exactly what I meant as the first time setup shouldn't be needed and may be an obstacle to a smooth setup.

gitbls, 2022-08-23:
I'm expecting to have an updated sdm-gburn for you to use no later than Friday. Still have a few rough edges to address. 

I am curious why you would want to pi-gen your own IMG. What's the motivation?

ShovelJockey, 2022-08-23:
Cool! and I am not sure really I wanted to keep it as an option in case there are issues cloning an image from an SD that make building one more attractive?

gitbls, 2022-08-24:
Um, if by "clone" you mean to actually do a disk-to-disk copy of an SD card, I'm definitely not a big fan of that. sdm is NOT cloning. using pi-clone or dd or whatever to physically copy an SD to another disk IS a clone. Hopefully you didn't mean that?

ShovelJockey, 2022-08-25:
I meant creating an image from an a sd install that's already setup, from what I understand this usually used more for making a backup of an image?

gitbls, 2022-08-25:
Exactly. Use "clone" for backups, not for making copies to run on other systems. There are system-specific settings such as ssh host keys and disk ID (and others?) that would need to be redone if the cloned copy is used on another system. 

Use sdm to build enhanced ready-for-first-boot images to run on (other) systems.

**sdm-gburn update:** Still to do: Final sdm-gburn testing, and some documentation. Late Friday US west coast time completion still looks good.

gitbls, 2022-08-26:
Updates to sdm-gburn and sdm are checked in. You'll need to update sdm (changed files) and get new sdm-gburn. See wiki https://github.com/gitbls/sdm/wiki/Batch-burn-SSD-SDs-with-sdm-gburn for details on sdm-gburn.

LMK if you have any questions up front, and please LMK when you've got it running. Happy to hear about problems/suggestions.

ShovelJockey, 2022-08-27:
Thank you very much, ill have a look later and get back to you when I try it out!

ShovelJockey, 2022-08-31:
Hey sorry for the delay been a busy few days, everything is working well using SDM has been great and has been really useful for getting more used to which pi files are key for various network settings.

gitbls, 2022-08-31:
Thanks for the update!, and thanks for the inspiration to build out sdm-gburn. Do LMK if you have any further questions or run into any gaps where improvements to sdm would be helpful for you.

gitbls, 2022-09-05:
Closing this issue. If you have any further issues/questions, feel free to either re-open this issue or fire up a new issue. Thanks!

closed 2022-09-05 by gitbls