gitbls/sdm #32: Help with adding a custom service to an image during customisation

ShovelJockey, 2022-09-23:
Hi there! sdm has been working great to setup many pi but would appreciate some help/direction with using custom phase scripts. I have had success adding python libraries to an image with a custom script but need some help adding a custom service and enabling it on an image. 

I think the enable service call in phase 1 is working but the service file itself seems to be failing to be copied across to the systemd/system folder in the image so the enable call is failing to find the service to enable. When mounting and investigating the image I can find no trace of the file, I am sure I am missing a step or have misinterpreted whats required.

Currently I am trying to copy the service file in phase 0 with:
cp /usr/local/sdm/sdm-custom-asset/my.service /mnt/sdm/etc/systemd/system

and enabling the service in phase 1 with:
sudo systemctl enable my.service

Anymore help you could provide would be great as you were fantastically helpful with my last request!

gitbls, 2022-09-24:
Generally, a systemd service has a .service file, and whatever other assets are referenced by the .service file, such as scripts, programs, etc. Let me know if this is helpful.

Let's assume a service, foo, which has a foo.service file, and a foo.bin (for name differentiation) executable. The foo.service file has `ExecStart=/usr/local/bin/foo.bin` in it. 

In Phase 0:
```
cp foo.service $SDMPT/etc/sytemd/system
cp foo.bin $SDMPT/usr/local/bin
```
Then in Phase 1:
```
systemctl enable foo.service 
```
So that's the general thing.

I'm guessing that's what you thought you did. BUT, does your custom phase script really reference /mnt/sdm/... or do you use $SDMPT? And, what version of sdm are we talking about? I updated sdm earlier this year to support multiple active sdm sessions. If you have later than V6.0 AND sdm failed out for some reason or you CTRL/C'd out, you could have had /mnt/sdm still mounted from a failed previous usage, and during your recent use it was /mnt/sdm.nnnnn (nnnn=a pid). 

In this scenario, referencing /mnt/sdm might not fail to write, but it's definitely not what you wanted.

I just updated sdm today so that these dangling mounts and loop devices should be FAR less frequent, as well as trapping and more cleanly handling CTRL/C  (trying to underpromise here). This would ensure that the scenario I described above shouldn't happen again.

Hope this helps!

ShovelJockey, 2022-09-24:
I referenced /mnt/sdm/ but realised I missed a well placed comment telling met use $SDMPT/ in the custom script template.... I assume this might be part of the issue? I'll check the version im using but definitely have not gotten the most recent update, but that sounds great.

gitbls, 2022-09-24:
As I suggested, this is likely the whole issue. If it's still a problem after you correct your custom script, we'll take the next step...

ShovelJockey, 2022-09-26:
This change has resolved my issue but I do occasionally have issues with mounts and loop devices so will update and see if those are less frequent too thanks for the help.

gitbls, 2022-09-26:
Absolutely! Closing this issue since resolved. After you update to the new version please let me know if you're still seeing any dangling mounts or loop devices.

closed 2022-09-26 by gitbls