gitbls/sdm #35: sdm does not work if called from a symlink

arpj-rebola, 2022-10-25:
````
$ sudo ln -s /usr/local/sdm/sdm /usr/bin/sdm
$ sudo sdm
/usr/bin/sdm: line 606: /usr/bin/sdm-cparse: No such file or directory
/usr/bin/sdm: line 607: iswsl: command not found
? No command specified (--aptmaint --burn, --customize, --explore, --mount)
````

This issue originates in `sdm` line 605: `src=$(dirname $0)` can be replaced by e.g. `src=$(realpath "$0")`.

gitbls, 2022-10-25:
Thanks for reporting this. I'll get it into the next update.

gitbls, 2022-11-02:
I'm nearly done with the next update and have fixed this, and I have updated EZsdmInstaller to add the link into /usr/local/bin

gitbls, 2022-11-04:
This edit is in V7.0. Thanks again for your suggestion!

closed 2022-11-04 by gitbls