gitbls/sdm #36: "Set new disk ID" fails for some versions of sfdisk

arpj-rebola, 2022-10-27:
````
$ sudo sdm --password-pi --host frambuesa --expand-root --burn /dev/mmcblk0 2022-09-22-raspios-bullseye-arm64.img
* Burn '2022-09-22-raspios-bullseye-arm64.img' to '/dev/mmcblk0'...
dd if=2022-09-22-raspios-bullseye-arm64.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
4265607168 bytes (4.3 GB, 4.0 GiB) copied, 286 s, 14.9 MB/s
254+1 records in
254+1 records out
4265607168 bytes (4.3 GB, 4.0 GiB) copied, 363.394 s, 11.7 MB/s
cat: /sys/block/mmcblk0/mmcblk02/size: No such file or directory
> Expand Root: Expand partition from 0.0GB (0.0GiB) to 55.5GB (59.6GiB)
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk0: 59.49 GiB, 63864569856 bytes, 124735488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x63ee4f38

Old situation:

Device         Boot  Start     End Sectors  Size Id Type
/dev/mmcblk0p1        8192  532479  524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 8331263 7798784  3.7G 83 Linux

/dev/mmcblk0p2: 
New situation:
Disklabel type: dos
Disk identifier: 0x63ee4f38

Device         Boot  Start       End   Sectors  Size Id Type
/dev/mmcblk0p1        8192    532479    524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 124735487 124203008 59.2G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
> Set new disk ID '828d21d2' on '/dev/mmcblk0'
sfdisk: unrecognized option '--disk-id'
Try 'sfdisk --help' for more information.
? sfdisk error setting diskid 1
````
This error originates in `sfdisk` - apparently, some versions of it do not admit the '--disk-id' option.
````
$ sfdisk --help

Usage:
 sfdisk [options] <dev> [[-N] <part>]
 sfdisk [options] <command>

Display or manipulate a disk partition table.

Commands:
 -A, --activate <dev> [<part> ...] list or set bootable (P)MBR partitions
 -d, --dump <dev>                  dump partition table (usable for later input)
 -J, --json <dev>                  dump partition table in JSON format
 -g, --show-geometry [<dev> ...]   list geometry of all or specified devices
 -l, --list [<dev> ...]            list partitions of each device
 -F, --list-free [<dev> ...]       list unpartitioned free areas of each device
 -r, --reorder <dev>               fix partitions order (by start offset)
 -s, --show-size [<dev> ...]       list sizes of all or specified devices
 -T, --list-types                  print the recognized types (see -X)
 -V, --verify [<dev> ...]          test whether partitions seem correct
     --delete <dev> [<part> ...]   delete all or specified partitions

 --part-label <dev> <part> [<str>] print or change partition label
 --part-type <dev> <part> [<type>] print or change partition type
 --part-uuid <dev> <part> [<uuid>] print or change partition uuid
 --part-attrs <dev> <part> [<str>] print or change partition attributes

 <dev>                     device (usually disk) path
 <part>                    partition number
 <type>                    partition type, GUID for GPT, hex for MBR

Options:
 -a, --append              append partitions to existing partition table
 -b, --backup              backup partition table sectors (see -O)
     --bytes               print SIZE in bytes rather than in human readable format
     --move-data[=<typescript>] move partition data after relocation (requires -N)
 -f, --force               disable all consistency checking
     --color[=<when>]      colorize output (auto, always or never)
                             colors are enabled by default
 -N, --partno <num>        specify partition number
 -n, --no-act              do everything except write to device
     --no-reread           do not check whether the device is in use
     --no-tell-kernel      do not tell kernel about changes
 -O, --backup-file <path>  override default backup file name
 -o, --output <list>       output columns
 -q, --quiet               suppress extra info messages
 -w, --wipe <mode>         wipe signatures (auto, always or never)
 -W, --wipe-partitions <mode>  wipe signatures from new partitions (auto, always or never)
 -X, --label <name>        specify label type (dos, gpt, ...)
 -Y, --label-nested <name> specify nested label type (dos, bsd)

 -G, --show-pt-geometry    deprecated, alias to --show-geometry
 -L, --Linux               deprecated, only for backward compatibility
 -u, --unit S              deprecated, only sector unit is supported

 -h, --help                display this help
 -v, --version             display version

Available output columns:
 gpt: Device Start End Sectors Size Type Type-UUID Attrs Name UUID
 dos: Device Start End Sectors Cylinders Size Type Id Attrs Boot End-C/H/S Start-C/H/S
 bsd: Slice Start End Sectors Cylinders Size Type Bsize Cpg Fsize
 sgi: Device Start End Sectors Cylinders Size Type Id Attrs
 sun: Device Start End Sectors Cylinders Size Type Id Flags

For more details see sfdisk(8).
````

gitbls, 2022-10-27:
Hi, thanks for the report. You didn't say which version of sdm you're using. sdm V6.13 checks for an sfdisk failure, and if so, uses fdisk instead. 

Can you please identify which version of sdm you're using, and what distro/version you're running it on?

arpj-rebola, 2022-10-28:
That's right - I updated `sdm` to version 6.13 and now it works. Thank you!

gitbls, 2022-10-28:
Excellent!

closed 2022-10-28 by gitbls