gitbls/sdm #37: Use of systemd --version instead of systemctl --version

alexandrebarsacq, 2022-11-08:
Hi, I'm trying to make this run on arch linux and I get an error when there are calls to `systemd --version`. Changing them to `systemctl --version` seems to do the job. When trying on debian it seems that both commands output the same info, so I'm wondering if there is any issue to use systemctl instead of systemd command ?

gitbls, 2022-11-08:
Curious. I wonder why arch decided to disable `systemd --version`. 

To answer your question, shouldn't be a problem. I'll look a bit more closely at this and switch to using systemctl if no issues (not that I'm expecting any!).

Let me know how your port is going and happy to help with questions you get stuck on.

gitbls, 2022-11-11:
The next update will have this change in it, as well as some related simplifications.

gitbls, 2022-11-17:
Fixed in V7.2

closed 2022-11-17 by gitbls