gitbls/sdm #48: Error when using `sdm --customize`

tiefenauer, 2023-01-14:
I read through the documentation and was testing out the sample command as shown in https://github.com/gitbls/sdm/wiki/Example:-Commands. However, I cannot build a customized image, because of the following error:

```bash
$ sudo sdm --customize --poptions apps --hdmigroup 2 --hdmimode 82 --user pi --nowpa 2020-02-13-raspbian-buster.img 
* Mount IMG '2020-02-13-raspbian-buster.img'
mount: /dev/loop27p2 mounted on /mnt/sdm.
mount: /mnt/sdm/boot: mount point does not exist.
? Error mounting IMG '2020-02-13-raspbian-buster.img'
umount: /mnt/sdm unmounted
```

Note: I changed the command from the example a bit by removing irrelevant switches and adding switches the script reported as mandatory:
removed:
- `--apps`
added:
- `--user`
- `--nowpa`

tiefenauer, 2023-01-14:
I'm so stupid: I didn't notice I have to unzip the image first.

```bash
$ sudo apt install p7zip-full p7zip-rar
$ 7z e 2022-09-22-raspios-bullseye-arm64.img.xz
```
Afterwards it works like a charm. I'm closing this issue but leave it for other noobs so they can learn.

closed 2023-01-14 by tiefenauer

gitbls, 2023-01-14:
Glad that you sorted it out. sdm checks for a ".zip" file type, but did not get it updated yet for ".xz". Will take care of this. Thanks for the reminder!