gitbls/sdm #50: Locale warning when customizing image

tiefenauer, 2023-01-15:
When customizing an image using the command below I get a warning about not supported locales:

```bash
$ sudo sdm --customize \
    --os raspios \
    --l10n \
    --locale en_US.UTF-8 \
    --disable bluetooth,piwiz,triggerhappy \
    --user pi \
    --password-same y \
    --password-pi "$user_password" \
    --password-root "$user_password" \
    --rootpwd \
    --host raspibox \
    --nowpa \
    --loadlocal wifi \
    --apssid raspibox \
    --poptions apps \
    --apps @raspibox-apps \
    --dtoverlay hifiberry-dac \
    --ssh service \
    --plugin samba:"" \
    --xmb 4048 \
    --showapt \
    2022-09-22-raspios-bullseye-raspibox-arm64.img
(...)
Reloading keymap. This may take a short while
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
     LANGUAGE = "en_US.UTF-8",
     LC_ALL = (unset),
     LC_PAPER = "de_CH.UTF-8",
     LC_NUMERIC = "de_CH.UTF-8",
     LC_IDENTIFICATION = "de_CH.UTF-8",
     LC_MEASUREMENT = "de_CH.UTF-8",
     LC_NAME = "de_CH.UTF-8",
     LC_TELEPHONE = "de_CH.UTF-8",
     LC_ADDRESS = "de_CH.UTF-8",
     LC_MONETARY = "de_CH.UTF-8",
     LC_TIME = "de_CH.UTF-8",
     LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
(...)
```

The same warning is repeated several times.

I'm running this on Ubuntu 22.04 LTS (64 bit) to customize the latests RaspianOS image found at https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-09-26/2022-09-22-raspios-bullseye-arm64.img.xz.

gitbls, 2023-01-15:
Using `--l10n` sets the locale as well. What are you trying to accomplish by also using `--locale`?

tiefenauer, 2023-01-15:
I was trying out stuff to get rid of the error. Also, I assumed that if both `--l10n` and `--locale` are set, the keymap and wifi country will be taken from `--l10n` while the locale specified by `--locale` will be used, i.e. more specific switches like `--locale` will override  switches aggregated with `--l10n`. 

The docs were not clear on this, so from your answer I take my assumption was wrong. I am trying to create a customized image based on RaspianOS. Above behavior would have been helpful for me because I want to create a script that modifies an image using SDM, leaving some configuration to the user (like the username) or taking some configuration from the user's machine where the script is run while on the other hand hardcoding some things like the locale, which I want to have a value that my script specified.

Anyway, I understand now that the locale will be set by `--l10n` using my host system's (ubuntu) locale, which should be `en_US.UTF-8` (value of `LANG`). But that does not seem to work.

For reference, here are the locales from my Ubuntu system:

```bash
$ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_CH.UTF-8
LC_TIME=de_CH.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_CH.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_CH.UTF-8
LC_NAME=de_CH.UTF-8
LC_ADDRESS=de_CH.UTF-8
LC_TELEPHONE=de_CH.UTF-8
LC_MEASUREMENT=de_CH.UTF-8
LC_IDENTIFICATION=de_CH.UTF-8
LC_ALL=
```
Following the [Ubuntu documentation](https://help.ubuntu.com/community/Locale#List_current_settings), `LANG` will be used for all unset `LC_*` values, while `LC_ALL` will override everything, so I would not want to set `LC_ALL`.

gitbls, 2023-01-15:
For `--l10n` sdm gets the locale via:
```
    source /etc/default/locale
    locale="$LANG"
```
What is the contents of /etc/default/locale on your ubuntu system?

Also, if you describe what you're trying to do, in terms of what you want the scenario to be for your eventual end user, I may be able to offer some suggestions on how to approach it with sdm.

tiefenauer, 2023-01-15:
```
$ cat /etc/default/locale
#  File generated by update-locale
LANG=en_US.UTF-8
LC_NUMERIC="de_CH.UTF-8"
LC_TIME="de_CH.UTF-8"
LC_MONETARY="de_CH.UTF-8"
LC_PAPER="de_CH.UTF-8"
LC_NAME="de_CH.UTF-8"
LC_ADDRESS="de_CH.UTF-8"
LC_TELEPHONE="de_CH.UTF-8"
LC_MEASUREMENT="de_CH.UTF-8"
LC_IDENTIFICATION="de_CH.UTF-8"
LANGUAGE=en_US.UTF-8

```

gitbls, 2023-01-15:
I need to look into this further. I've not seen any locale errors like this. Ever.

gitbls, 2023-01-19:
I have a fresh Ubuntu 22.04 install, and sdm works correctlly without this error. However, my /etc/default/locale only has the LANG=en_US.UTF-8 line in it.

Can you detail how you change your locale on Ubuntu such that I can repro the contents of your /etc/default/local?

gitbls, 2023-01-19:
Well, I of course tried putting your /etc/default/locale onto my system and did not get the errors your saw. I don't know Ubuntu management very well, so was hoping to figure out an easy repro.

At this point, after an hour or two trying and without a reliable way for me to create a repro, I'm not able to proceed further on investigating this issue.

tiefenauer, 2023-01-28:
I think it has to do with the way Debian ships locales:

> Debian ships locales in source form. They need to be compiled explicitly. The reason for this is that compiled locales use a lot more disk space, but most people only use a few of them.

Source: https://unix.stackexchange.com/a/269293

I added `sudo dpkg-reconfigure locales` to the custom phase script (in phase 1), and then manually selected the languages I wanted to generate. In my case it was `en_US.UTF-8`, `en_US.ISO-8859-15`, `de_CH.UTF-8` and  `de_CH.ISO-8859-1`, since those were the languages used in my `/etc/default/locale`.

After selecting "OK", the locales were generated:

```
Generating locales (this might take a while)...
  en_US.UTF-8... done
  de_CH.ISO-8859-1... done
  de_CH.UTF-8... done
  en_US.ISO-8859-15... done
```

Comparing that to the output of SDM (without custom phase script) I think there's something wrong with SDM:

```
> Set Locale to 'en_US.UTF-8'
Generating locales (this might take a while)...
  en_US.UTF-8... done
```
It seems only `en_US.UTF8` is being generated, althought the `--l10n` switch should read out the locales from `/etc/default/locale`.

Not sure if this helps. Everything seems to be working correctly, but with all those errors on the console I'm not 100% sure.

gitbls, 2023-01-28:
sdm currently picks up whatever is in LANG (from /etc/default/locale) when you use `--l10n`. If you want additional locales, you can add them, as you noted, using `dpkg-reconfigure locales` in your custom phase script, so you have a workaround.

[This link](https://askubuntu.com/a/1027038) demonstrates how to add multiple locales non-interactively, which is probably what you want to do in your custom phase script:
```
echo "locales locales/default_environment_locale select en_HK.UTF-8" | debconf-set-selections
echo "locales locales/locales_to_be_generated multiselect en_HK.UTF-8 UTF-8" | debconf-set-selections
rm "/etc/locale.gen"
dpkg-reconfigure --frontend noninteractive locales
```
I haven't tried this, but it looks close to correct.

gitbls, 2023-03-18:
Closing this due to lack of further activity. Please re-open if you want to pursue this further.

closed 2023-03-18 by gitbls