gitbls/sdm #54: `sdm --burn` fails with errors

tiefenauer, 2023-01-15:
I downloaded the latest RaspianOS image from https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-09-26/2022-09-22-raspios-bullseye-arm64.img.xz and tried to burn it to an SD card using sdm. However, the process does not run through cleanly, as there are errors on the console after the actual burning has finished:

```
$ sdm --burn /dev/mmcblk0 2022-09-22-raspios-bullseye-arm64.img 
[sudo] password for daniel: 
% hostname not specified with --host; will not be written
* Burn '2022-09-22-raspios-bullseye-arm64.img' to '/dev/mmcblk0'...
dd if=2022-09-22-raspios-bullseye-arm64.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
4261412864 bytes (4.3 GB, 4.0 GiB) copied, 186 s, 23.0 MB/s
254+1 records in
254+1 records out
4265607168 bytes (4.3 GB, 4.0 GiB) copied, 185.808 s, 23.0 MB/s
> Set new disk ID 'ce5cc681' on '/dev/mmcblk0'
Disk identifier changed from 0x63ee4f38 to 0xce5cc681.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or partx(8).
Syncing disks.
> Disk ID ce5cc681 set successfully with sfdisk
mount: /mnt/sdm.167869: mount(2) system call failed: File exists.
mount: /mnt/sdm.167869/boot: mount point does not exist.
> Set new Disk ID 'ce5cc681' in /etc/fstab and /boot/cmdline.txt
sed: can't read /mnt/sdm.167869/etc/fstab: No such file or directory
sed: can't read /mnt/sdm.167869/boot/cmdline.txt: No such file or directory
sed: can't read /mnt/sdm.167869/boot/cmdline.txt: No such file or directory
grep: /mnt/sdm.167869/boot/cmdline.txt: No such file or directory
sed: can't read /mnt/sdm.167869/boot/cmdline.txt: No such file or directory
sed: can't read /mnt/sdm.167869/boot/cmdline.txt: No such file or directory
> RasPiOS firstboot service not disabled; at least one of --expand-root and --regen-ssh-host-keys not set
% IMG '2022-09-22-raspios-bullseye-arm64.img' is not sdm-enhanced
  Logs and configuration updates will not be written
* Storage '/dev/mmcblk0' is ready
umount: /mnt/sdm: target is busy.
```
The Raspberry doesn't boot either, I guess the image wasn't flashed correctly.

gitbls, 2023-01-15:
It looks to me like you have some dangling mounts from prior runs. The mount error on /mnt/sdm.167869 tells me that there is something mounted at /mnt/sdm. At a minimum, you need to find and umount any dangling mounts, and then `losetup -d` all of the sdm-related loop devices shown with `losetup -a`.

I'm guessing that one of your previous runs of sdm errored out, or you ctrl/c'd out of it an an inopportune time and sdm's cleanup handling didn't work. If you have any recollection of how that happened, I'd like to hear about it, or a way to repro it.

The messages
```
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot...
```
Could indicate that you're using an IMG that is still mounted as per above.

Any input you can provide on how you got to this state (from previous customize or burn commands) would be very helpful. In the meantime, I'll have a look at the burn code to see if it can be more observant of such instances and provide a more corherent message.

tiefenauer, 2023-01-15:
I did ctrl-c out of prior runs, but reading the logs I thought smd cleaned up successfully. It's impossible to tell now as the logs are gone. However, your guess makes very much sense, so I'll check next week.

gitbls, 2023-01-17:
In looking intot this I realized that I mis-spoke. The issue is likely twofold:

- Your SD card (/dev/mmcblk0) is still mounted 
- While sdm checks for SD cards connected via USB that are mounted, it doesn't check for them on /dev/mmcblk0

The checking will be improved in the next release

tiefenauer, 2023-01-19:
Confirming that burning succeeds when using `sdm V7.6`when burning an umodified RaspbianOS using built-in SD card reader.

```bash
$ sdm --version
sdm V7.5
$ curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
$ sdm --version
sdm V7.6
$ sudo sdm --burn /dev/mmcblk0 2022-09-22-raspios-bullseye-armhf.img 
% hostname not specified with --host; will not be written
? Partition /dev/mmcblk0p1 is mounted
? Partition /dev/mmcblk0p2 is mounted
? Use 'sudo umount' to unmount each mounted partition, then redo the burn command
$ umount /dev/mmcblk0p1
$ umount /dev/mmcblk0p2
$ sudo sdm --burn /dev/mmcblk0 2022-09-22-raspios-bullseye-armhf.img 
% hostname not specified with --host; will not be written
* Burn '2022-09-22-raspios-bullseye-armhf.img' to '/dev/mmcblk0'...
dd if=2022-09-22-raspios-bullseye-armhf.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
4261412864 bytes (4.3 GB, 4.0 GiB) copied, 220 s, 19.4 MB/s
254+0 records in
254+0 records out
4261412864 bytes (4.3 GB, 4.0 GiB) copied, 253.657 s, 16.8 MB/s
> Set new disk ID '96bb42c7' on '/dev/mmcblk0'
Disk identifier changed from 0x1d124286 to 0x96bb42c7.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID 96bb42c7 set successfully with sfdisk
mount: /dev/mmcblk0p2 mounted on /mnt/sdm.
mount: /dev/mmcblk0p1 mounted on /mnt/sdm/boot.
> Set new Disk ID '96bb42c7' in /etc/fstab and /boot/cmdline.txt
> RasPiOS firstboot service not disabled; at least one of --expand-root and --regen-ssh-host-keys not set
% IMG '2022-09-22-raspios-bullseye-armhf.img' is not sdm-enhanced
  Logs and configuration updates will not be written
* Storage '/dev/mmcblk0' is ready
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted

```

closed 2023-01-19 by tiefenauer