gitbls/sdm #66: Using `--rclocal` multiple times does not work

tiefenauer, 2023-01-27:
I use SDM with the following switches:
```bash
sudo sdm --customize \
(...)
    --rclocal "exec 1>/home/$user_name/rc.local.log 2>&1  # send stdout and stderr from rc.local to a log file" \
    --rclocal "set -x                         # tell sh to display commands before execution" \
    --rclocal "bash /home/$user_name/hotspot.sh" \
```

The resulting /etc/rc.local in the image is:

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

exec 1>/tmp/rc.local.log 2>&1
exit 0
```

Issues:
1. Only the first command was added to /etc/rc.local (I would like to include the comments for later reference and documentation)
2. the comment was not included

Maybe both issues are connected.

gitbls, 2023-01-27:
The code that restores variables from /etc/sdm/cparams mis-handles this. Everything after the first '#' is removed.

For a workaround, you can remove the comments for each line and the code will end up in /etc/rc.local

Will investigate a solution that enables your goal.

tiefenauer, 2023-01-27:
Thx. It's not terribly important, but would be nice to have. In the meantime I removed the comment. Everything works well with the following switches:

```bash
    --rclocal "exec 1>/home/$user_name/rc.local.log 2>&1" \
    --rclocal "set -x" \
    --rclocal "bash /home/$user_name/hotspot.sh" \
```

gitbls, 2023-01-28:
This should be corrected in V7.7. Please test and verify. Thanks!

gitbls, 2023-03-18:
Closing this as fixed.