gitbls/sdm #67: Am I able to rename the default pi user?

1stcall, 2023-02-01:
Am I able to rename the default pi user to something else using sdm --customize ?

gitbls, 2023-02-01:
sdm does not provide an integrated function to rename the pi user. You can, however, create a new user with the `--user` switch, which creates a user that you can customize as you desire.

In recent RasPiOS builds and all builds going forward, the Pi user comes without a password, and is thus unusable unless you change the password. 

If you really want to rename the pi user for some specific reason, you can do so in a custom phase script or a plugin. Something like this in the Phase 1 code should work:
```
usermod --login myusername --home /home/myusername pi
```
Alternatively, you can use the sdm `--user` switch to create a new user, and then simply `userdel pi`, or just leave it, since the account can't be used, as I mentioned above. Personally, I always create a new user, and leave the pi user as is, and have never had an issue.

If you set up the user (using either method) as you'd like, and also set the localization settings (using either `--l10n`, or `--keyboard`, `--locale`, `--timezeone`, and `--wifi-country`) you can also add `--disable piwiz`, which will disable the stuff that RasPiOS runs at first boot to query you for all of these settings. Why do something manually that can be automated? 🤔

1stcall, 2023-02-11:
This would help. Thank you.

I've use the pi-gen feature to simplify nfs share permissions across different systems.

gitbls, 2023-02-11:
I've implemented a `--rename-pi` switch, and it will be in the next version...coming next week.

I do have a couple of questions for you. These will help me understand what you're doing, and I may be able to offer you other sdm-usage suggestions, or you may enlighten me on some useful sdm ideas.

- Sounds like you've been using pi-gen? if that's the case, what is your interest in sdm? Is this instead of pi-gen, or in addition to? What can you do in sdm that you can't do in pi-gen?
- Could you expand on how you "simplify nfs share permissions?" I'm curious because I use nfs on my network but basically have all shares open to all systems on my LAN. That's really easy to do with sdm, including setting up common UID/GIDs across all the systems. Is that what you're doing, or are you doing something more complex?

Thanks!

1stcall, 2023-02-11:
This is instead of pi-gen.  My intention is to build multiple images for a PI cluster.  The cluster will comprise of 4 Raspberry PI4s and a Raspberry CM4.  The CM4 will act as a switch, firewall, router and network boot host.

I wish to make the setup & config automated across Debian updates, while maintaining as few files as i can. 

sdm seem like it will help me to achieve this. 

"including setting up common UID/GIDs across all the systems. Is that what you're doing,"
Yes.

gitbls, 2023-02-11:
Thanks for that! I believe that sdm will help you reduce the config file maintenance. My own personal *stuff* for sdm consists of:

- A bash script to run the sdm customize. I'll be updating the wiki with a newer example in the next few days, incorporating plugins, which made several things easier
- A personal plugin to do all the customizations that sdm doesn't directly support, such as copying my login files and configuring my NFS shares. This will be updated on the wiki as well

Yep, just two files for my own customized sdm usage.

Several of the plugins included with sdm started out in my personal customizations, but once plugins were implemented, I migrated them to plugins so they can be more easily used by others (e.g., apt-cacher-ng, apt-file, and others coming in the next version).

I encourage you to use plugins liberally, as they provide the ability to easily add/remove capabilities in a modular fashion. Also, they can be used both when customizing the IMG (so all burn targets get the common features) or when burning (for only specific targets). This is handy, for instance when setting up servers with different functionality. 

Your use case sounds interesting! If you run into situations where an addition to sdm would be helpful, let me know. I'm always looking for good ideas to incorporate, or maybe the capability is already there! 🤔

1stcall, 2023-02-11:
The next feature I expect to be looking for is to output a output/${rootfs},${bootfs}/ directory structure with the --burn parameter to facilitate network booting.

gitbls, 2023-02-11:
hmm...I've done tftp network booting with x64 systems, but never with the Pi. Are you thinking of a single directory structure for all Pis, or one tree per pi? If the former, how are the per-system items, such as the hostname, applied?

1stcall, 2023-02-11:
I am testing by using sdm --mount and copying the files.

my tree currently looks  like :-

/srv/tftp/${pi-mac-address}/bootfs/
&
/srv/ntfs/${pi-hostname}/rootfs/

gitbls, 2023-02-11:
Ah, cool. I was going to suggest that. There's also the `--directory` switch that lets sdm operate directly on a directory tree. Full disclosure: I haven't tested it in a while, but it _should_ work. If you give it a go and it falls over, LMK that I broke it 🙄

1stcall, 2023-02-11:
this is my fork that I'm experimenting in my fork https://github.com/1stcall/sdm 
take a look at my customize.sh, burnfile.sh & sdm-customphase for my current tests.

gitbls, 2023-02-11:
Looks like you'll be explaining to me how this all works fairly soon!

1stcall, 2023-02-11:
lol.  Have a look at my repo https://github.com/1stcall/build-kernel 

I'm working on including a custom kernel and packaging it as a .deb
I then intend to use sdm to install it on my images.

1stcall, 2023-02-11:
I like the look of the plugins.  I'll play around with them soon.

gitbls, 2023-02-14:
sdm is updated with a `--rename-pi` switch, which renames the pi user. Please validate that it works for you.

1stcall, 2023-02-28:
I can confirm this works well.  Thank you.

1stcall, 2023-02-28:
Also, I'm loving the plugins.  They have significantly  simplified things for me and work very well.

gitbls, 2023-02-28:
Great, thanks for confirming. I'm loving the plugins as well. Glad you find them helpful and simplifying also!

closed 2023-02-28 by gitbls