gitbls/sdm #78: sdm does not work as expected on Quick Start example

ballen4705, 2023-05-06:
I've run the commands from the quick start example to produce a bootable SD card.  When I boot from the card, I am dropped into a 'set your keyboard' dialog.  The documentation says that this should not happen. Am I doing something incorrectly?

Attached is a script (run as root) to reproduce the problem.
[error.txt](https://github.com/gitbls/sdm/files/11412230/error.txt)

gitbls, 2023-05-06:
Ah! Thanks for noticing that the quick start example needs updating. Adding `--disable piwiz` will eliminate the 'set your keyboard' dialog. 

Also, you can remove the need to stuff "exit" by using the `--batch` switch on the customize.

If you add `--regen-ssh-host-keys` to the customize command sdm will regenerate the SSH host keys during first boot rather than running the RasPiOS firstboot that does it. sdm's regenerate SSH host keys waits until the system time is synchronized so that the date/times on the SSH host keys are correct.

ballen4705, 2023-05-07:
Thanks for the quick feedback.  The "moving target" aspect of Linux (and computers in general) makes it a to produce something that will continue to work "as documented".  I'll give this a try later today.

ballen4705, 2023-05-14:
Sorry for the slow feedback.  I just tested the README.md quickstart instructions and it worked correctly.

gitbls, 2023-08-12:
Added a note that user needs to type 'exit' in V8. Closing this issue due to lack of activity.

closed 2023-08-12 by gitbls