gitbls/sdm #79: --bootscripts documentation confusion

wreedamz, 2023-06-22:
Hi! Thanks for the work on sdm, it's proving to be a great help in consistently creating images. I am interested in running a script on the first boot for my use case. I see the documentation in [Command-Details.md](https://github.com/gitbls/sdm/blob/master/Docs/Command-Details.md) states 
> `--bootscripts` — Directs sdm-firstboot to run the boot scripts in 1piboot/*.sh.

When I couldn't get this to work by creating a directory next to where I was running `sdm`, I dug in further and saw [here](https://github.com/gitbls/sdm/blob/d33b36426a128f095d83bcafb66290c05f7d0bdd/sdm#L396) that the `--bootscripts` flag will

> Run the scripts /usr/local/sdm/1piboot/0*-*.sh during first boot

1. The files have a different naming scheme than mentioned in command details
2. The files must be in a location that is different than the one mentioned in command-details.md

After some more digging I see in `Custom-Phase-Script.md` that 

> If a Custom Phase Script wants to run a script at boot time, even if --bootscripts is not specified, the Custom Phase script should put the script in the IMG in /etc/sdm/0piboot and named 0*-*.sh (e.g., 010-customize-something.sh).

This is actually exactly what I wanted to begin with, it just took a while to find this information. I am using a Plugin but it seems to work with one as well. Now my plugin will copy the scripts that I want into the appropriate directory. This information is not present in the plugin page however so it wouldn't be obvious from looking at that alone.

I'm not sure what the best way to rectify all this is, it just appears to be confusing as is and hard to find details on scripts at first boot. There may even be somewhere this is spelled out better, but it hasn't been easy to find. Perhaps a doc called `first-boot-scripts` could clarify this?

gitbls, 2023-06-26:
I've updated the docs to address this. Please review and let me know if there are any additional clarifications/additions required.

wreedamz, 2023-06-27:
Thank you! This is just what I was looking for

closed 2023-06-27 by wreedamz