gitbls/sdm #80: cron file owner and WRONG FILE OWNER error

wreedamz, 2023-07-07:
When running`sdm` I supply a cron file via `--cron-d foo`. I also specify a user via `--user`. Running `ls -la foo` I see:

```
-rw------- 1 pi pi 405 Jul  7  2023 /etc/cron.d/foo
```

Why is the file owner `pi`? Is this a user account that is always created on the device even though I specify something else via `--user` (I do indeed see `pi` in `/etc/passwd`)? When investing why my task isn't running I see in `/var/log/syslog` (ignore the incorrect datetime)
```
May  3 03:13:01 raspberrypi cron[411]: (*system*foo) WRONG FILE OWNER (/etc/cron.d/foo)
```

files in `/etc/cron.d/` require a user associated with each task as far as I can tell. I was initially trying with the user specified via `--user`. Also the same error message when trying with user `pi` in each task of the cron file.

Reading Debian's [`man cron 8`](https://manpages.debian.org/jessie/cron/cron.8.en.html) I see
> /etc/crontab and the files in /etc/cron.d must be owned by root, and must not be group- or other-writable.

Running `sudo chown root:root /etc/cron.d/foo` seemed to fix my issue. I can change the file owner on boot but I'm guessing this is something sdm should be doing if my understanding is correct.

gitbls, 2023-07-07:
You are absolutely correct that sdm should be handling this. sdm needs to properly chown/chmod all files that it copies into the IMG on the user's behalf. Fix coming soon. Thanks for reporting!

gitbls, 2023-07-07:
This is a critical bug since the resulting IMG does not work correctly/as expected. V7.18 corrects this. Please test and verify.

wreedamz, 2023-07-07:
File owners look correct in explore mode. I'll be able to fully test the image on Monday. Thanks for the quick turnaround!

wreedamz, 2023-07-10:
Thanks again for the quick fix; seems to be working as intended now!

closed 2023-07-10 by wreedamz