gitbls/sdm #81: [feature request] add entries to ssh authorized_keys

nagyben, 2023-07-08:
Hi,

Loving this project so far!

Would it be possible to add a feature to populate `.ssh/authorized_keys` with entries as part of the customize / burn process?

Maybe like specifying the path to a local `authorized_keys` file which is copied into the image, similar to how `wpa_supplicant.conf` is done

gitbls, 2023-07-08:
Great to hear that you're enjoying sdm!

As far as your feature request, it's possible and quite easy to do that today, with a very small amount of work on your part. Have a look at [Example-Plugin](https://github.com/gitbls/sdm/blob/master/Docs/Example-Plugin.md), which I use for customizing my home directory.

You'll find code there that (in Phase 0) copies my .ssh directory from a saved location into the IMG ( cp -a $csrc/home/bls/ssh-mydomain $SDMPT/home/$myuser/.ssh). Note the use of the cp switch `-a` so that owner and protection get copied as well.

The example does this in conjunction with the `--user` switch.

In Phase 0 (before the plugin Phase 0 runs), sdm creates the home directory for the `--user` user (symbol is $myuser), but can't set the ownership of the files in that directory since Phase 0 runs in the context of the host OS.

In Phase 1, running in the context of the OS in the IMG, sdm sets the file ownership of all the files in the /home/$myuser directory tree so that everything is completely correct.

So, you can create a plugin by following [Building Plugins](https://github.com/gitbls/sdm/blob/master/Docs/Programming-Plugins-and-Custom-Phase-Scripts.md#building-plugins). It should be the basic sdm-plugin-template with a couple of lines of code (depending on what you want to do) to copy your .ssh files into the IMG. 

Once your plugin is created, you can use the `--plugin myplugin` switch on either the `--customize` or `--burn` command line to run the plugin. In either case, the plugin will run at the appropriate time with the correct environment and execute the code in your plugin to copy the ssh files into your `$HOME/$myuser/.ssh` directory.

As an aside, plugins are extremely powerful. I'd encourage you to read through a couple of them to get a better understanding of their capabilities. Good examples that are fairly simple: `rxapp`, `apt-file`, and `clockfake`. These are all located in the `/usr/local/sdm/plugins` directory.

Hope this helps! Ask if anything's unclear or if you need further guidance/assistance.

nagyben, 2023-07-17:
Hi,

I was able to implement this using a plugin

```bash
## ssh-authorized-keys

if [ "$phase" == "0" ]
then

    if [ "$myuser" != "" ]
    then
      logtoboth "> Plugin $pfx: Copy $myuser login files to $SDMPT/home/$myuser"
      [ ! -d $SDMPT/home/$myuser ] && mkdir $SDMPT/home/$myuser

      logtoboth "> Plugin $pfx: Copy SSH authorized keys to $SDMPT/home/$myuser"
      mkdir -p $SDMPT/home/$myuser/.ssh
      cp -a authorized_keys $SDMPT/home/$myuser/.ssh/authorized_keys
      chmod 700 $SDMPT/home/$myuser/.ssh

      # backup og root user files
      [ ! -d $SDMPT/root/orig ] && mkdir $SDMPT/root/orig && mv $SDMPT/root/.bashrc $SDMPT/root/orig
      [ -f $SDMPT/root/.bash_profile ] && mv $SDMPT/root/.bash_profile $SDMPT/root/orig

      cp -ar $SDMPT/home/$myuser/.ssh $SDMPT/root/.ssh
      chown -R root.root $SDMPT/root/.ssh
    fi

    logtoboth "* Plugin $pfx: Complete Phase 0"

fi
```

Then run using `sdm ... --plugin ssh-authorized-keys ...`

Thanks for your help!

closed 2023-07-17 by nagyben

gitbls, 2023-07-17:
Great! Thanks for letting me know.  Looking at your script, I have a couple of questions/comments:

- You don't have a /full/path/to/file for authorized_keys. Presumably  you've just omitted it in the snippet you provided? If not, I suggest you add the full path to the source file, since the default directory may not be where you think it is.
- I discovered in Bookworm (the upcoming release of RasPiOS)  that `chown root.root` gets you a warning. The chown command, even in the current release, specifies the owner to be `root:root`. If you change it now, you'll eliminate one future fixit.

nagyben, 2023-07-18:
Yeah, it's fine being a relative path as that works for my specific use case. I was leaving this here as an example in case other people wanted to achieve the same thing.

Thanks for the heads up about `chown`!