gitbls/sdm #83: question: Will sdm work in macOS? Has anyone tried?

stepheneb, 2023-07-20:


stepheneb, 2023-07-20:
SDM scripts use linux specific packages. Would need to be ported to work on macOS.

closed 2023-07-20 by stepheneb

gitbls, 2023-07-21:
Apologies for the late reply, was in the wilderness with poor/no internet.

I don't know macOS well, but indeed, sdm relies on the packages you identified, with the following exceptions.

sdm can use chroot on systems that don't support systemd-container, such as Windows WSL. **Q for you: Is chroot supported on macOS?**

qemu may NOT be required on macOS. **Q: What is the output from the command `file /bin/ls` ?** (binfmt-support is required only for use in conjunction with qemu)

`parted` is required for the `--extend`, `--expand-root` , and `--shrink` functions. 

If there's any possibility of getting sdm to run on macOS via workarounds in sdm, I would be interested in working closely with someone to test and verify sdm there.

stepheneb, 2024-02-13:
It appears looking at the replies to this gist that with the more recent hardened runtime chroot may. not work anymore in macos: https://gist.github.com/eddieh/b3c04436eb6b28826939cd57f1b704bb

gitbls, 2024-02-13:
> It appears looking at the replies to this gist that with the more recent hardened runtime chroot may. not work anymore in macos: https://gist.github.com/eddieh/b3c04436eb6b28826939cd57f1b704bb
> 
> If I find out otherwise and if I can answer your other questions I'll re-open this.

I'm happy to do what I can to enable sdm on MacOS, but there are no Macs here, so unless someone sends me one, I'm 100% dependent on MacOS users to sort out what changes are required.

Of the packages listed in the OP, systemd-container is optional, assuming that `chroot` works. parted is required for anything that mucks with partitions (extend, add, shrink, etc), I don't know how to make sdm work without `qemu` and `binfmt`, since those are required to do a cross-platform chroot, AIUI.