gitbls/sdm #84: Unable to use ':' in plugin argument like  BASE_URL=https\://somwhere.com

benzht, 2023-08-06:
When attempting to pass any 'value' with a colon as a plugin argument, the sdm argument parser wrongly splits 

So, with an sdm-switch like this
`--runonly plugins
--plugin sms-imprint=plugin:"BASE_URL=https\://somwhere.com"
`
the parses uses the _second :_ as the separator between plugin name and arguments and subsequently fails to find `sms-imprint` because it is looking for a file called `sms-imprint=plugin:"BASE_URL=https`. The `\:` is escaped because for parameter parsing, `IFS="|:" read ...` is used and this function requires escaping.

The culprit seems to be operation `p="${p%:*}"` in `sdm` and `sdm-cparse` splitting the arguments off at _the last `:` in the string_. It looks like using `p="${p%%:*}"` instead solves the problem at the 'cost' that `:` cannot appear in the module path or name any more - but I don't see why this would be an issue.

Better still would be to use `IFS=':' read -a plug <<< "$p" because it would allow consistent escaping of the special separation character.

Next issue is in `smd-cparse.runplugins()' where the code `IFS="~" read -a plugs <<< "$plugins"` apparently expands the plugin-string and _strips_ `https\:` to `https:`. Double escaping can solve the issue here (`https\\\:`->`https\:`) but that then fails then in other places.

For now, my work-around will be to use two parameters: one for the protocol and one for the rest of the url after the `:`.

Perhaps some bash magician out there could help to solve this in a more convenient way? Thanks a lot in advance!

gitbls, 2023-08-06:
While I agree that the sdm plugin parser is, er, _weak_, your plugin switch value looks wrong to me. 

As noted in xxx, the canonical plugin switch format is:
```
--plugin plugname:"key1=val1|key2=val2|key3=val3"
```
So, your switch should be
```
--runonly plugins --plugin sms-imprint:plugin="BASE_URL=https\://somwhere.com" 
```
Or similar. I'm actually not sure what _key=value_ you want to pass to your plugin.

That said, I agree with you that the handling of plugin keys/values is a bit weak, and could use some improvement. 

Could you please clarify your plugin name and key/value you want to use?

Thanks!

gitbls, 2023-08-06:
After some focused "tweaks" (aka bug fixes) here's what I'm seeing:
```
p84/ssd/work/lite# sdm --runonly plugins --plugin plugdebug:"arg1=https://somwhere.com"  --oklive --directory /
* Run selected plugins on the host system
> Run Plugins Phase '0'
> Run Plugin 'plugdebug' (/usr/local/sdm/plugins/plugdebug) Phase 0 with arguments: 
  'arg1=https://somwhere.com'
* Plugin plugdebug: Start Phase 0
> Plugin plugdebug: Keys/values found:
   arg1: https://somwhere.com

```
Note that I incorrectly stated what you switch should be in my first reply. The correct syntax is in the example above.

LMK if the above syntax does what you're looking to do. If it doesn't, please explain further.

benzht, 2023-08-07:
Hi - thanks for the fast reply. In my first post, the 'cleaning' of the command line went a step too far. It should have looked like this
`--runonly plugins --plugin sms-imprint-plugin:"BASE_URL=https://somwhere.com|OTHER_PARM=test"`

The syntax as proposed looks like what I'd expected!

gitbls, 2023-08-07:
Great! Hoping to get V8.0 out later this week or early next. 

Please keep this open until you determine that V8.0 corrects the problem.

Thanks for finding this rather nasty bug!

benzht, 2023-08-07:
Thank you for building this rather nifty project 👏 🍰

gitbls, 2023-08-12:
V8 is now available that has this problem corrected. Please verify to make sure that I didn't duff it after fixing it. Thanks!

benzht, 2023-08-12:
Looks like you forgot to commit `plugins/apps` :-)

gitbls, 2023-08-12:
ARRGH. It's there now.

The name= is not needed if you only use the apps plugin once. A default name of 'default' is used. However, if there's more than one apps plugin on your command line, you need to name at least one of them.

Why? Better messages (you can tell which invocation is being worked on in the log). Also, the apps plugin creates a file in /etc/sdm/assets/apps-{name} during phase 0. Then in Phase 1 it picks up that file (based on the name= or 'default') and installs the apps listed there.

There are undoubtedly other ways to do this without requiring the name=, but I thought this was good enough and useful, especially for the logging detail.

Why do you need `--nowpa`? All it did was avoid sdm insisting that you provide a wpa_supplicant file. That restriction/insistence has been removed.

Thanks for your feedback and questions...definitely helpful!

benzht, 2023-08-12:
So, I can just drop the `--nowpa` switch if I understand you right?
Looks like the ':' is now passed correctly to the plugin - still need to check that with phase 3 burning later in the day. Others may still have an issue if one would want the `|` inside a parameter because that does not seem to be escapable (or I' missing how). Not an issue for me, though.

gitbls, 2023-08-12:
Yes, you can drop nowpa. 

As far as the | in values, yes, that's a problem, you might be able to escape it, but I figured I wouldn't worry about it until someone actually found a use for it and raised an issue 🙄.

benzht, 2023-08-12:
> until someone actually found a use for it and raised an issue 🙄.
Good approach. I understand that the IFS splitting caters for `\` as an escape char and would allow `\|` to be part of a value. But I guess somewhere in the processing the string gets interpolated and the '\' vanishes.

Anyhow: I've been running the complete build & flash and everything seems to be working fine.

Thanks again for the fast fix!
    Happy coding!

closed 2023-08-13 by benzht