gitbls/sdm #85: Issues with apt-cache when updating the image and installing apps

tinker2much, 2023-08-11:
I *had* been working on a series of scripts to help reproduce my many custom configurations when RaspiOS bookworm hits, but I just ran across sdm and thought I shouldn't be re-inventing any wheels, and yours might be a LOT easier  ;-)   So far it's all making sense to me and I think I'll be able to do everything I want, though perhaps with some custom scripting. This here is my first trial run on bullseye - an attempt to do everything I can just with the command line - and I'm having some trouble with apt-cacher-ng.  

Here's my command line: 
sudo sdm --extend --xmb 512 --customize --wpa /etc/wpa_supplicant/wpa_supplicant.conf --L10n --restart --disable piwiz --regen-ssh-host-keys --user pi --password-pi mypassword --ssh service --poptions apps,xapps --apps "figlet hdparm mc nmap screen stress" --xapps "passwordsafe" --aptcache 192.168.0.118 --apt-dist-upgrade --disable bluetooth --swap 1024 /mnt/backup/bullseye64.img 

NOTE: sdm is running on "pi8", a pi4-8GB with a 240GB SSD, running 64-bit bullseye. I'm already a fan of apt-cacher-ng, so this machine was ALREADY set up as my apt-cacher-ng server. On this and all of the other client pi's, I add the following line to 02proxy:
`Acquire::http { Proxy "http://pi8.local:3142"; }` And have had no trouble at all with pi8 or any of my other pi's. 

Here, I am (I hope correctly) requesting that sdm set up this image as an apt-cacher-ng client, and it seems to be trying to use my CURRENT cache during its own update and install processes??? I think so because no matter how I code the --aptcache line (with pi8.local, or 192.168.0.118), the error messages reference pi8.local (see below). 

l also noted while nosing around within the sdm shell that you are constructing the apt proxy line differently than I am (see yours at the very end of the pasted text below: 

Here's part of the sdm output and what I did in the shell: 
`2023-08-11 14:18:15 apt update

W: Failed to fetch http://deb.debian.org/debian/dists/bullseye/InRelease  Could not resolve 'pi8.local'
W: Failed to fetch http://security.debian.org/debian-security/dists/bullseye-security/InRelease  Could not resolve 'pi8.local'
W: Failed to fetch http://deb.debian.org/debian/dists/bullseye-updates/InRelease  Could not resolve 'pi8.local'
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Could not resolve 'pi8.local'
W: Some index files failed to download. They have been ignored, or old ones used instead.

2023-08-11 14:18:23 apt install --no-install-recommends --yes figlet hdparm mc nmap screen stress

E: Failed to fetch http://deb.debian.org/debian/pool/main/f/figlet/figlet_2.2.5-3_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/h/hdparm/hdparm_9.60%2bds-1_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/libl/liblinear/liblinear4_2.3.0%2bdfsg-5_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/l/lua5.3/liblua5.3-0_5.3.3-1.1%2bb1_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/libu/libutempter/libutempter0_1.2.1-2_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/l/lua-lpeg/lua-lpeg_1.0.2-1_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/m/mc/mc-data_4.8.26-1.1_all.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/m/mc/mc_4.8.26-1.1_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/n/nmap/nmap-common_7.91%2bdfsg1%2breally7.80%2bdfsg1-2_all.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/n/nmap/nmap_7.91%2bdfsg1%2breally7.80%2bdfsg1-2_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/s/screen/screen_4.8.0-6_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/s/stress/stress_1.0.4-7_arm64.deb  Could not resolve 'pi8.local'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

2023-08-11 14:18:26 apt install --no-install-recommends --yes passwordsafe

E: Failed to fetch http://deb.debian.org/debian/pool/main/q/qrencode/libqrencode4_4.1.1-1_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/w/wxwidgets3.0/libwxbase3.0-0v5_3.0.5.1%2bdfsg-2_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/w/wxwidgets3.0/libwxgtk3.0-gtk3-0v5_3.0.5.1%2bdfsg-2_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/x/xerces-c/libxerces-c3.2_3.2.3%2bdebian-3_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/liby/libyubikey/libyubikey0_1.13-6_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/y/yubikey-personalization/libykpers-1-1_1.20.0-3_arm64.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/p/passwordsafe/passwordsafe-common_1.12.0%2bdfsg-1_all.deb  Could not resolve 'pi8.local'
E: Failed to fetch http://deb.debian.org/debian/pool/main/p/passwordsafe/passwordsafe_1.12.0%2bdfsg-1_arm64.deb  Could not resolve 'pi8.local'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

2023-08-11 14:18:28 apt -y dist-upgrade

2023-08-11 14:18:30 apt --yes autoremove

root@sdm:/etc# cd apt
root@sdm:/etc/apt# cd apt.conf.d
root@sdm:/etc/apt/apt.conf.d# ls
01autoremove  01autoremove-kernels  02proxy  20listchanges  20packagekit  50raspi  70debconf
root@sdm:/etc/apt/apt.conf.d# cat 02*
Acquire::http::proxy "http://pi8.local:3142";
`

gitbls, 2023-08-11:
Hi, thanks for taking a look at sdm. I agree with you, it's going to be a lifesaver when moving from bullseye to bookworm. 

As far as your apt-cacher problem, I guess I'm not terribly surprised that pi8.local doesn't work inside sdm. I'm guessing that MDNS (implemented in Avahi on Linux) doesn't work inside the systemd-nspawn container so it fails.

Can you try the following?
```
sudo sdm --explore /path/to/imgfile
root# ping pi8.local
```
If that does not resolve, then my hypothesis is correct. If it doesn't...well, it will be time to do further investigation. 

I don't run MDNS on my network. (I use [ndm](https://github.com/gitbls/ndm) on one of my Pi systems and assign all my systems DHCP-server assigned static addresses). 

But, if your test refutes my hypothesis, I can set up an MDNS environment to figure out what's going on.

It's certainly possible that the syntax sdm uses in 02proxy doesn't work in some scenario or another, but nobody has reported an issue with it. Until now. 🤔

You could test your hypothesis that the 02proxy format is wrong by changing it to your format and doing an `apt update`. You mentioned that you did this, but didn't say what happens in this case.  Could you `sdm --explore` the IMG, change the proxy to your format, and show me the result?

Also, you didn't mention what the result was if you used the IP address rather than pi8.local. I'm interested in that result as well.

As an aside, I've been working on bookworm support in sdm, and will be posting an updated sdm in the next few days that works well with the current beta IMG that RPL has posted. Once the "final" (is anything ever _final_ in software?) bookworm release comes out, I'll make any necessary updates.

Lastly, congrats on using apt-cacher! I think it's awesome and makes sdm just that much better.

tinker2much, 2023-08-11:
> root@sdm:/# ping pi8.local
> ping: pi8.local: Name or service not known

Both your format proxy line and mine work when doing an apt update from within the shell _IF used with a numeric IP address_, and both **fail** if used with the pi8.local address. 

I just reran the configure specifying the numeric IP address, and the apt update and apt install worked:

> Set image as apt-cacher-ng client using server 192.168.0.118
> System journal configuration not modified
> Set Swap Size to 1024MB
> Start 'apt update'
> IMG '/mnt/backup/bullseye64.img' has 2836152 1K-blocks free at start of 'apt update'
> IMG '/mnt/backup/bullseye64.img' has 2836152 1K-blocks free at end of 'apt update'
> IMG '/mnt/backup/bullseye64.img' has 2836152 1K-blocks free at start of Application Installs
> Start Application Installs
> Application Installs: figlet hdparm mc nmap screen stress
> Application Installs Completed
> IMG '/mnt/backup/bullseye64.img' has 2789220 1K-blocks free at start of xapp/XWindows Installs
> Start xapp/XWindows Installs
> xapp/XWindows Installs: passwordsafe
> xapp/XWindows Installs Completed
> IMG '/mnt/backup/bullseye64.img' has 2736284 1K-blocks free at end of xapp/XWindows Installs
> Phase 1 post-app/xapp installation/configuration
> Start 'apt dist-upgrade'
> IMG '/mnt/backup/bullseye64.img' has 2736284 1K-blocks free at start of 'apt dist-upgrade'
> IMG '/mnt/backup/bullseye64.img' has 2453372 1K-blocks free at end of 'apt dist-upgrade'
> Start 'apt autoremove'
> IMG '/mnt/backup/bullseye64.img' has 2453372 1K-blocks free at start of 'apt autoremove'
> IMG '/mnt/backup/bullseye64.img' has 2453372 1K-blocks free at end of 'apt autoremove'
> Set RasPiOS Full to 'console no login' for First Boot
> Desktop Autologin setting 'B3' will be re-enabled after First Boot
> IMG '/mnt/backup/bullseye64.img' has 2453368 1K-blocks free at end of Phase 1 image customization
> Phase 1 Completed
> IMG '/mnt/backup/bullseye64.img' has 2453368 1K-blocks free at end of image customization
> apt reported errors; review /etc/sdm/apt.log

Despite the "apt reported errors", I looked at the log and didn't see anything that looked like a failure. 

I swear I had tried the numeric IP before and it hadn't worked, but I'm not always a careful and systematic tester...

If I ultimately want the pi8.local form of address, I could change this later in a custom script?

gitbls, 2023-08-11:
Thanks for the follow-up testing. So my hypothesis is unfortunately correct. Interestingly, if you had a DNS server on your network instead of relying on MDNS, it would work correctly, since DNS doesn't depend on any additional services that aren't running in the systemd-nspawn.

I'd suggest that you leave the IP address in the IMG in case you ever go back in there and want to do any apt operations. But, that said, you can certainly change it as part of the first system boot. If it were me, I'd do something like this in my personal plugin (or custom phase script if you prefer). I did not test this 🙄
```
cat > /etc/sdm/0piboot/010-fix-apt.sh <<EOF
#!/bin/bash
rm -f /etc/apt/apt.conf.d/02proxy
echo "Acquire::http::proxy \"http://pi8.local:3142\";" > /etc/apt/apt.conf.d/02proxy
EOF
logger "FirstBoot: Establish desired apt proxy host address"
chmod 755 /etc/sdm/0piboot/010-fix-apt.sh
```
For examples of writing 0piboot scripts, have a look at, for instance, /usr/local/sdm/plugins/apt-cacher-ng, at the bottom of the script.

the 0piboot scripts are run by the sdm-firstboot process the first time the system boots. You can see the output in the system log by looking back at the first boot with `sudo journalctl -b -1` (replace -1 by the appropriate number of boots to get back to the first one, of course. You can get the correct N via `sudo journalctl --list-boots`).

Lastly, the "apt reported errors" is curious. sdm simply looks for a line starting with **E:**:
```
    grep -q "^E:" /etc/sdm/apt.log && logtoboth "?!apt reported errors; review /etc/sdm/apt.log"
```
Are you sure there weren't any errors reported in /etc/sdm/apt.log?

tinker2much, 2023-08-12:
Upon trying this again on a new day on a clean image file, there were no reported errors. Perhaps the log is cumulative and there had been older errors? In any case, it worked great. I had previously been running against an image file on a network share, but now I'm running directly on the attached SSD, and the whole process takes 2 minutes and 30 seconds. Awesome!

I'm moving on to see what other of my customizations I can accomplish with custom scripts. I think I'll start with your personal example and nudge it into what I want. I like some of the things you're doing, such as copying .bashrc files and panel or terminal settings.

gitbls, 2023-08-12:
It could certainly be leftover cruft in /etc/sdm/apt.log. It simply looks at the _whole_ log for that E:, there's no easy way to tell if there's something remaining from an earlier run. I've run into that myself at least once.

The personal customizations are what makes sdm so great, IMHO. When I fire up a new system it's got literally _everything_ I need already in place, and ready to fully participate on my network.

If I run into something that's not working correctly on a new system, I fix it in my personal plugin so I never see that issue again. I haven't had anything not working for a while, but back in the early sdm days it happened more frequently. 😁

One thing that I do, which you may not notice, is that I have a network share (/rpi) which has all the stuff I want to pick up in sdm. When I need to change something (my .bashrc for instance), I test the changes somewhere, and when I'm happy with it, put it on the /rpi share so future sdms pick it up.

tinker2much, 2023-08-12:
So are you using /rpi as your $csrc ? 

Could you provide an example of how invoke your personal plugin on the command line?

gitbls, 2023-08-12:
Yes, I use /rpi as $csrc. /rpi is an NFS share. In my fstab addon I of course have /rpi mounted in there for new systems.

My personal plugin is invoked with
```sh
--plugin bls-plugin:random=rng-tools5|nfsserver=1
```
I'll update my personal plugin example in the next couple of days. the random argument tells the plugin to use that package instead of rng-tools (which has way too much log spew for my liking).

nfsserver=1 has the plugin install the nfs-kernel-server package, which I typically do on most of my systems, although at the current time I don't take advantage of that. Fixing that (to take advantage of it) is on my list of things to think about for switching to Bookworm.

I've also done a bunch of work on the script that I use to run sdm (Docs/Example-Commands.md). It's much simpler to use, but more complex internally (go figure that out!). 

I'll update those two Docs soon.

gitbls, 2023-08-12:
PS Note that I could just as easily pass the /rpi pointer into my plugin with an argument instead of using $csrc, but time keeps me from fixing everything all at once 🙄

tinker2much, 2023-08-13:
I have had some success with a minimal custom plugin - I created a directory skeleton so far just leanly-fleshed with .config files and other stuff for my home directory that I pass in as $csrc, I tar it up and store it in /etc/sdm/local-assets during phase0, then cd / and untar it in the post-install phase. I could include stuff from/for any other directory anywhere on the system and the tar would throw it there. 

One or two oddities: ONE: there was an initial quick keyboard dialog, though it had been defaulted to english-us, and TWO: even though I added the command to "--disable piwiz", it still ran when I burned the image and booted. (I used someone else's software to do the burning - could that matter?

gitbls, 2023-08-13:
What! You used other software! You hussy!!! 🤣🤣🤣

Seriously, though, your approach of copying a tarball into the img is certainly a lot easier than copying the files individually (which I do). My excuse is that my plugin grew organically, and it's not busted, so... But the first time it breaks, I'll be overhauling that!

The only thing I'd caution you on is to make sure that you chown the files appropriately for your configuration. I have a nonstandard uid/gid for my user, and I do chown/chmod -R to ensure everything is correct.

As for the oddities, no explanations. Was this on Desktop or Lite, and what version OS (Buster, Bullseye?). Can you post the appropriately redacted command line you used? Nothing is done during the sdm burn that is required to make disable piwiz work.

Lastly, The one reason I'd suggest that you use sdm for burning rather than other software is the burn-time `--expand-root` switch. Why waste boot time for something that can be easily done at burn-time? 🤷‍♂️

tinker2much, 2023-08-13:
Desktop, Bullseye, 64-bit, the latest image right off the pi site. 

Here’s the script I’ve been running to play with things, nothing there worth redacting:

# run sdm to customize a 64-bit base image

cd ~/Downloads

# start with a clean copy each time (fast at 200-300MB/sec)
cp -p  2023-05-03-raspios-bullseye-arm64.img bullseye64.img & progress -mp $!

# customize 
time sudo sdm --batch --extend --xmb 512 --customize --wpa /etc/wpa_supplicant/wpa_supplicant.conf --L10n --restart --disable piwiz --regen-ssh-host-keys --user pi --password-pi mypassword --ssh service --poptions apps,xapps --apps "figlet hdparm mc nmap screen stress" --xapps "passwordsafe" --aptcache 192.168.0.118 --apt-dist-upgrade --disable bluetooth --swap 1024 --csrc /home/pi/bin/csrc --plugin pl-plugin /home/pi/Downloads/bullseye64.img 

# explore the customized image 
#sudo sdm --explore /home/pi/Downloads/bullseye64.img
 
I copied it to a network drive and then “burned” it to a 16GB chunk of a 480GB SSD on a pi4 using RonR’s Multiboot script. I know this may not be the most efficient burning process, but this way I can run lots of different test images on the same hardware, very fast.

tinker2much, 2023-08-13:
By the way, I see that you’ve released some or all of v8, but all the code I’m using is still the previous version from a week or so ago. 

Soon, I’ll need to get the new code and see what I need to change in my plug-in or command line…

gitbls, 2023-08-13:
Yep, noticed that you are not using V8 yet, but no worries. I modified the command line and ran it. Repro'd the problem so multiboot is absolved 👍 Barring unforeseen further issues, should have a fix later today.

gitbls, 2023-08-13:
You found an actual bug 😵 the code that proceses `--disable` was not handling the case of `--disable piwiz --disable bluetooth` correctly.

Actually, it wasn't handling at all. It simply jammed the latest value found , forgetting about the first one, and forged ahead.

I've fixed the code, but you can work around it in your command line by doing `--disable piwiz,bluetooth` instead of having them as two different and separate switches.

Sorry about that. I've got a couple of other things that I need to fix, will take care of them and post a fix for this (in V8) in the next day or so.

gitbls, 2023-08-14:
Since 8.1 will be coming out shortly (likely late tomorrow [Monday] USPT), I'd encourage you to wait for that. As enticement, I'll have a few comments for you tomorrow on your script to help you move to 8.1.

gitbls, 2023-08-14:
Hostname being set on burn is a design decision. Such decisions are based on the principle:
```
Do common things as early as possible (e.g., during customize)
Do system-specific things as late as possible (e.g., during burn).
```
You can set the hostname during customize yourself. Here's the code that sdm uses ($hname has the hostname). 
```
            logtoboth "> Set hostname '$hname'"
            [ "$domain" != "" ] && shn="${hname}.${domain}" || shn=""
            echo $hname > $SDMPT/etc/hostname
            sed -i "s/127.0.1.1.*raspberrypi/127.0.1.1\t$hname $shn/g" $SDMPT/etc/hosts
````
Note that the above assumes that the original hostname is `raspberrypi`, which it is today. For completeness, I'm going to modify the sdm code to get the current hostname rather than _assuming_. You can see how this is done in raspi-config, function `get_hostname` and `do_hostname`. Note that raspi-config's do_hostname also does a `hostnamectl` because the system is up and running. This isn't necessary if done during customize.

Fair question as to what other functions are available only in --burn, will sort that later today, along with the other accumulated 8.1 items.

And, will sort out what's going on with the country code issue.

tinker2much, 2023-08-14:
In my pi world, I always do certain things on a new image - I always tweak some GUI settings and I always install a handful of debian packages and I always install a handful of my own projects from my own git server. Then, depending on the pi, I may install or build another two to twenty git projects, some from my own git server or others from Github. I have long had project-specific scripts to handle installing/updating/building each such project. But the imminent release of RaspiOS bookworm made me think I REALLY needed a _system_ to perform all of this more easily and quickly. Thus, my recent personal attempt to do pi customization... 

Since I did not know how to use nspawn or modify images directly, all I did was add two more levels of scrpting - the first level being a pair of scripts (one for full, and one for lite) which would make generic or universally desired customizations to a just booted standard image, including installing and customizing a subset of my own software, and a second level being a much larger set of scripts, one per pi, which would assume a lite or full base produced by the first level, and make the specific and numerous and potentially very complicated customizations to reproduce each pi, including installing and customizing more of my own software, plus installing or building or customizing multiple packages from GitHub. I had the lite and full scripts working, and I had enough examples of the pi-specific ones that I figured I could manage all the rest. 

Now, I'm considering how MY systems/infrastructure could best work with YOUR systems/infrastructure, and I have a few thoughts on which I invite your comments. First, after playing with it the last few days, I think that your system, with enough command-line options and a plugin or two, can easily handle creating my desired generic bases, full or lite. Second, I want to keep and leverage my several dozen existing per-package scripts (they're sometimes complicated and I've sunk a lot of work into them, and they're still useful for ADDING random packages to existing systems.) Third, I think I still need my per-pi specific level of scripts, since they essentially represent the COMBINATIONS of packages I want on each system. 

The question is whether, after using sdm to create the bases, I then just manually run one of my per-pi specific scripts each time after booting a copy of the base, which is very easy, OR, whether I attempt to have sdm run the appropriate per-pi script for me. Perhaps a lot of work to figure that out, just to avoid me running one script of my own after booting. (Although it would be cool, I admit, to have the pis come up with EVERYTHING done.) I have to decide _when_ and _how_ I would invoke my pi-specific scripts in sdm - a plugin during customization, or perhaps during burning, or perhaps during first boot (i'm leaning towards first boot, because that's the real live environment my scripts assume.) Your thoughts?

gitbls, 2023-08-14:
I'll start my response with a story. I really hate doing manual steps, but automating stuff, like Pi configuration, would have taken more time than I had for a very long time due to other projects (on this github), and general time requirements for life 🙄.

That changed in early 2020 🤔. By that time, I had a collection of a dozen or so scripts that I would run after booting the system. But even THAT was getting annoying. So, I started working on a script that would run all my scripts for me and do other system configuration.

After about a month of diddling around with that (remember, this was early covid, big-worry-time for going out, and I was recovering from a "procedure", so I had lots of time 😄)

I re-discovered systemd-nspawn, and figured that if I wanted to do this type of thing, others would as well. So,...sdm.

Early on when there were literally no features, I started adding features from my long list. In retrospect I should probably have done plugins much earlier, but it was only a germ of a thought, so put it aside for later. Once I fully realized how powerful they were, and conveniently how similar to Custom Phase Scripts they were, I knew it was time to build them. Over time I expect to move more sdm features into plugins, as I did in V8 with network, graphics, and apps.

Plugins facilitate adding functionality in a highly modular way. It's super-easy to pass in arguments to a plugin, which means they're easy to extend. They are well-structured, and the rules for what can be done in each phase are clear. They can be run in different ways (during customization, during boot, or on a running system after boot), and since they are well-structured and have relatively simple environmental requirements, they can be run outside of sdm. So, while plugins are integrated into and structured for sdm, they can be used in other systems, or run manually without sdm with minimal effort.

So, that's how sdm began, and how I think about plugins.

When I upgraded to Bullseye, I did partial automation, but used the burn switches b0script and b1script to install stuff, since plugins did not exist at that time. I will be turning those per-host scripts into plugins when I move them forward to Bookworm.

OK, enough about me. On to your comments, although I'm sure you have figured out what I'm going to say. 😄 As I read your note, I realized that you and I went down similar paths in deciding how to solve our "configuration" problem.

My suggestion would be for you pick the N (N between 1 and 3) per-pi or per-package scripts and convert them to plugins. You'll understand a lot about plugins, and when you go through it, you'll understand how to call a plugin without sdm, should you ever want to do that. And then you can decide your path forward based on experience and usage.

If you decide to go totally plugin, there are two approaches. One would be to build a plugin for each package. It sounds like you do things like download from github or a local repository (did you know that the network works inside systemd-nspawn?), and then you probably do some config work, and then build the project. This can all be done inside systemd-nspawn, of course.

The other approach is to put all of them in one script and select the packages to install with a plugin argument. Both viable approaches. I think it's a matter of which you prefer, and also take into account the resulting code complexity.

As far as the per-pi scripts that you want to run at boot, a quick and dirty way to do those would be to build a small plugin that takes the name of a script to run at boot. The plugin copies the script to /etc/sdm/assets with a unique name, and writes a script in /etc/sdm/0piboot with a name in the format 0*-*.sh (e.g., 050-myscript1.sh). As long as you use unique names, you can run the plugin multiple times during a customize, ending up with a set of scripts that all get run at boot time. Alternatively, you can simply put the scripts in /etc/sdm/0piboot, but I prefer a little more structure.

Huh, now that I've written this, I think I'm going to add this plugin to sdm. 😄

But, before doing too much of the above, I'd suggest copying all of your scripts into an amply-oversided IMG, `--explore` into it and run each of them to see how well they run. Or not. If they run well, great! If there are issues, you now have an idea of the scope of the work that you'd need to undertake. Pick off some easy-to-resolve items, do the above, and see how that works for you.

I'm happy to have an in-depth look at your scripts so I can offer more concrete suggestions. If you want to share them with me, you can post them here or email me (email on the Docs/Description page).

tinker2much, 2023-08-14:
Thank you for your comments! I think I will upgrade to 8.1, modify whatever I’ve done so far as needed for 8.1, THEN start playing with invoking various of my other scripts from within the sdm framework. Some parameterized way would be good, and some way that would let me leave those scripts unchanged, at least for now, allowing me to also run them my way. 

"If you decide to go totally plugin, there are two approaches. One would be to build a plugin for each package. It sounds like you do things like download from github or a local repository (did you know that the network works inside systemd-nspawn?), and then you probably do some config work, and then build the project. This can all be done inside systemd-nspawn, of course.”

… except that lots of my scripts reference my git server or other sources of data as .local addresses, which don’t work within systems-nspawn.  Which is why I was thinking of invoking the new sdm-over-my-stuff wrapper during boot.

gitbls, 2023-08-14:
> I did try "--disable piwiz,bluetooth" and that's working fine. 8.1 sounds worth the wait. However, I did rerun my latest test this morning, just to make sure I was seeing two things straight last night. One, the hostname is not set, now looking back at the docs and code, I see that the --host command is not performed unless --burn, but I would like to use it during --customize too, at least in my current workflow. What other commands are only available in --burn? Two, when I boot the resulting image and startx, the wifi is working as shown by hovering over the the networking icon, but if I click on it, it prompts me to set the country code, i.e., it seems to think the country code is not set. And it IS there in the wpa_supplicant.conf I passed in. I don't know if this is an sdm thing. And I did check to make sure it wasn't the country code dos line ending problem [see here](https://forums.raspberrypi.com/viewtopic.php?t=353086), that we had both contributed to.

I did a build with Desktop version and the WiFi starts fine. You mentioned `startx`. Was this a Lite build with a desktop added? I need to be able to replicate your build as I can't see an issue here.

gitbls, 2023-08-14:
> Thank you for your comments! I think I will upgrade to 8.1, modify whatever I’ve done so far as needed for 8.1, THEN start playing with invoking various of my other scripts from within the sdm [...]

Remember that plugins are run in the lexical order they are provided on the command line, so you can orchestrate a plugin to run first and modify /etc/hosts in Phase 0 to put in a defintion for xyz.local, and then set it back in the post-install phase. Sorry I didn't think of this sooner, 😵

tinker2much, 2023-08-14:
> > when I boot the resulting image and startx, the wifi is working as shown by hovering over the the networking icon, but if I click on it, it prompts me to set the country code, i.e., it seems to think the country code is not set. And it IS there in the wpa_supplicant.conf I passed in.
> 
> I did a build with Desktop version and the WiFi starts fine. You mentioned `startx`. Was this a Lite build with a desktop added? I need to be able to replicate your build as I can't see an issue here.

This is the full version, but set to boot to the command line with no auto login. I login, then run the startx command to get a desktop. The networking icon, when left clicked, suggests I set a country code.

gitbls, 2023-08-14:
> This is the full version, but set to boot to the command line with no auto login. I login, then run the startx command to get a desktop. The networking icon, when left clicked, suggests I set a [...]

Ah, got it. Thx

gitbls, 2023-08-14:
> I did try "--disable piwiz,bluetooth" and that's working fine. 8.1 sounds worth the wait. However, I did rerun my latest test this morning, just to make sure I was seeing two things straight last [...]

Switches that are only processed in `--burn`:
- `--expand-root` &mdash; Expands the root partition on the burned device. Doesn't make much sense on an IMG
- `--hostname` &mdash; As already discussed
- `--b0script`, `--b1script` &mdash; Only acted on during `--burn`
That's it!

gitbls, 2023-08-14:
Well, as much as I'd like to finish 8.1 today, it's going to have to wait until tomorrow so I can do a little polishing and a lot more testing. The issues you identified are corrected: Multiple `--disable` switches and WiFi country not properly set.

Tomorrow for sure!

gitbls, 2023-08-16:
V8.1 is posted. Everything we've discussed in this thread is corrected 🤞😄.

Just want to point out ICYMI that you can create an entry in your /etc/hosts file for your foo.local apt-cacher-ng server to work around the no mdns in systemd-nspawn.

tinker2much, 2023-08-16:
regarding the new runatboot plugin, about which you write: "The runatboot plugin provides a way to run an arbitrary script during the First Boot of the system. The script is run as root with no other provisions or control made by sdm." 

This would seem like a good way to run my pi-specific script(s). But I don't think I want to run them as root. My scripts assume they're being run as user pi, are intended to install things to /home/pi, but don't always hard-code that, using instead many references to "~" or things off "~".  Plus I want the pi user to be the owner. I think I could fix this by adding the following to the front of the called script (ref: [see here](https://superuser.com/questions/759799/how-do-i-run-commands-in-a-script-as-another-user))

```
if [ "`whoami`" != "pi" ]; then
  sudo -H -u pi bash "$0" "$@"
  exit
fi

```

gitbls, 2023-08-16:
Understood.

You could do that, but based on your feedback I'm going to add a `user` parameter that will do that for you, along with another parameter that enables you to specify the sudo switches (`-H` in your case), as there are a few other switches that someone might find useful.

tinker2much, 2023-08-16:
EZsdmInstaller does not deploy runatboot. I downloaded it myself.

gitbls, 2023-08-16:
Rats and Thanks. Added fix to my growing list of today's updates. 😳

gitbls, 2023-08-16:
Updated to V8.2. Fixed EZInstaller, added args to runatboot, and added a plugin for ufw. Also removed some pointless chmods that I finally maxed out on 🤣

tinker2much, 2023-08-17:
I'm confused by the locations of the double quotes in the runatboot examples [seen here](https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#example).
There's also a reference to stdout that wasn't changed to output.

gitbls, 2023-08-17:
Yep, I'm confused by the double quotes as well. I guess you can tell that I originally named 'output' as 'stdout'. Sigh.

tinker2much, 2023-08-17:
I'm using the following --lxde-config string:
`--lxde-config lxterminal:/home/pi/Documents/sdm/myfiles/lxterminal.conf `
I do see in the log how that file is picked up and put in the image: 
`> Copy --lxde-config 'lxterminal' config file '/home/pi/Documents/sdm/myfiles/lxterminal.conf' to /mnt/sdm/etc/sdm/assets`
but I'm not seeing anything in the log about moving it to ~/.config, and, in fact, I end up with a generic lxterminal.conf, not this one.

gitbls, 2023-08-17:
Ah. I moved the code that actually processes the lxde config files into the graphics plugin. You don't need to provide any arguments to it, just add `--plugin graphics` to your command line. LMK if this does the trick for you.

tinker2much, 2023-08-17:
I'm attempting to use the new runatboot plugin to invoke a specific customization script of my own. I see the set up happening, to make it run at boot:
```
> Run Plugin 'runatboot' (/mnt/sdm/usr/local/sdm/plugins/runatboot) Phase 0 with arguments: 
  'user=pi|sudoswitches=-H|script=/home/pi/bin/refresh_pi3a|output=/home/pi/Documents/sdm/runatboo
  t_refresh_pi3a.log'
* Plugin runatboot: Start Phase 0
> Plugin runatboot: Keys/values found:
   user: pi
   sudoswitches: -H
   script: /home/pi/bin/refresh_pi3a
   output: /home/pi/Documents/sdm/runatboot_refresh_pi3a.log
> Plugin runatboot: Set plugin 'refresh_pi3a' in '/mnt/sdm/etc/sdm/0piboot/060-runatboot.sh' to 
  run at first system boot as user 'pi'
* Plugin runatboot: Complete Phase 0
```
I see the phase 1 and post-install log entries for runatboot, where nothing's supposed to happen (and it doesn't   ;-)  ).
In /var/log/user.log I see it supposedly getting run:
```
Aug 17 11:20:29 raspberrypi root: FirstBoot: Run runatboot scripts
Aug 17 11:20:29 raspberrypi root: FirstBoot: Run runatboot script 'refresh_pi3a' as user 'pi': /etc/sdm/assets/runatboot-refresh_pi3a  >/home/pi/Documents/sdm/runatboot_refresh_pi3a.log 2>&1
```
But there's no log in my home. I think I was confused when I set this up about what directories exist where - that's one that exists on the machine doing the customize but wouldn't exist on the machine that's booting.  
On some previous run, I believe I saw some message about permissions, but I can't find that in a log and don't have any more detail to report.  I wondered at first if the permissions were about my script, but perhaps they're about that log. 'll try routing the stdout to some place that WILL exist on the booting machine and see what happens.

gitbls, 2023-08-17:
Currently, sdm doesn't do anything to make sure that the directory exists, or even complain if it doesn't. There's a case to be made that it should, but there are owner/protection questions that you can resolve more appropriately than sdm.

Arguably some better error checking/reporting would be helpful.

Your thoughts?

tinker2much, 2023-08-17:
I will keep trying to see what's happening and eliminate whatever silly things I'm doing first, before I speak to what sdm might do

tinker2much, 2023-08-17:
Regarding runatboot, I put the output in a better place and now it says this:
`sudo: unable to execute /etc/sdm/assets/runatboot-refresh_pi3a: Permission denied `

tinker2much, 2023-08-17:
> Ah. I moved the code that actually processes the lxde config files into the graphics plugin. You don't need to provide any arguments to it, just add `--plugin graphics` to your command line. LMK if this does the trick for you.
> 
> I'll update the documentation to note this.

That worked, thanks.

gitbls, 2023-08-17:
> Regarding runatboot, I put the output in a better place and now it says this: `sudo: unable to execute /etc/sdm/assets/runatboot-refresh_pi3a: Permission denied `

Potentially two problems. I'd appreciate your thoughts on the 2nd issue

First problem: If you provide a script that doesn't have mode 755, sdm doesn't ensure that it's executable

Second problem: /etc/sdm is protected 700 since there can potentially be critical information there (passwords, etc). Obviously, running a script down the tree while a non-root user isn't going to work. I'm thinking about how to best address this.

tinker2much, 2023-08-17:
I had imagined that you would generate a 0piboot script which would invoke whatever script I gave you, wherever I gave it to you. IOW, the path/to/myscript is assumed to be on the booting box already (however I've managed that). Whereas, you're assuming the path/to/script is coming from the customizing box and therefore you are copying it to and running it from /etc/sdm/assets?

gitbls, 2023-08-17:
> I had imagined that you would generate a 0piboot script which would invoke whatever script I gave you, wherever I gave it to you. IOW, the path/to/myscript is assumed to be on the booting box [...]

Yea, I took a shortcut and got caught. sdm can't assume that the script is actually on the booting box, so it copies it during Phase 0.

I think the correct answer is: if user is specified, before running the script copy the file to /tmp, chown and chmod it appropriately, and run it from there. 

Does that work for you?

tinker2much, 2023-08-17:
I'm OK with your proposal to move it to temp and execute from there (it's your code), but I'm not sure why you can't assume its already on the box (or up to the user to have gotten it to the box via some previous sdm step)? Then you could just execute it from where you're told, and there's no moving.

gitbls, 2023-08-17:
> I'm OK with your proposal to move it to temp and execute from there (it's your code), but I'm not sure why you can't assume its already on the box (or up to the user to have gotten it to the box via [...]

Your point is good, but in literally every other instance where a file is provided by the user, the file is on the sdm host, not in the IMG. I think it would be confusing to explain, and for users, if sdm starts growing exceptions like this one.

The code is relatively straightforward, and keeps this consistent with the rest of sdm.

tinker2much, 2023-08-17:
OK, I'm happy for it to be as you proposed. I get the need for design consistency, both for the designer and those trying to grasp the tool. However, I must warn you that I was known for finding creative ways to use my colleagues' software. They wrote things imagining they would be used in certain ways, and I enjoyed finding others. This is easier (and less disconcerting) for someone other than the designer.
 
And, just having to be contrary, the scripts I'm going to be calling via this mechanism are going to make all kinds of assumptions internally about all kinds of other stuff of mine being there already, and I know it's my responsibility to have gotten such there already.

tinker2much, 2023-08-17:
A side thought - what's your thinking on setting or changing things in boot/config.txt via editing settings in 1piboot/1piboot.conf, or changing them via commands like  --bootadd or --bootconfig or --bootset

gitbls, 2023-08-17:
> OK, I'm happy for it to be as you proposed. I get the need for design consistency, both for the designer and those trying to grasp the tool. However, I must warn you that I was known for finding [...]

All I can say is: Bring it on! 🤣

gitbls, 2023-08-17:
Indeed, there are (at least) 3 ways to modify /boot/config.txt: a) do it yourself, b) use 1piboot/1piboot.conf, and c) use command line switches.

I personally like the command line switches, since they are visible from, er, the command line, which is captured in /etc/sdm/history

1piboot.conf is invisible and hidden unless you know to go look for it. I could log 1piboot.conf if it's actually got anything in it, to eliminate the (in)visibility issue, but at the moment, it remains more or less invisible.

As far as doing it yourself, that's of course an option. You could simply append your settings to the end of the file, or `sed` away. 

It's really a matter of personal preference and comfort.  

I made a note a few days ago to think about building a bootset plugin that would support the features of the command line switches, just to get them out of the mainline code. But, _someone_ is keeping me busy with their tortured uses of sdm 🤣🤣🤣

gitbls, 2023-08-17:
plugin `runatboot` updated to copy and execute the file in /tmp if `user` argument provided.

tinker2much, 2023-08-17:
I hope to try that this evening, else tomorrow, thank you. 

Re: "tortured uses of sdm", I would still appreciate your letting --host happen in the --customize call.

Justifications (seriously): 1) there IS a use case for not using your --burn in which case allowing --host would be much _easier_ on those users and 2) while the few other things with this limitation only seem to make sense in --burn, there's nothing about --host that means it _couldn't_ work anywhere.

gitbls, 2023-08-17:
🤯 Sigh. There's no technical reason not to do it, so I'll put my sdm righteousness aside and enable `--host` on customize.

gitbls, 2023-08-18:
I will push an update tomorrow morning USPT that enables hostname on customize, and also checks (at first boot) for a non-existent user given to the runatboot plugin.

gitbls, 2023-08-18:
V8.3 up. Set your hostname during customize is there, as is runatboot checking for an unknown user (late refinement 😄)

tinker2much, 2023-08-18:
Thank you! I happened to try a git pull just now and got v8.3, already starting a test. Will verify hostname and runatboot-as-user.

tinker2much, 2023-08-18:
Hostname got set just fine, thanks again. I do appreciate the effort. 
Also, regarding runatboot, you and sdm v8.3 invoke my script just fine - as user pi, sudoswitches applied, the output goes where I send it, etc., so all of that infrastructure of yours is working fine too. 
BUT I'm in a bit of a harder place. It would be very cool to do the entire install and customization via sdm, but I think I'll need to do a good bit of work _on my side_ to allow this final complicated script, or scripts called by that script,  some of which I don't have control over, to function in a non-interactive environment. Some pieces worked fine, but some pieces popped up questions, and there's nobody there to respond. I'll keep playing for a while. Otherwise, I'll just have to run a script or two right after booting what sdm produces, that's not onerous. And sdm still does a LOT for me, and provides the framework to easily do any number of customizations that I might want to add in the future. 
Thank you again for your help and support and responsiveness. Off to play some more...

gitbls, 2023-08-18:
I wonder if you could set runatboot's output and error to the console (`/dev/console`) to enable interactivity?

Thanks for the extended interactivity on making it work for you. 👍

Happy playtime. I'm off to work on the next round of _stuff_.

tinker2much, 2023-08-23:
OK, I just typed a huge comment here and somehow I erased it instead of sending it, but I'll try to recreate it while its fresh in my mind...

As a programmer, It makes sense to consolidate similar functionality into one place in the code. If you leave the arguments the same, just move them about a bit, as a user, I wouldn't care beyond the slight onetime bit of work to switch over.

I DO have some questions about the meaning of these settings, regardless of where you implement them, and maybe that's just a docs thing (or maybe just a me not reading the docs enough thing.) 

At one point, I saw the following three as a set covering the boot/config.txt space as I'll describe below, though I think now that I've got the third one wrong:
```
--bootadd key:value,key:value,... — Add new keys/values to /boot/config.txt

--bootconfig key:value,key:value,... — Update existing, commented keys in /boot/config.txt

--bootset key:value,key:value,... — Change system configuration settings. See Bootset and 1piboot. Note that key=value can also be used.

```
I thought that bootadd covered adding "new" keys that didn't appear in /boot/config.txt, that bootconfig covered keys that existed but were commented, and that bootset must therefore cover keys that existed and were live, NOT commented. The recently noticed "See Bootset and 1piboot" now makes me think I had that wrong. But still, how to handle existing uncommented keys??

I had asked previously about your opinion of using 1piboot.conf versus the various boot commands; you had thought that using the overt settings is preferable, and I agree. What I may not understand, though I have a couple of guesses, is precisely how the Venn diagrams of what the two can effect overlap. One idea (from some doc and code skimming, but no in-depth code study) is that every value in 1piboot.conf relates to your batch use of raaspi-config to set various things, some of which are set in boot/config.txt, and some of which are set elsewhere, _and_ only those items that appear in 1piboot.conf are open to setting. The second idea is that the bootadd, bootconfig, hdmiforcehotplug, and other commands you mentioned above are _only_ to set things in /boot/config.txt, although they can set _anything_ in that file's purview.

tinker2much, 2023-08-23:
... and BTW, I am continuing to try to mash your sdm and my customizing scripts together. Originally, pre-sdm, I had imagined the many steps it would take to get 23 pi's re-customized under bookworm - first, burn and boot a new image and customize it by hand with the (mostly) GUI settings that I always wanted. Then write that image to a shared drive using RonR's Image File Utilities, from which I would be able to burn it to many other cards, each of which would have to be booted and where I would then apply one or two additional layers of scripted customizations, one for common software, and one for very specific software at the individual pi level, both of which depended on some mix of very detailed third-level scripts, each of which would handle installing one project.

Once I found sdm, I found I could skip the mostly GUI customization and even booting to do it, yay. I had been trying to get sdm to call both the layers of software-customizing scripting, via runatboot, but that had pitfalls because of possible and hard to remove interactivity in my second level of pi-specific scripts.

BUT, I kept going and set out to see how much of the _first_ level of common scripted software installation and customization I could do via sdm, and I think I've got _that_ working! And that's good enough - I'm fine with running one remaining pi-specific script one time on each pi, that's still hugely more efficient. I run a script which invokes sdm's customize and get a very nice common customized image (or maybe several, depending on 32 vs 64-bit, and full vs lite), with both my desired GUI customizations and various common software already installed. Then, I do a pretty lean burn invocation, with specific hostname and bootadd-level customizations, once per pi. After I've booted each and you've applied those last bits for me, I just run one existing very specific script per pi, thus (re)creating my extremely diverse fleet of pi's, as if by magic compared to the imagined tinkery scripted solution. (And WAY easier than the entirely manual way - from notes - that I would have had to do, before either sort of automation!)

The only hard part has been that I had to make some modest changes to A LOT of my underlying pieces and parts, to let them be called in multiple different ways, either how I need them to be called for sdm or how my pi-specific scripts call them, while till letting them be called on their own, as when I would use one to add some single package to a pi at a later date. But thats what git branches are for, and I don't seem to have broken anything major. A bit more testing, merge the changes into main, and I think I'm ready for bookworm. I hope I'm just in time, because after all this effort, I want RasPiOS Bookworm released _soon_ so that I can just get to it!

gitbls, 2023-08-23:
> As a programmer, It makes sense to consolidate similar functionality into one place in the code. If you leave the arguments the same, just move them about a bit, as a user, I wouldn't care beyond the slight onetime bit of work to switch over.
> 
That's pretty much my feeling about this kind of stuff (yes, happens to me), which is why I try not to make incompatible changes, and when I do (like in this case), I try to minimize the pain.

> I DO have some questions about the meaning of these settings, regardless of where you implement them, and maybe that's just a docs thing (or maybe just a me not reading the docs enough thing.)
>
> At [...]

You have a correct understanding of them. Now that I'm revisiting these, I think it makes sense to just get rid of bootadd, and have bootconfig do one of 3 things:
* If it's in config.txt: a) commented out>>uncomment and set the value, or b) update the value
* If it's not in config.txt, add to the end
Seems like that would eliminate confusion and certainly simplify documentation. Obviously didn't think about that when I did this first time around.

As for bootset, it's for the list of settings that are in 1piboot.conf, which are "simple" raspi-config settings. The switch name isn't really right for what it does. Any thoughts?

> I had asked previously about your opinion of using 1piboot.conf versus the various boot commands; you had thought that using the overt settings is preferable, and I agree. What I may not understand, [...]

Way back when,  raspi-config had a bunch of "simple" settings. 1piboot and bootset provided a simple way to get at those, Not sure why I did both methods, but I definitely wasn't sure where sdm was headed at that point. 

My thinking is that 1piboot.conf is pretty much going to stay the way it is, and remove it at some point in the future. Not in any rush, since it's done, but not going to extend it unless there's a really good reason.

Does this help your understanding?

[7 more replies]

closed 2023-09-23 by gitbls