gitbls/sdm #87: please excuse my ignorance - How can I add a bash script to the image and have it run at first boot?

AlanEbell, 2023-08-18:
Trying to figure out how to add a bash script to the image and make it run after all the rest of the processes are done.  Any help or examples greatly appreciated.

Alan Bell

gitbls, 2023-08-18:
There are several ways to do that.  Can I ask a few questions to better help you?

When you say "add a bash script to the IMG", presumably the script already exists on your host computer or network somewhere, is that correct?

"make it run after all the rest of the processes are done". Do you mean you want it to run at the end of customizing an IMG or at the end of the first system boot? And, by "after all the rest" does it need to run absolutely last, or just "towards the end" of the First Boot?

AlanEbell, 2023-08-18:
It does not I am wanting to add the file when I set up the img using sdm.

Alan

Sent from my Verizon, Samsung Galaxy smartphone

gitbls, 2023-08-18:
"It does not". WHAT does not do what? You're going to have to explain to me what you tried, or what you want to do. 

It's easy to add the script to the IMG (personal plugin or `runatboot`).  

The `runatboot` plugin runs a script as part of the first boot of the system. 

Alternatively, you could use your own personal plugin. With that you could copy your script into the IMG and then cause it to run (by some mechanism) during the first system boot, doing effectively what the `runatboot` plugin does.

If neither of the above help you sufficiently, then please explain in sufficient detail what you're trying to do. My crystal ball is in the repair shop.

gitbls, 2023-08-18:
References for the two methods mentioned above.

* https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#runatboot &mdash; `runatboot` plugin
* https://github.com/gitbls/sdm/blob/master/Docs/Example-Plugin.md &mdash; Code example for a plugin

AlanEbell, 2023-08-18:
Very sorry that was in response se to the email I received which asked of the file was already on the system. I have batch file that I want to deposit if you will in the image file using sdm and then have it run during the first boot up.

Sent from my Verizon, Samsung Galaxy smartphone

gitbls, 2023-08-18:
The resources I previously mentioned document two different ways for doing this:

* https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#runatboot &mdash; `runatboot` plugin
* https://github.com/gitbls/sdm/blob/master/Docs/Example-Plugin.md &mdash; Code example for a plugin

With the first method, `runatboot`, you provide a /path/to/script on your host system. The runatboot plugin copies that into the IMG and runs it during the first boot of the system

With the second method, you would do all the steps that the `runatboot` plugin does, but with your own code. 

It sounds to me like the `runatboot` plugin would be a better approach for you. Try the following:

* `sdm --customize /path/to/img.img --plugin runatboot:"script=/path/to/script" [...plus any other switches you want]`
*  When you boot the system, the script will run as part of the first boot of the system

If you're going to use sdm much, you will benefit by becoming familiar with the sdm documentation, perhaps by skimming through it. You'll get a much better understanding of how it works and how you can make it work for you.

AlanEbell, 2023-08-19:
Thank you.  I am doing so tests.  If I want to make sure a file which my startup script depends on would I use --exports to add that file to the img and then change the path in my startup script to look for it in the /etc/exports directory?  How about if I need more that one file do I just call the --exports command multiple times?

gitbls, 2023-08-20:
I think you are confused about the `--exports` switch. `--exports` is used with the network file sharing system (NFS) and specifies what directories on the system are to be shared (made available) to other systems via NFS, and with what protections.

In other words, it's not the correct switch for you to be using as you have suggested.

sdm doesn't provide a command-line switch to copy files into the IMG, but it's really easy for you to do it yourself.

If you have files that you want to put into the IMG during a `--customize` so that they are available when the system boots, the easiest way to do it is to write your own plugin.

It's pretty simple. use sudo and copy the file /usr/local/sdm/plugins/sdm-plugin-template to /usr/local/sdm/local-plugins/myplugin (or whatever name you want to give it).

Then, edit that file using sudo and your editor, and make the changes you need. For examples of copying files into an IMG in a plugin, see https://github.com/gitbls/sdm/blob/master/Docs/Example-Plugin.md, and look for the line
```
cp -f $csrc/home/$myuser/{.bashrc,.colordiffrc,.dircolors,.emacs,.inputrc,.vimrc,.Xmodmap,.Xdefaults,.tmux.conf} $SDMPT/home/$myuser
```
That line copies some files from a directory on my local system ($csrc/home/$myuser) into the IMG in the $SDMPT/home/$myuser directory.

A simpler example: Suppose you want to copy the file /home/pi/my-special-file.txt into the IMG at /home/$myuser:
```
cp /home/pi/my-special-file.txt $SDMPT/home/$myuser
```

**Some details:**
$csrc holds the value from the command line `--csrc` switch. $myuser is the name of the user provided via the `--user` switch. I recommend using this switch rather than defaulting to the "pi" user, but it's totally up to you. If you don't use the `--user` switch, $myuser will not be defined. 

$SDMPT is a special variable defined by sdm. During sdm Phase 0, the IMG is mounted and is accessed like any other mounted disk. To make sure that you're writing into the IMG rather than your running host system, use $SDMPT.

$SDMPT is not required during the phase 1 or post-install phases, but it doesn't hurt to use it in those phases, since it's defined 
to be the null string then.

Note that the example plugin copies many files into the IMG. You can do the same thing, you don't need to have each file copy in a separate plugin, unless you want to, of course.

Lastly, sdm plugins run as root. If the file needs to be owned by a different user, you'll have to `chown user:group file` to set the file ownership, and you may need to `chmod` it to set the correct file protection.

Also, you might find this page to be helpful, as it describes the sdm phases, and an overall view of writing a plugin: 
 https://github.com/gitbls/sdm/blob/master/Docs/Programming-Plugins-and-Custom-Phase-Scripts.mdI

I hope this helps. If anything is unclear, please ask!

gitbls, 2023-08-20:
If you're having any issues figuring out how to build a plugin to copy the files you want into the IMG, hold off for a bit.

I'm putting together a plugin to do exactly that: Copy specified files into the IMG where you want them, and with the specified owner and protection. 

I'll have it done later today or tomorrow at the latest, and will let you know.

gitbls, 2023-08-21:
Ok, the `copyfile` plugin is available. Easiest would be to simply reinstall via EZsdmInstaller. Hope you find it useful.

AlanEbell, 2023-08-21:
Thank you so much! I am sure I will find it very useful!

Going to give it a try first thing in the morning. It's getting late out here on the east coast!

I am very grateful that you took the time to do this.

Sent from my Verizon, Samsung Galaxy smartphone

gitbls, 2023-08-21:
Apologies...forgot to point to the doc: https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#copyfile

tinker2much, 2023-08-22:
Something to tweak in the doc. The description of the "to" argument says it must be a directory, not a file, but the example shows it as a file...
```
Arguments

from — /full/path/to/sourcefile on the host system
to — /path/in/IMG to place the file in the IMG. This must be a directory, not a file. The directory must already exist
...

Examples

--plugin copyfile:"from=/usr/local/bin/myconf.conf|to=/usr/local/etc/myconf.conf" The config file will be copied from /usr/local/bin/myconf.conf on the host system to /usr/local/etc/myconf.conf in the IMG during Phase1. The file will be owned by the same user:group as on the host, the file protection will be the same as well.
```

gitbls, 2023-08-22:
OOPS.

Since you're using the bootset stuff, can I get your opinion on something? I'm thinking about moving the following command switches into a plugin (and hence removing the command line switches). The names and arguments would stay the same, just changing them into a plugin: bootadd, bootconfig, bootset, dtoverlay, dtparam, hdmigroup, hdmimode, hdmiforcehotplug, hdmiignoreedid.

It reduces complexity in sdm and co-locates all the boot configuration switches (and documentation) in a single plugin.

Your thoughts are appreciated...

tinker2much, 2023-08-23:
BTW, I'm not trying to barge into someone else's thread, but I did try the copy file plugin and it worked great. I had just a single file I wanted to deploy during the burn (having one such file specific to each pi), and copyfile worked great.

gitbls, 2023-08-23:
> BTW, I'm not trying to barge into someone else's thread, but I did try the copy file plugin and it worked great. I had just a single file I wanted to deploy during the burn (having one such file specific to each pi), and copyfile worked great.

No worries, I appreciate the feedback. Good to hear that it worked for you. Still waiting to hear from @AlanEbell on his experience with it, hoping to turn him into a satisfied user as well 👍

AlanEbell, 2023-08-23:
It is working fantastic! I am still wrapping my head around the whole process and it's taking me time to familiarize myself with SDM most of the issues I am finding are with my personal knowledge base so I have been slow to respond as I take my time testing.

I appreciate the plug-in and the software a great deal.

gitbls, 2023-08-23:
Great! Be aware that there's a bug in the plugin: calling it multiple times on the command line does not work correctly. Will be fixed later this morning USPT.

gitbls, 2023-08-23:
Multiple invocation bug is corrected in V8.6. Rerun EZsdmInstaller to pick it up.

AlanEbell, 2023-08-23:
Thank you!

Sent from my Verizon, Samsung Galaxy smartphone

gitbls, 2023-09-23:
Closing this issue since no updates in several weeks.

closed 2023-09-23 by gitbls