gitbls/sdm #88: --bootadd item processed during --customize, but not --burn?

tinker2much, 2023-08-20:
I've just now been playing with setting some things in --burn (having previously done all modifications in --customize, and then burning using another image burning product). This is for an image intended for a pi 3a+ with a pimoroni fan shim. Currently, in either burn case, I install their fan control software in a post-boot customization script of my own. Then I run some stress testing to see how the arm frequency and temperature vary under stress. Besides using the fan, I want to increase the system's temp_soft_limit to 70 in  /boot/config.txt (the default being to start throttling at 60 degrees). I had successfully used --bootadd temp_soft_limit:70 during --customize to set that setting. Didn't seem to be happening when I tried the same parameter during --boot - the throttling was happening at 60 degrees, so I checked and the line DID NOT appear in /boot/config.txt.  

Here are excerpts from the two scripts I used, the first to do --bootadd in --customiize, the second to do it in --burn:
```
sudo sdm \
--batch \
--extend --xmb 512 \
--customize \
--wpa /home/pi/Documents/sdm/myfiles/wpa_supplicant.conf \
--L10n \
--reboot 5 \
--disable piwiz,bluetooth \
--regen-ssh-host-keys \
--user pi --password-pi mypassword \
--ssh service \
--plugin apps:"apps=figlet hdparm mc nmap screen stress|name=apps" \
--plugin apps:"apps=passwordsafe|name=xapps" \
--aptcache 192.168.0.118 \
--apt-dist-upgrade \
--swap 1024 \
--csrc /home/pi/Documents/sdm/csrc \
--plugin pl-plugin \
--lxde-config lxterminal:/home/pi/Documents/sdm/myfiles/lxterminal.conf --plugin graphics \
--bootset boot_splash=1,boot_behaviour=B1 \
--host pi3a83 \
--bootadd temp_soft_limit:70 \
/home/pi/Downloads/bullseye64.img 

```
```
IMGDIR=~/Downloads
CSRC=/home/pi/Documents/sdm/csrc

time sudo sdm \
--batch \
--extend --xmb 512 \
--customize \
--wpa /home/pi/Documents/sdm/myfiles/wpa_supplicant.conf \
--L10n \
--reboot 5 \
--disable piwiz,bluetooth \
--regen-ssh-host-keys \
--user pi --password-pi mypassword \
--ssh service \
--plugin apps:"apps=figlet hdparm mc nmap screen stress|name=apps" \
--plugin apps:"apps=passwordsafe|name=xapps" \
--aptcache 192.168.0.118 \
--apt-dist-upgrade \
--swap 1024 \
--csrc $CSRC \
--plugin pl-plugin \
--lxde-config lxterminal:/home/pi/Documents/sdm/myfiles/lxterminal.conf --plugin graphics \
--bootset boot_splash=1,boot_behaviour=B1 \
$IMGDIR/bullseye64.img
# --plugin runatboot:"user=pi|sudoswitches=-H|script=/home/pi/bin/refresh_pi3a|output=/home/pi/runatboot_refresh_pi3a.log" \
# --host pi3a83 \
# --bootadd temp_soft_limit:70 \

time sudo sdm \
--burn /dev/sdc \
--expand-root \
--csrc $CSRC \
--plugin pl-plugin \
--host pi3a83b \
--bootadd temp_soft_limit:70 \
$IMGDIR/bullseye64.img

```

gitbls, 2023-08-21:
Yep, it's busted for burn. Apparently nobody us using it there.  🤔

 Will get a fix out tomorrow.

gitbls, 2023-08-21:
Fixed in V8.5. Support added for `--bootadd` on burn.

tinker2much, 2023-08-21:
I tried it and it worked fine. Thank you!

gitbls, 2023-08-21:
👍 Thanks for the quick test turnaround!

closed 2023-08-21 by gitbls