gitbls/sdm #89: Error expanding root fs on --burn

damoclark, 2023-08-21:
hi @benn,

Great project mate.  Lots of work has gone into this and it works really well.  

I was tinkering with sdm to generate standardised images for different types of projects, so rather than have a custom image for each project type, I am configuring defaults for everything in standardised images, and then more specific customisations while burning.  

I came across an error when specifying the `--expand-root` option to `--burn`.  The full command was:

```bash
sudo sdm --burn /dev/mmcblk2 \
        ~/Downloads/2023-05-03-raspios-bullseye-arm64.img \
        --expand-root \
        --restart \
        --bootset serial=0,boot_wait=1,camera=0,i2c=0,onewire=1,blanking=0,overscan=0,boot_behaviour=B4 \
        --regen-ssh-host-keys \
        --hostname xxx
```

The error was:

```
cat: /sys/block/mmcblk2/mmcblk22/size: No such file or directory
```

The path should be: `/sys/block/mmcblk2/mmcblk2p2/size` to denote the second partition on the mmcblk2 device.  Full log output below for context.  The partition table for the root partition does get expanded fully.  I checked using fdisk, and the last sector matches the size of the 16G card.  But the ext4 filesystem doesn't get expanded to fill the partition, and retains its original size from the image file.

Just noticed another error too where the need for firstboot is confused.  See following full log output.

```
* Burn '/home/clarkd/Downloads/2023-05-03-raspios-bullseye-arm64.img' to '/dev/mmcblk2'...
dd if=/home/clarkd/Downloads/2023-05-03-raspios-bullseye-arm64.img of=/dev/mmcblk2 status=progress bs=16M iflag=direct
4622123008 bytes (4.6 GB, 4.3 GiB) copied, 250 s, 18.5 MB/s
275+1 records in
275+1 records out
4622123008 bytes (4.6 GB, 4.3 GiB) copied, 348.221 s, 13.3 MB/s
cat: /sys/block/mmcblk2/mmcblk22/size: No such file or directory
> Expand Root: Expand partition from 0.0GB (0.0GiB) to 10.5GB (11.3GiB)
> Set new disk ID '???' on '/dev/mmcblk2'
Disk identifier changed from 0x??? to 0x???.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID ??? set successfully with sfdisk
mount: /dev/mmcblk2p2 mounted on /mnt/sdm.
mount: /dev/mmcblk2p1 mounted on /mnt/sdm/boot.
> Set new Disk ID '???' in /etc/fstab and /boot/cmdline.txt
> Flush accumulated burn log messages
% sdm will try to use systemd-nspawn on this 'x86-64' system
> Disable unneeded RasPiOS firstboot service in cmdline.txt; both --expand-root and 
  --regen-ssh-host-keys set
> Disable regenerate_ssh_host_keys service; First Boot will run it instead
Failed to disable unit: Unit file regenerate_ssh_host_keys.service does not exist.
```

The `--customize` command used to generate the image follows.

```bash
sudo sdm --customize \
        ~/Downloads/2023-05-03-raspios-bullseye-arm64.img \
        --l10n \
        --bootset serial=0,boot_wait=1,camera=0,i2c=0,onewire=1,blanking=0,overscan=0,boot_behaviour=B4 \
        --user clarkd \
        --disable piwiz \
        --showapt \
        --ssh service \
        --extend --xmb 200
```

gitbls, 2023-08-21:
Thanks!

Can you tell me about the SD card reader you're using? I obviously don't do much testing with anything other than a USB card reader that identifies as /dev/sdX. I imagine (hope!) that using the built-in SD Card reader on the Pi would elicit the same error you're seeing so i can easily repro it.

damoclark, 2023-08-21:
> Can you tell me about the SD card reader you're using? I obviously don't do much testing with anything other than a USB card reader that identifies as /dev/sdX. I imagine (hope!) that using the built-in SD Card reader on the Pi would elicit the same error you're seeing so i can easily repro it.

Sure.  The computer I am using is a Seeed Studio [Odyssey](https://www.seeedstudio.com/ODYSSEY-X86J4105864-p-4447.html).  It has a built in 64GB eMMC for OS, and it also has an integrated Micro SD card Socket, which I am using to write cards.  The card socket uses the device file  `/dev/mmcblk2`.

damoclark, 2023-08-21:
I have the following additional info about the controller for the SD card reader from `lspci -v`.

```
00:1c.0 SD Host controller: Intel Corporation Celeron/Pentium Silver Processor SDA Standard Compliant SD Host Controller (rev 03) (prog-if 01)
        DeviceName: Onboard - Other
        Subsystem: Intel Corporation Celeron/Pentium Silver Processor SDA Standard Compliant SD Host Controller
        Flags: bus master, fast devsel, latency 0, IRQ 39
        Memory at a141e000 (64-bit, non-prefetchable) [size=4K]
        Memory at a141d000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: sdhci-pci
        Kernel modules: sdhci_pci

00:1e.0 SD Host controller: Intel Corporation Device 31d0 (rev 03) (prog-if 01)
        DeviceName: Onboard - Other
        Subsystem: Intel Corporation Device 7270
        Flags: bus master, fast devsel, latency 0, IRQ 42
        Memory at a141c000 (64-bit, non-prefetchable) [size=4K]
        Memory at a141b000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: sdhci-pci
        Kernel modules: sdhci_pci

```

gitbls, 2023-08-21:
Thx. sdm doesn't care about hardware device info (thankfully!). It was all about the device _name_ (mmcblkN), where partition names are **p**n rather than _n_. Anyhow, thanks for finding, fixed in V8.5

damoclark, 2023-08-21:
Wow, that was fast.  Thanks for resolving this one.  So can I just re-run the EZsdmInstaller script again to update to 8.5?

gitbls, 2023-08-21:
> Wow, that was fast. Thanks for resolving this one. So can I just re-run the EZsdmInstaller script again to update to 8.5?

Yep. It's EZ 🤣

gitbls, 2023-08-22:
By the way, on this error
```
> Disable regenerate_ssh_host_keys service; First Boot will run it instead
Failed to disable unit: Unit file regenerate_ssh_host_keys.service does not exist.
```
I don't see that here. 

When I look at an untouched Bullseye desktop IMG, the service regenerate_ssh_host_keys.service exists, which has me wondering about previous modifications to the IMG you're working with.

Can you replicate the error with a freshly downloaded IMG?

damoclark, 2023-08-22:
> Yep. It's EZ 🤣

Marketing and programming talents. 🤣

> Can you replicate the error with a freshly downloaded IMG?

I'll give it a try later today, and let you know.

gitbls, 2023-08-22:
I ALWAYS start with a fresh IMG and am unable to replicate it.

damoclark, 2023-08-27:
hi @gitbls 

I had hoped to get to this on the weekend but it wasn't to be.  I do wonder whether the issue was a side effect of the device file problem.

damoclark, 2023-08-27:
I tested v8.6 using --burn on a fresh image and there was no error. 

Just testing again on a customised image as originally reported to confirm this is resolved as well.

damoclark, 2023-08-27:
Yep, all fixed, although another error came up. I took a quick look and can see the issue - I'll submit a separate issue - its a quick fix.

Thanks again for all your hard work on this project.  You should consider seeing if you can get some sponsorship from the raspberry pi foundation to help support your work.

closed 2023-08-27 by damoclark

gitbls, 2023-08-28:
Thanks for following up on this. 

As far as RPF support, the support ... interesting idea. Will think about it.