gitbls/sdm #90: Using the copyfile plugin multiple times copies only the last file

nzottmann, 2023-08-23:
I want to copy 3 files in different target directories and included the new copyfile plugin 3 times. Only the last files ends up in the image. If I delete copyfile call 2 and 3, leaving only the first, this file ends up in the image, so my call seems to be correct.

The logs show all three calls of copyfiles with the different files `.bash_profile`, `.xinitrc` and `postcardscanner.service`, but it says that it copies always `postcardscanner.service`

```
> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments: 
  'from=./files/.bash_profile|to=/home/pi/.bash_profile|mkdirif=yes'
* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy 'postcardscanner.service' from /etc/sdm/assets/copyfile to 
  '/etc/systemd/system/postcardscanner.service'
* Plugin copyfile: Complete Phase 1

> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments: 
  'from=./files/.xinitrc|to=/home/pi/.xinitrc|mkdirif=yes'
* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy 'postcardscanner.service' from /etc/sdm/assets/copyfile to 
  '/etc/systemd/system/postcardscanner.service'
* Plugin copyfile: Complete Phase 1

> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments: 
  'from=./files/postcardscanner.service|to=/etc/systemd/system/postcardscanner.service|mkdirif=yes
  '
* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy 'postcardscanner.service' from /etc/sdm/assets/copyfile to 
  '/etc/systemd/system/postcardscanner.service'
* Plugin copyfile: Complete Phase 1
* ```

The docs say
> Multiple --plugin switches can be used on the command line. This includes specifying the same plugin multiple times

The apps plugin is explicitly mentioned as supported here, but copyfile is not mentioned. Is this a bug or does copyfile just not support multiple calls?

gitbls, 2023-08-23:
It WAS working, but then I "improved" it, broke it, and clearly failed to retest multiple invocations of the plugin.

Will fix later this morning USPT.

gitbls, 2023-08-23:
Fixed in V8.6. You can simply re-run EZsdmInstaller to pick it up.

nzottmann, 2023-08-24:
In my first try it didn't work at all now, but I'm still figuring out if this is my fault or not. Will report back after checking.

gitbls, 2023-08-24:
> In my first try it didn't work at all now, but I'm still figuring out if this is my fault or not. Will report back after checking.

If you can't sort it out, post the sdm command you're working with I can have a look at it.

nzottmann, 2023-08-24:
Thank you, I tested more and it seems to be a problem with the changes to the copyfile plugin.

This is my sdm command:  
`sudo sdm --customize --plugin copyfile:"from=$PWD/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=root:root|chmod=755" 2023-05-03-raspios-bullseye-armhf-lite.img`
The file `050-test.sh` and the image are in my working dir.

Using the old copyfile https://github.com/gitbls/sdm/blob/fcca7014105b3d6f801bbb1f89ab04674523201c/plugins/copyfile, the file ends up in the image. Using the new copyfile https://github.com/gitbls/sdm/blob/e24a533f6d693b73c83cd9c2bf37a1e4d9cae3bf/plugins/copyfile, the file is missing in the image. You can also see a difference in the relevant sections of the log output, with the new version, the lines where it actually copies the file are missing:

Old copyfile, working:
```
> Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments:
  'from=/workingdir/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=
  root:root|chmod=755'
* Plugin copyfile: Start Phase 0
> Plugin copyfile: Keys/values found:
   from: /workingdir/050-test.sh
   to: /usr/local/sdm/1piboot/
   mkdirif: yes
   chown: root:root
   chmod: 755
> Plugin copyfile: Add '/workingdir/050-test.sh' to files to copy
* Plugin copyfile: Complete Phase 0

* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy '050-test.sh' from /etc/sdm/assets/copyfile to '/usr/local/sdm/1piboot/'
* Plugin copyfile: Complete Phase 1

> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase post-install with arguments:
  'from=/workingdir/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=
  root:root|chmod=755'
* Plugin copyfile: Start Phase post-install
* Plugin copyfile: Complete Phase post-install
```

New copyfile, not working:
```
> Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments:
  'from=/workingdir/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=
  root:root|chmod=755'
* Plugin copyfile: Start Phase 0
> Plugin copyfile: Keys/values found:
   from: /workingdir/050-test.sh
   to: /usr/local/sdm/1piboot/
   mkdirif: yes
   chown: root:root
   chmod: 755

> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments:
  'from=/workingdir/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=
  root:root|chmod=755'
* Plugin copyfile: Start Phase 1
* Plugin copyfile: Complete Phase 1

> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase post-install with arguments:
  'from=/workingdir/050-test.sh|to=/usr/local/sdm/1piboot/|mkdirif=yes|chown=
  root:root|chmod=755'
* Plugin copyfile: Start Phase post-install
* Plugin copyfile: Complete Phase post-install
```

gitbls, 2023-08-24:
It appears that I duffed an edit. There is a line with a "naked" `exit` on it in the plugin.  Delete that line and append `&& exit` to the line above it (logtoboth Nothing to Do). 

Not sure how I managed that 😳, but it will be fixed in the next update

nzottmann, 2023-08-24:
Oops 😀 

After moving the `exit`, it works. Thank you for the quick help!

gitbls, 2023-09-23:
This is corrected in V9.1.1. Please test, verify, and close this issue.

closed 2023-10-23 by gitbls

nzottmann, 2023-10-24:
I can confirm this is fixed in V9.5. Thank you!