gitbls/sdm #91: couple of things I am puzziling through

AlanEbell, 2023-08-23:
Here is a the command I am running:
sudo sdm --customize --network:"netman=nm|wifissid=Netgear96|wifipassword=xxxx|wificountry=US|noipv6" --L10n --restart --disable piwiz --regen-ssh-host-keys --user trg --password-user xxxx --password-same y --autologin --xmb 2048 --plugin copyfile:"filelist=/home/alan/Documents/raspios/filestore/trg/filelist.txt" 2023-05-03-raspios-bullseye-armhf-lite.img

two things are happening 
1) the wlan0 and dhcpcd do not seem to be setup so when I first log in the pi zero has not network access.
2) the other is autologin does not seem to be working and there is also a user labeled pi on the system after it boots up as well as the trg user.

I am pretty sure this has got to be something I am misunderstanding about the sdm so I added the command above in hopes you can educate me.

gitbls, 2023-08-23:
In your command line you have `netman=nm`. This means use Network Manager instead of dhcpcd. So, dhcpcd will not be configured. Since you're using a Bullseye V11 (I presume) IMG, replace `netman=nm` with `netman=dhcpcd` and see what happens. This should bring your network online assuming your wifi ssid and password are correct.

As far as the user `pi` being there, that is correct. `--user` creates a new user. If you want to replace the user pi with user trg, you can use `--rename-pi`. Note that the user pi doesn't have a password, so it can't be logged into.

Personally, I create my new user with a userid and gid that I use across all my Linux systems so that files on NFS network shares have correct ownership. (This is a very poor man's network directory!)

Every one of my systems has a pi user on it, but with no password, so the account can't be logged into.

gitbls, 2023-08-24:
Just realized that I neglected to address the autologin question. 

I'll have a look at that tomorrow.

gitbls, 2023-08-24:
> Just realized that I neglected to address the autologin question.
> 
> I'll have a look at that tomorrow.

Autologin was not handled correctly in this case. Will fix in the next update.

gitbls, 2023-09-23:
V9.1.1 is released and the autologin issue with CLI is corrected. The NetworkManager support has been improved, but I will admit that most of the testing I did on it was with bookworm. That said, I can't think of a reason that it won't work for you.

HOWEVER, users are handled differently now. For details please review [https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#user](https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#user)

Please test and verify that your issues are corrected on V9.1.1, close this issue if they are, or update with V9.1.1-based details.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls