gitbls/sdm #94: Feature request: Install a specific version of sdm

nzottmann, 2023-08-24:
First of all: Thank you for this very nice piece of software! The possibilities and documentation are awesome.

I would like to run sdm in a GitHub Action, to have a customized pi image build automatically. This works already fine.

sdm is installed with the `EZsdmInstaller` script, which downloads always the most recent commit in the master branch:
https://github.com/gitbls/sdm/blob/210897b5eb1e81a404338a81833d07c253682084/EZsdmInstaller#L60

This is a problem for me, because breaking changes like in sdm V8.x or new introduced bugs can break my build. This would not have to happen, as the build would continue to work fine using the old sdm version. To have a reliable auto build, I need to have the control when to update sdm, which means I need to be able to install a specific version of sdm. Another use case is testing something with a different version of sdm, for example to check for bugs.

Without any changes to sdm, I have this workaround:
- Download `EZsdmInstaller` at a specific commit
- Replace `master` in the download link with the id of the specific commit using for example sed
- Install

So the feature request is a clean way for installing a specific version. There are multiple ways to achieve this. For example, the installer script could have a parameter to specify a commit id replacing `master`. Adding tags to the release commits would even allow to specify a version instead of a commit id.

Please see this feature request just as something to start thinking, I don't see a high priority as I listed an easy workaround.

Thank you 😃

gitbls, 2023-08-24:
Thanks for your thoughts on this. I recognize the need for better version control, so will add this to my TODO list.

As an aside, while I have used other version control systems in my deep dark past, my git and github knowledge is, er, spare. So, before implementing this, I'm going to have to learn more about git/github. Not a big deal, but features are always more fun 🤣

nzottmann, 2023-08-24:
Thank you for thinking about this 😃 But no pressure, the workaround is easy.

sdm's features are amazing, thank you for the great work!

gitbls, 2023-09-14:
A quick update: I believe I've sorted out how to do this and I'm planning to use it starting with the v9.0 release that's coming soon 🤞

gitbls, 2023-09-23:
the `master` branch (aka the default) is now V9.1.1, and V8.6 is also available [here](https://github.com/gitbls/sdm/releases/tag/V8.6). There is some information about installing from the non-default branch [here](https://github.com/gitbls/sdm/blob/master/Docs/Detailed-Installation-Guide.md).

Please test, verify, and close this issue, or (of course) update this if you have problems.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls

nzottmann, 2023-10-24:
Sorry for the late reply, I did not have time too test this until now.

Thank you for implementing this feature, this is exactly what I was looking for. Very nice!