gitbls/sdm #95: Enabling camera using bootset does not work in current pi images

nzottmann, 2023-08-24:
`--bootset` has the option `camera`, but this does not enable the camera in a current image, probably because they changed the way this is done in `config.txt`.

In `2023-05-03-raspios-bullseye-armhf-lite`, there is now the option `camera_auto_detect=0`, which needs to be changed to `1`.

The docs https://www.raspberrypi.com/documentation/computers/config_txt.html#camera_auto_detect say that this option is on by default, but at least for the mentioned lite image this is not the case, it is set to 0.

Of course there are several possible workarounds for sdm like using a first boot script, but I had to debug why my camera wasn't working at first although I specified `--bootset camera=1`.

gitbls, 2023-08-25:
Yea, it's been quite a while since I've looked at these. On the list to bring it up to date, probably will not be in the immediate next release for a complete overhaul, but will take a look at camera* since you're interested in it.

nzottmann, 2023-08-25:
Thank you. Currently I am doing this using `sed -i "s/camera_auto_detect=0/camera_auto_detect=1/" /boot/config.txt` in a custom firstboot script, which works fine.

But I am not sure why exactly this is necessary, because when checking the image with `sudo sdm --explore` before or after customization, `camera_auto_detect` is set to `1`. But after booting on a pi, it is set to `0` when not using my firstboot script. I don't know where this switch happens.

gitbls, 2023-08-25:
Oh, wait...my bad. Things in config.txt will all be automatically "supported" since the new plugin won't check validity. "Supported" meaning whatever you provide will get into config.txt, syntax unchecked. So, `xyzzy=10` will work also 🤯

I haven't done anything with Pi cameras for years, so can't offer any help for `camera_auto_detect`. Maybe post a question in the Raspberry Pi forums to get more eyes on the question?

nzottmann, 2023-08-25:
I tried `--bootset camera_auto_detect=1` but I says
```
? Unrecognized --bootset key 'camera_auto_detect'
```

Is there another plugin for populating `config.txt`?

gitbls, 2023-08-25:
> I tried `--bootset camera_auto_detect=1` but I says
> 
> ```
> ? Unrecognized --bootset key 'camera_auto_detect'
> ```
> 
> Is there another plugin for populating `config.txt`?

Oh, sorry! There are a few new plugins coming in the next release that will replace `--bootset`, `--bootadd`, and `--bootconfig`. 

I'm hoping to get it out early next week, but it may be another week due to other commitments later in the week.

nzottmann, 2023-08-25:
Ah ok, just thought I overlooked something. Take your time, I have a workaround, so no quick action need here on my side.

Thank you!

gitbls, 2023-08-26:
See https://github.com/gitbls/sdm/discussions/98 for a discussion of the new plugins. You'll be able to add camera_auto_detect using the `bootconfig` plugin

gitbls, 2023-09-23:
V9.1.1 has been released with the `bootconfig` plugin. It does no validity checking, so you should be able to add camera_auto_detect to config.txt. Of course, since there's no validity checking you can also misspell settings and have fun sorting out what you did 😵

Please test and verify that this is corrected and close this issue.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls

nzottmann, 2023-10-24:
Using V9.5, setting `camera_auto_detect=1` in config.txt works fine using the `raspiconfig` plugin. Thank you!