gitbls/sdm #100: Error in /usr/local/sdm/sdm-cmdsubs when using --burn option

damoclark, 2023-08-27:
Received the following error:

```
% sdm will try to use systemd-nspawn on this 'x86-64' system
  If this does not work please post a GitHub issue
/usr/local/sdm/sdm-cmdsubs: line 215: [: !=: unary operator expected
> Disable unneeded RasPiOS firstboot service in cmdline.txt; both --expand-root and 
  --regen-ssh-host-keys set
```
Took a quick look at line 215, and its just missing double quotes around the variable `$bbootadd` - presumably the variable was an empty string.

```bash
        [ $bbootadd != "" ] && bootadd="$bbootadd"
```

gitbls, 2023-08-28:
Yea, saw this while I was in there actually removing this line of code because of https://github.com/gitbls/sdm/discussions/98

The next version replaces several command line switches with plugins, and this exact line of code is one of those no longer needed.

gitbls, 2023-09-23:
This is fixed in V9.1.1. Please test, verify, and close this issue.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls