gitbls/sdm #99: Customise on fresh arm64 image error when no local-plugins used

damoclark, 2023-08-27:
hi @gitbls 

Just testing on a fresh image after re-running EZ installer again (over top of my previous version).  Now with version 8.6, I got the following error.  It looks harmless, but thought I'd report it.  I checked the `/mnt/sdm/usr/local/sdm/local-plugins` directory when sdm dropped me to a root shell and the directory does exist and it is empty.

```
* Mount IMG '/home/user/Downloads/2023-05-03-raspios-bullseye-arm64.img.test'
mount: /dev/loop11p2 mounted on /mnt/sdm.
mount: /dev/loop11p1 mounted on /mnt/sdm/boot.
* Unmount IMG for --extend
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
* Extend IMG by 200MB...
200+0 records in
200+0 records out
209715200 bytes (210 MB, 200 MiB) copied, 0.283942 s, 739 MB/s
* Resize partition 2 of '/home/user/Downloads/2023-05-03-raspios-bullseye-arm64.img.test
* Remount IMG to resize the file system
* Mount IMG '/home/user/Downloads/2023-05-03-raspios-bullseye-arm64.img.test'
mount: /dev/loop11p2 mounted on /mnt/sdm.
mount: /dev/loop11p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore on-line resizing message)
resize2fs 1.46.5 (30-Dec-2021)
Filesystem at /dev/loop11p2 is mounted on /mnt/sdm; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop11p2 is now 1061858 (4k) blocks long.

chmod: cannot access '/mnt/sdm//usr/local/sdm/local-plugins/*': No such file or directory
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize 
  /home/user/Downloads/2023-05-03-raspios-bullseye-arm64.img.test --l10n --restart --bootset 
  serial=0,boot_wait=1,camera=0,i2c=0,onewire=1,blanking=0,overscan=0,boot_behaviour=B4 
  --regen-ssh-host-keys --user user --disable piwiz --showapt --ssh service --extend --xmb 200
* Host Information
   Hostname: hostname
   uname: Linux hostname 5.15.0-78-generic #85-Ubuntu SMP Fri Jul 7 15:25:09 UTC 2023 x86_64 
  x86_64 x86_64 GNU/Linux
   os-release Name: "Ubuntu"
   os-release Version: "22.04.3 LTS (Jammy Jellyfish)"
   sdm Version: V8.6
* IMG Information
   Date: 2023-05-03
   RasPiOS Version: 11
   os-release Version: "11 (bullseye)"
% sdm will try to use systemd-nspawn on this 'x86-64' system
  If this does not work please post a GitHub issue
> IMG '/home/user/Downloads/2023-05-03-raspios-bullseye-arm64.img.test' has 656616 1K-blocks free 
> Copy sdm to /usr/local/sdm in the IMG

```

gitbls, 2023-08-28:
Huh. Will have a look at that.

gitbls, 2023-08-29:
The code ALWAYS creates the directory before the `chmod`, so I'm completely baffled as to how it could fail.

Does it reproduce every time you do a customize?

damoclark, 2023-08-30:
The problem is that the directory is empty.  The asterisk wildcard when evaluated by the shell, returns no file arguments so Chmod is executed with out any file paths, which generates the error.  I’ve tested it and that is the error returned.

you might need to do recursive perms on the path, or test if files exist before performing the chmod.

gitbls, 2023-09-23:
This is fixed in V9.1.1. Please test, verify, and close this issue.

damoclark, 2023-09-27:
Hey @gitbls 

I just went to test V9.1.1 but I can't see such a version in your releases, or tags in the repository.

gitbls, 2023-09-27:
> Hey @gitbls
> 
> I just went to test V9.1.1 but I can't see such a version in your releases, or tags in the repository.

I'm still adapting my workflow to using releases and branches. Sorry for the confusion.

The 'master' branch is currently V9.1.1, so install from that.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls