gitbls/sdm #101: Mount command description

Mark-81, 2023-08-29:
[Here](https://github.com/gitbls/sdm/blob/master/Docs/Command-Details.md) I read:

> sudo sdm --mount raspios-image.img
> 
> OR
> 
> sudo sdm --mount /dev/sdX
> 
> Mounts the IMG file (first example) or SD Card (second example) onto the running system. This enables you to manually and easily copy files from the running RasPiOS system into the IMG.

I'm not sure to understand.
This command is intended to be used on the development machine (where the IMG file is located) and it will hardly be a RasPiOS system. It can, but usually it would be any common distro like Ubuntu, Debian, Arch... RasPiOS is the target os in my opinion.

Hence why you assume the running system is the RasPiOS?

gitbls, 2023-09-23:
This is corrected in V9.1.1. Please verify and close this issue.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls