gitbls/sdm #103: --extend does nothing

Mark-81, 2023-08-29:
I downloaded the Bullseye xz file from the official website and extract to my hdd.
Then I tried to extend the partition:

```
$ sudo sdm --extend  ~/Scaricati/2023-05-03-raspios-bullseye-arm64.img
* Extend IMG by 2048MB...
2048+0 Record in
2048+0 registra
2147483648 byte ( 2,1 GB, 2,0 GiB) copiato, 0,557759 s, 3,9 GB/s
* Resize partition 2 of '/home/mark/Scaricati/2023-05-03-raspios-bullseye-arm64.img
* Remount IMG to resize the file system
* Mount IMG '/home/mark/Scaricati/2023-05-03-raspios-bullseye-arm64.img'
mount: /dev/loop24p2 mounted on /mnt/sdm.
mount: /dev/loop24p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore on-line resizing message)
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 1010688 (4k) blocks long.  Nothing to do!

umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
```

As the output says, it does not resize anything.
I also tried to add the same command along with `--customize` but the behavior is the same.

Am I doing it wrong?

gitbls, 2023-08-29:
This problem does not repro on hosts running Debian Bullseye or RasPiOS.

Please provide details on your host system, specifically: 
* Is the host system running on bare metal or in WSL?
* The contents of /etc/os-release
* The output of `uname -a`

Thanks

Mark-81, 2023-08-30:
Ubuntu 23.04 running on bare metal.

```
$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 23.04"
NAME="Ubuntu"
VERSION_ID="23.04"
VERSION="23.04 (Lunar Lobster)"
VERSION_CODENAME=lunar
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=lunar
LOGO=ubuntu-logo
``` 
```
$ uname -a
Linux femtolab 6.2.0-27-generic #28-Ubuntu SMP PREEMPT_DYNAMIC Wed Jul 12 22:39:51 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

gitbls, 2023-08-30:
I won't be able to set up a test system with this for a couple of weeks. If you're game, here's something you could try:

sudo edit (with your favorite editor) /usr/local/sdm/sdm-cmdsubs 

Find the following 4 lines:
```
    parted -m /dev/$dev u s resizepart 2 $newsize
    errifrc $? "? parted resizepart failed with status"
    #echo ",+" | sfdisk -N 2 $fulldevname
    #errifrc $? "? sfdisk expand last partition failed with status"
```
Comment out the first 2 lines, and uncomment the last 2, so it looks like:
```
    #parted -m /dev/$dev u s resizepart 2 $newsize
    #errifrc $? "? parted resizepart failed with status"
    echo ",+" | sfdisk -N 2 $fulldevname
    errifrc $? "? sfdisk expand last partition failed with status"
```
Save and exit the file, try the `--extend` again, and let me know the result.

gitbls, 2023-08-30:
As I mentioned, I won't be able to check this out on x64 for while, but I did throw Ubuntu 23.04 onto a Pi4. Doing the extend works fine:
```
p83/ssd/work$ sudo sdm --extend bullseye-lite.img 
* Extend IMG by 2048MB...
2019557376 bytes (2.0 GB, 1.9 GiB) copied, 11 s, 184 MB/s
2048+0 records in
2048+0 records out
2147483648 bytes (2.1 GB, 2.0 GiB) copied, 23.2939 s, 92.2 MB/s
* Resize partition 2 of 'bullseye-lite.img
* Remount IMG to resize the file system
* Mount IMG 'bullseye-lite.img'
mount: /dev/loop8p2 mounted on /mnt/sdm.
mount: /dev/loop8p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore on-line resizing message)
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/loop8p2 is mounted on /mnt/sdm; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop8p2 is now 1626067 (4k) blocks long.

umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
```
I 'm hard-pressed to believe that this is an x64/Arm difference, but stranger things have happened, I guess. 

I'll await the results of the test I asked you to do above.

Mark-81, 2023-08-31:
I tried what you suggested but unfortunately the output is the same:

```
$ sudo sdm --extend 2023-05-03-raspios-bullseye-arm64.img
* Extend IMG by 2048MB...
2048+0 Record in
2048+0 registra
2147483648 byte ( 2,1 GB, 2,0 GiB) copiato, 0,556056 s, 3,9 GB/s
* Resize partition 2 of '2023-05-03-raspios-bullseye-arm64.img
* Remount IMG to resize the file system
* Mount IMG '2023-05-03-raspios-bullseye-arm64.img'
mount: /dev/loop7p2 mounted on /mnt/sdm.
mount: /dev/loop7p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore on-line resizing message)
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 1010688 (4k) blocks long.  Nothing to do!

umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
```

But as you can see it actually expanded the partition but not the filesystem:

[image]

"Spazio libero" means empty space in Italian.

gitbls, 2023-08-31:
OK, thx. Need to do more research on this.

Mark-81, 2023-08-31:
Yeah, now it works. I tried without the 2nd parameter (+512 MB) and extending it again of 2048 MB:

```
$ sudo ./script.sh 2023-05-03-raspios-bullseye-arm64.img 2048

sfdisk before extend and resize
Disk 2023-05-03-raspios-bullseye-arm64.img: 4,61 GiB, 4949278720 bytes, 9666560 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3e247b30

Dispositivo                            Avvio  Start    Fine Settori  Size Id Tipo
2023-05-03-raspios-bullseye-arm64.img1         8192  532479  524288  256M  c W95 FAT32 (LBA)
2023-05-03-raspios-bullseye-arm64.img2       532480 9619140 9086661  4,3G 83 Linux

parted list before extend and resize
Modello:  (file)
Disco /home/mark/dev/performer/2023-05-03-raspios-bullseye-arm64.img: 4949MB
Dimensione del settore (logica/fisica): 512B/512B
Tabella delle partizioni: msdos
Flag del disco: 

Numero  Inizio  Fine    Dimensione  Tipo     File system  Flag
 1      4194kB  273MB   268MB       primary  fat32        lba
 2      273MB   4925MB  4652MB      primary  ext4

** Do extend and resize
* Extend  by 2048MB...
2048+0 Record in
2048+0 registra
2147483648 byte ( 2,1 GB, 2,0 GiB) copiato, 0,552822 s, 3,9 GB/s
* Resize partition 2 of '2023-05-03-raspios-bullseye-arm64.img to 6700MB (end=6973MB)
* Remount 2023-05-03-raspios-bullseye-arm64.img to resize the file system
mount: /dev/loop25p2 mounted on /mnt.
mount: /dev/loop25p1 mounted on /mnt/boot.
* Resize the file system
% (Ignore on-line resizing message)
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/loop25p2 is mounted on /mnt; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop25p2 is now 1635832 (4k) blocks long.

sfdisk after extend and resize
Disk 2023-05-03-raspios-bullseye-arm64.img: 6,61 GiB, 7096762368 bytes, 13860864 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3e247b30

Dispositivo                            Avvio  Start     Fine  Settori  Size Id Tipo
2023-05-03-raspios-bullseye-arm64.img1         8192   532479   524288  256M  c W95 FAT32 (LBA)
2023-05-03-raspios-bullseye-arm64.img2       532480 13619140 13086661  6,2G 83 Linux
parted list after extend and resize
Modello:  (file)
Disco /home/mark/dev/performer/2023-05-03-raspios-bullseye-arm64.img: 7097MB
Dimensione del settore (logica/fisica): 512B/512B
Tabella delle partizioni: msdos
Flag del disco: 

Numero  Inizio  Fine    Dimensione  Tipo     File system  Flag
 1      4194kB  273MB   268MB       primary  fat32        lba
 2      273MB   6973MB  6700MB      primary  ext4
```
[image]

It leaves some spare space at the end of the image, but it should not harm. But the root partition is now extended as expected! Good job!

gitbls, 2023-08-31:
Thanks for confirming! I knew that it wasn't consuming all the free space, but wanted to check this on your system.

I'll be tuning this up and will include it in the next release of sdm, target release the week of 9/11.

gitbls, 2023-08-31:
I updated the script above so that it correctly consumes all available space. Please give it a try. Thx

Mark-81, 2023-08-31:
Yep, I confirm!

[image]

gitbls, 2023-08-31:
Thx for the quick test turnaround!

gitbls, 2023-09-23:
This is corrected in sdm V9.1.1. Please test, verify, and close this issue or if it fails, update this with additional information based on sdm v9.1.1.

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls