gitbls/sdm #104: --fstab not being run on burn

tinker2much, 2023-09-03:
And I know it doesn't list --fstab as something that occurs in --burn, but I sure wish it did - I have specific, not general, fstab lines that I would like to do only when burning specific pi's.

gitbls, 2023-09-04:
As I mentioned in https://github.com/gitbls/sdm/discussions/98, fstab will be included in the `system` plugin with the upcoming release. I didn't go into detail, but here's the scoop:

When `fstab` is provided as an argument to the `system` plugin, it will accept a comma-separated list of files on the host system, each of which will be appended to /etc/fstab per below.

Each file will be copied into the IMG, and a deferred action will be added to /etc/sdm/auto-1piboot.conf for each file. auto-1piboot.conf is processed as part of the first system boot, and each of the `fstab` files will be appended to /etc/fstab as part of the first boot in the order in which they are specified.

The reason I do it this way is that other actions (fstab dependencies) may also be deferred until the first system boot . If the dependencies aren't in place, that could cause fstab to fail, thus causing the first boot to fail.

Deferring the fstab modifications to the first system boot reduces the likelihood that will happen, but of course that's ultimately up to you.

Since fstab will be part of the `system` plugin, you can use it during customization or burning.

Hope this will provide you with the fstab flexibility you want. Other arguments to the `system` plugin will also accept multiple files. This of course will all be documented when the release is published sometime during the week of 9/11.

tinker2much, 2023-09-04:
I missed that part, sorry. (sheepish face...) That will completely take care of fstab for me.

gitbls, 2023-09-04:
No worries! 20-some command switches are changing all at once.  JFYI `exports` will work the same way, except the append won't be deferred to the first system boot. modprobe, sysctl, and all the cron-* arguments also work the same, but copy new files into the prescribed target directory, rather than appending to a single file. 

I'm looking forward to the release, b/c it will clean up a LOT of things I wasn't totally happy with. But there's testing and documentation yet to be done.

gitbls, 2023-09-15:
Would you please email me at gitbls @ outlook.com ? I'd like to enlist your assistance with something, if possible.

Thx

gitbls, 2023-09-23:
This is fully corrected in V9.1.1 which is now released. The `system` plugin can be applied during a burn and supports your use case. Please verify and close. Thx!

closed 2023-10-23 by gitbls

1stcall, 2024-08-20:
My customise command line is :-
sudo sdm --customize 2024-07-04-raspios-bookworm-arm64-lite.img \
    --extend --xmb 1024 \
    --showapt \
    --apt-dist-upgrade \
    --plugin-debug \
    --plugin @./customize.plugin \
    --restart \
    --batch

customize.plugin contains :-
user:"deluser=pi"
user:"addgroup=docker,2000"
user:"adduser=carl|uid=1000|password=letmein123|groupadd=docker"
apt-file
chrony
apps:"name=removal1|remove=fake-hwclock"
apps:"name=baseapps|apps=git,command-not-found,dnsutils,kpartx,binfmt-support,gdisk,parted,qemu-user-static,rsync,systemd-container,uuid,jq,usbutils,dnsutils,networkd-dispatcher,nmap,tmux,curl,wget,xz-utils,file,zram-tools,i2c-tools,qemu-utils,parted,nfs-kernel-server,python3-systemd,iptables,firewalld,fail2ban,btrfs-progs,python3-icmplib,sg3-utils,lsscsi"
disables:"piwiz|triggerhappy"
system:"name=1stcall|service-disable=apt-daily.timer,apt-daily-upgrade.timer,systemd-timesyncd|eeprom=beta"
quietness:"consoleblank|noquiet=keep|nosplash=keep|noplymouth"
copyfile:"from=/home/carl/dev/mysdm/override.conf|to=/lib/systemd/system/nfs-blkmap.service.d/|mkdirif"

My burn command line is :-
sudo sdm --burn /dev/mmcblk0 \
    --host rpicm4-1.1stcall.uk \
    --convert-root btrfs \
    --expand-root \
    --regen-ssh-host-keys \
    --gpt \
    --showapt \
    --plugin-debug \
    --plugin @./burn_rpicm4-1.plugin \
    2024-07-04-raspios-bookworm-arm64-lite.img

and burn_rpicm4-1.plugin contains :-
apt-addrepo:"repo=deb https://download.docker.com/linux/debian bookworm stable|gpgkey=/home/carl/dev/mysdm/docker.asc"
apps:"name=docker|apps=docker-ce,docker-ce-cli,containerd.io,docker-buildx-plugin,docker-compose-plugin"
#apps:"name=mate|apps=task-mate-desktop,mate-desktop-environment-extras"
#apps:"name=guiapps|apps=gparted,firefox,code"
bootconfig:"section=[cm4]|dtparam=i2c_vc=on|dtoverlay=i2c-rtc,pcf85063a,i2c_csi_dsi"
L10n:"host"
system:"name=myfstab|fstab=/home/carl/dev/mysdm/myfstab"
trim-enable:"disks=all"
imon

The resulting history file is :-

2024-08-20 00:41:59 * Start Configuration
2024-08-20 00:41:59 > Command Line: /usr/local/bin/sdm --customize 2024-07-04-raspios-bookworm-arm64-lite.img 
2024-08-20 00:41:59   --extend --xmb 1024 --showapt --apt-dist-upgrade --plugin-debug --plugin @./customize.plugin 
2024-08-20 00:41:59   --restart --batch
2024-08-20 00:41:59 * Host Information
2024-08-20 00:41:59    Hostname:  rescue.1stcall.uk
2024-08-20 00:41:59    Memory:    7996444 kB
2024-08-20 00:41:59    uname:     Linux rescue.1stcall.uk 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
2024-08-20 00:41:59    Raspberry Pi Model: Raspberry Pi Compute Module 4 Rev 1.0
2024-08-20 00:41:59    os-release Name: Debian GNU/Linux 12 (bookworm)
2024-08-20 00:41:59    Version: 12 (bookworm)
2024-08-20 00:41:59    sdm Version: V12.6
2024-08-20 00:41:59 * IMG Information
2024-08-20 00:41:59    Name: 2024-07-04-raspios-bookworm-arm64-lite.img
2024-08-20 00:41:59    Date: 2024-07-04
2024-08-20 00:41:59    RasPiOS Version: 12
2024-08-20 00:41:59    RasPiOS Architecture: 64-bit aarch64
2024-08-20 00:41:59    os-release Version: 12 (bookworm)
2024-08-20 00:41:59 * Plugins selected:
2024-08-20 00:41:59    * user
2024-08-20 00:41:59        Args: deluser=pi
2024-08-20 00:41:59    * user
[... 381 lines cut ...]
2024-08-20 00:49:51   'repo=deb'
2024-08-20 00:49:52 * Plugin apt-addrepo: Start Phase 1
2024-08-20 00:49:52 * Plugin apt-addrepo: Complete Phase 1
2024-08-20 00:49:52 > Run Plugins Phase 'post-install'
2024-08-20 00:49:52 > Run Plugin 'apt-addrepo' (/usr/local/sdm/plugins/apt-addrepo) Phase post-install with 
2024-08-20 00:49:52   arguments: 'repo=deb'
2024-08-20 00:49:52 * Plugin apt-addrepo: Start Phase post-install
2024-08-20 00:49:52 * Plugin apt-addrepo: Complete Phase post-install
2024-08-20 00:49:52 > Enable btrfs in initramfs
2024-08-20 00:50:22 * Burn Completed
2024-08-20 00:50:22 > Burn elapsed time: 00:03:02
2024-08-20 00:54:44 > sdm FirstBoot: Start sdm FirstBoot Service
2024-08-20 00:54:44 > sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
2024-08-20 00:54:44 > sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
2024-08-20 00:54:44 > sdm FirstBoot: Set WiFi Country to 'US'
2024-08-20 00:54:53 > sdm FirstBoot: System time synchronization achieved
2024-08-20 00:54:53 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/030-rpi-eeprom.sh'
2024-08-20 00:54:53 > sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/090-reset-quietness.sh'
2024-08-20 00:54:53 > sdm FirstBoot: Skip Custom sdm FirstBoot Scripts
2024-08-20 00:54:53 > sdm FirstBoot: Disable sdm-firstboot service
2024-08-20 00:54:54 > sdm FirstBoot: Regenerate SSH Host Keys
2024-08-20 00:55:00 > sdm FirstBoot: set boot behaviour to B1
2024-08-20 00:55:03 > sdm FirstBoot: Re-enable getty@tty1
2024-08-20 00:55:05 > sdm FirstBoot: System will restart in 20 seconds
2024-08-20 00:55:27 > sdm FirstBoot: System restarting now

gitbls, 2024-08-20:
> My customise command line is :- sudo sdm --customize 2024-07-04-raspios-bookworm-arm64-lite.img --extend --xmb 1024 --showapt --apt-dist-upgrade --plugin-debug --plugin @./customize.plugin --restart --batch

Thx for all that. Could you also do the following:
* On the burned output disk, what is the contents of /etc/sdm/auto-1piboot.conf?
* After booting the disk:
    * Identify the first boot (where sdm-firstboot service runs) and do `journalctl -b -N | grep fstab` Replace `-N` with the number to get the journal for that boot. On a freshly booted system this will be `-1`
 
With what you've provided (which all looks fine) plus the above, I'll have sufficient info to sort out what's happening. Thx

1stcall, 2024-08-20:
/etc/sdm/auto-1piboot.conf contains

# sdm added these settings from the command line (see /etc/sdm/history)

journalctl -b -1 | grep fstab yeilds no results.

journalctl -b -1 --unit sdm-firstboot.service :-

Aug 20 00:54:44 rpicm4-1.1stcall.uk systemd[1]: Starting sdm-firstboot.service - sdm FirstBoot System Configuration...
Aug 20 00:54:44 rpicm4-1.1stcall.uk systemd[1]: Started sdm-firstboot.service - sdm FirstBoot System Configuration.
Aug 20 00:54:44 rpicm4-1.1stcall.uk root[930]: sdm FirstBoot: Start sdm FirstBoot Service
Aug 20 00:54:44 rpicm4-1.1stcall.uk root[935]: sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
Aug 20 00:54:44 rpicm4-1.1stcall.uk root[949]: sdm FirstBoot: Set WiFi Country to 'US'
Aug 20 00:54:53 rpicm4-1.1stcall.uk root[1197]: sdm FirstBoot: System time synchronization achieved
Aug 20 00:54:53 rpicm4-1.1stcall.uk root[1203]: sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/030-rpi-eeprom.sh'
Aug 20 00:54:53 rpicm4-1.1stcall.uk root[1207]: sdm FirstBoot: Set rpi-eeprom to 'beta'
Aug 20 00:54:53 rpicm4-1.1stcall.uk root[1209]: sdm FirstBoot: Run Generated sdm FirstBoot Script '/etc/sdm/0piboot/090-reset-quietness.sh'
Aug 20 00:54:53 rpicm4-1.1stcall.uk root[1217]: sdm FirstBoot: Skip Custom sdm FirstBoot Scripts
Aug 20 00:54:54 rpicm4-1.1stcall.uk root[1242]: sdm FirstBoot: Regenerate SSH Host Keys
Aug 20 00:55:03 rpicm4-1.1stcall.uk root[1322]: sdm FirstBoot: Re-enable ***@***.***
Aug 20 00:55:27 rpicm4-1.1stcall.uk root[1375]: sdm FirstBoot: System restarting now
Aug 20 00:55:27 rpicm4-1.1stcall.uk systemd[1]: sdm-firstboot.service: Deactivated successfully.
Aug 20 00:55:27 rpicm4-1.1stcall.uk systemd[1]: Stopped sdm-firstboot.service - sdm FirstBoot System Configuration.
Aug 20 00:55:27 rpicm4-1.1stcall.uk systemd[1]: sdm-firstboot.service: Consumed 1.402s CPU time.

gitbls, 2024-08-20:
Thx! Will read through this and see what's duffed up.

gitbls, 2024-08-20:
Well, very strange. After the burn is done, phase 0 for all the plugins specified on the burn command line are run. However, not all of them are run for phase 1 or the post-install phase. Something is borking the plugin list. Not nice! Will sort it tomorrow. Thx for reporting, this looks like a good one!

gitbls, 2024-08-20:
I have a very simple (remove double quotes from a  single line) fix, which should work in all cases. 

I can't yet explain why the original code works correctly in Phase 0, but failed in Phase 1 and post-install phases, and I've been unable to simplify a repro case outside of sdm that fails in any scenario.

Will do some more testing here, then push the fix and update this issue so you can test it in your environment.

gitbls, 2024-08-20:
OK, I pushed an updated sdm-cparse for you to try.

In case you're interested, both of these commands are supposed to be equivalent:
```
	IFS="~" read -r -a plugs <<< "$theseplugs"  # what it used to be
```
and
```
	IFS="~" read -r -a plugs <<< $theseplugs   # what it is now
```
I learned [here](https://unix.stackexchange.com/questions/397205/split-string-using-ifs) that as of bash 4.4 (a while ago, I presume) you no longer need to quote the string. Which I was doing, and it generally worked. Except for some reason it doesn't work when plugins run during phase 1 and post-install. In that scenario, it for some reason drops most of the plugins string, like it is having IFS confusion of some sort.

Removing the double quotes fixes it. Not what I would have expected.

gitbls, 2024-08-20:
BTW, when you're using pluglist files you don't need the quotes around a plugin argument list.
For example:
```
system:name=myfstab|fstab=/rpi/etc/fstab.lan
```
Much easier on the eyes IMHO

gitbls, 2024-08-20:
Damn. Ignore the above. Test set failure.

It's not working with the apt-addrepo plugin, because of the spaces. Still mystified.

gitbls, 2024-08-21:
Remove the apt-addrepo plugin and fstab should work fine.

I'm a bit mystified at the moment how that plugin ever worked, so after fixing the plugin parsing problem the plugin needs some work as well.

1stcall, 2024-08-21:
No probs.  I should be able to achieve the same result using the copyfile plugin.

gitbls, 2024-08-22:
OK, fixed in V12.8. apt-addrepo plugin works better now, as well as any arg values with spaces in them. Thanks for pointing this out, it was a fairly fundamental bug.