gitbls/sdm #105: Copyfile is doing nothing

freumichCH, 2023-09-06:
Hello @gitbls 

i was coming across sdm from a couple days and find it very useful. Thanks for your work 

currently I'm facing a issue with the copyfile plugin. Unfortunately it does not copy any of my files.
Here my sdm command: 

```
sudo sdm --customize \
          --batch \
          --L10n \
          --restart \
          --disable piwiz \
          --host myhostname\
          --plugin apps:"apps=samba incron vim tree" \
          --plugin copyfile:"from=/root/s2c-builder/src/incron.allow|to=/etc|chown=root:incron" \
          --plugin copyfile:"from=/root/s2c-builder/src/test.txt|to=/root" \
          --plugin samba:smbconf=/root/s2c-builder/src/smb.conf \
          --regen-ssh-host-keys \
          --user "${PI_USER}" \
          --password-user "${PI_PASSWORD}" /root/s2c-builder/images/2023-05-03-raspios-bullseye-armhf-lite.img)"

```
The history looks like this. (Phase 0 and post-install)

```
2023-09-06 07:22:30 * Start Configuration
2023-09-06 07:22:30 > Command Line: /usr/local/bin/sdm --customize --batch --L10n --restart --disable piwiz --host 
2023-09-06 07:22:30   scan2cloud --plugin apps:apps=samba incron vim tree --plugin 
2023-09-06 07:22:30   copyfile:from=/root/s2c-builder/src/incron.allow|to=/etc/|chown=root:incron --plugin 
2023-09-06 07:22:30   copyfile:from=/root/s2c-builder/src/test.txt|to=/root --plugin 
2023-09-06 07:22:30   samba:smbconf=/root/s2c-builder/src/smb.conf --regen-ssh-host-keys --user myuser 
2023-09-06 07:22:30   --password-user mystroongpassword /root/s2c-builder/images/2023-05-03-raspios-bullseye-armhf-lite.img
2023-09-06 07:22:30 * Host Information
....
2023-09-06 07:22:30 * Start Phase 0 image customization
2023-09-06 07:22:30 *+Phase 0 sdm configuration
2023-09-06 07:22:30 *+Phase 0 Network configuration
2023-09-06 07:22:30 *+Phase 0 User configuration
2023-09-06 07:22:30 *+Phase 0 System configuration
2023-09-06 07:22:30 *+Phase 0 Raspberry Pi-specific configuration
2023-09-06 07:22:30 > Create user 'anykey' home directory /home/anykey
2023-09-06 07:22:30 > Run Plugins Phase '0'
2023-09-06 07:22:30 > Run Plugin 'apps' (/mnt/sdm/usr/local/sdm/plugins/apps) Phase 0 with arguments: 'apps=samba
2023-09-06 07:22:30   incron vim tree'
2023-09-06 07:22:30 * Plugin apps: Start Phase 0
2023-09-06 07:22:30 > Plugin apps: Keys/values found:
2023-09-06 07:22:30    apps: samba incron vim tree
2023-09-06 07:22:30 > Plugin apps [default]: Saved apps list: samba incron vim tree
2023-09-06 07:22:30 * Plugin apps: Complete Phase 0
2023-09-06 07:22:30 > Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments:
2023-09-06 07:22:30   'from=/root/s2c-builder/src/incron.allow|to=/etc/|chown=root:incron'
2023-09-06 07:22:30 * Plugin copyfile: Start Phase 0
2023-09-06 07:22:30 > Plugin copyfile: Keys/values found:
2023-09-06 07:22:30    from: /root/s2c-builder/src/incron.allow
2023-09-06 07:22:30    to: /etc/
2023-09-06 07:22:30    chown: root:incron
2023-09-06 07:22:30 > Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments:
2023-09-06 07:22:30   'from=/root/s2c-builder/src/test.txt|to=/root'
2023-09-06 07:22:30 * Plugin copyfile: Start Phase 0
2023-09-06 07:22:30 > Plugin copyfile: Keys/values found:
2023-09-06 07:22:30    from: /root/s2c-builder/src/test.txt
2023-09-06 07:22:30    to: /root
2023-09-06 07:22:30 > Run Plugin 'samba' (/mnt/sdm/usr/local/sdm/plugins/samba) Phase 0 with arguments:
2023-09-06 07:22:30   'smbconf=/root/s2c-builder/src/smb.conf'
2023-09-06 07:22:30 * Plugin samba: Start Phase 0
2023-09-06 07:22:30 > Plugin samba: Keys/values found:
2023-09-06 07:22:30    smbconf: /root/s2c-builder/src/smb.conf
2023-09-06 07:22:30 > Plugin samba: Copy samba smbconf '/root/s2c-builder/src/smb.conf' into IMG
2023-09-06 07:22:30   /etc/sdm/assets/samba-smb.conf
2023-09-06 07:22:30 * Plugin samba: Complete Phase 0
2023-09-06 07:22:30 * Phase 0 Completed
...
2023-09-06 07:26:39 > Run Plugins Phase 'post-install'
2023-09-06 07:26:39 > Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase post-install with arguments:
2023-09-06 07:26:39   'apps=samba incron vim tree'
2023-09-06 07:26:39 * Plugin apps: Start Phase post-install
2023-09-06 07:26:39 * Plugin apps: Complete Phase post-install
2023-09-06 07:26:39 > Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase post-install with arguments:
2023-09-06 07:26:39   'from=/root/s2c-builder/src/incron.allow|to=/etc/|chown=root:incron'
2023-09-06 07:26:40 * Plugin copyfile: Start Phase post-install
2023-09-06 07:26:40 * Plugin copyfile: Complete Phase post-install
2023-09-06 07:26:40 > Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase post-install with arguments:
2023-09-06 07:26:40   'from=/root/s2c-builder/src/test.txt|to=/root'
2023-09-06 07:26:40 * Plugin copyfile: Start Phase post-install
2023-09-06 07:26:40 * Plugin copyfile: Complete Phase post-install
2023-09-06 07:26:40 > Run Plugin 'samba' (/usr/local/sdm/plugins/samba) Phase post-install with arguments:
2023-09-06 07:26:40   'smbconf=/root/s2c-builder/src/smb.conf'
2023-09-06 07:26:41 * Plugin samba: Start Phase post-install
2023-09-06 07:26:41
2023-09-06 07:26:41 > Plugin samba: *** samba service notes ***
2023-09-06 07:26:41 * Plugin samba:  Ensure that /etc/samba/smb.conf is as you want it
2023-09-06 07:26:41 * Plugin samba:  Add users and passwords using 'sudo smbpasswd'
2023-09-06 07:26:41 * Plugin samba:  The service should be enabled when the system boots; review the system logs
2023-09-06 07:26:41   for possible errors
2023-09-06 07:26:41
2023-09-06 07:26:41 * Plugin samba: Complete Phase post-install
2023-09-06 07:26:41 %%%
2023-09-06 07:26:41 % IMG '/root/s2c-builder/images/2023-05-03-raspios-bullseye-armhf-lite.img' is 100% full (10308 1K-blocks free)
2023-09-06 07:26:41 % Review /etc/sdm/apt.log in the image for insufficient disk space errors
2023-09-06 07:26:41 % If needed use --extend --xmb nnnn with --customize to create more space
2023-09-06 07:26:42 %%%
2023-09-06 07:26:42 * Customization complete
2023-09-06 07:26:42 * Batch Mode exit
```

Curently installed Version 
`sdm V8.6`

Hope this helps to sort it out.

gitbls, 2023-09-06:
Weird. I will have a look at this when I get back to my desktop system later today. 

In the meantime, I notice that your IMG is full. Could you try your command again and add `--extend`, which will extend the IMG by 2048MB. While I doubt that this is the root cause, it would help to rule it out.

freumichCH, 2023-09-06:
appending `&& exit`( https://github.com/gitbls/sdm/issues/90#issuecomment-1692285701) did the trick.  
Thank you! Was not thinking that could be the solution. All files have been copied like expected. 

Thanks again for your Work.

closed 2023-09-06 by freumichCH

gitbls, 2023-09-06:
Great! I of course have fixed this in my sources, but didn't upload it, was holding off to fix it in the next release, didn't realize that more people were using it. I'll upload the updated version. Thx for the git push, but not needed, and I have too much going on to risk spending time wrestling with git(hub) on this at the moment.

freumichCH, 2023-09-07:
Fantastic!  Was thinking i could support you with my pull request. 
glad that it works now. Thanks again for your work.

gitbls, 2023-09-07:
I appreciate the thought, and frankly, it's on me to learn enough git/github to be able to accomodate the help. That's on my Post-V9.0 release ToDo list.

And, I most definitely appreciate your picking up and working with sdm. I was quite excited about the capabilities once I realized what it could do (3 years ago 😵), and am very glad that others find it useful as well.