gitbls/sdm #107: hotspot code not recognizing hostconfig (hotconfig)

dwatrous, 2023-09-15:
I ran this customize command:

`sudo sdm --customize --wpa wpa_supplicant.conf --disable piwiz --L10n --rename-pi mm --password-pi secret --hotspot hotspot.cfg --ssh service --regen-ssh-host-keys --host mmdwatrous --apt-dist-upgrade  2023-05-03-raspios-bullseye-armhf-lite.img`

With this in my host-config file:

`config="routed"`

But I see this in my logs

`* Start Access Point (hotspot) installation and configuration
' Unrecognized Access Point configuration type 'routed"`

This is being executed here: https://github.com/gitbls/sdm/blob/56017520b915f7d6ee32db3b38e957a617f5d352/sdm-hotspot#L138C3-L138C3

Based on the error message, the variable "hotconfig" is properly set to 'routed', but the conditional is still false for some reason.

I'm running WSL2 Debian:
`systemctl --version
systemd 252 (252.12-1~deb12u1)
+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified`

gitbls, 2023-09-15:
Hi, sorry that you're running into a problem with the hotspot. I've not paid attention to it for a while b/c I received no feedback from...anyone indicating that it was being used, or that there was interest in it.

At the current time, I'm in the middle of finishing up sdm V9.0, which I'm expecting to release sometime in the next several days.

I have been planning to revisit the hotspot code once V9.0 is out and get it working again, restructured into a new hotspot plugin. This doesn't mean a major overhaul, though...the existing code will form 99.9% of the new plugin. The only visible change will be that to use the hotspot plugin you'll add `--plugin hotspot:"args"` rather than using the `--hotspot` switch. Oh, and of course, it will work. 😁

Regarding your current issue, what is the result if you change your hotspot.cfg line to be `config=routed` (without the double quotes)?

dwatrous, 2023-09-16:
removing the quotes didn't help. I'll keep debugging.

gitbls, 2023-09-16:
Unfortunately, I'm not able to easily work on this here at the moment, but if you do a quick test for me, I should be able to direct you toward a fix. 

If you sudoedit sdm-hotspot you'll find
```
#
# Validity checks
#
if [[ ! "local|routed|bridged" =~ "$hotconfig" ]] || [ "$hotconfig" == "" ]
then
    logtoboth "? Unrecognized Access Point configuration type '$hotconfig'"
    exit
fi
```
Please add 
```
logtoboth "DEBUG hotconfig:|$hotconfig|"
```
immediately before the if statement. rerun your test and let me know what it returns.

Thx

dwatrous, 2023-09-16:
Here's what you asked for:

```
* Start Access Point (hotspot) installation and configuration 
|EBUG hotconfig:|routed                                       
' Unrecognized Access Point configuration type 'routed        
```

I'm not sure why the output is placing the last character at the front.

I also added `set -x` and ran the script and saw these results

```
+ case "${rpifun,,}" in  
' hotenable='true"
+ IFS=:=
+ read rpifun value   
|'logtoboth 'DEBUG hotconfig:|routed
|'local 'msg=DEBUG hotconfig:|routed
+ str=()
+ local str i spc=
+ '[' 25 -le 96 -o '' == nosplit ']'
|'logit 'DEBUG hotconfig:|routed
+ '[' -f /etc/sdm/history ']'
++ thisdate 
+++ date '+%Y-%m-%d %H:%M:%S'
++ echo '2023-09-15 20:23:48'
|'echo '2023-09-15 20:23:48 DEBUG hotconfig:|routed 
|'echo 'DEBUG hotconfig:|routed 
|EBUG hotconfig:|routed  
 ]][ ! local|routed|bridged =~ routed
'\'''toboth '? Unrecognized Access Point configuration type '\''routed
'\'''al 'msg=? Unrecognized Access Point configuration type '\''routed
+ str=()
+ local str i spc=
+ '[' 56 -le 96 -o '' == nosplit ']'
'\'''it '? Unrecognized Access Point configuration type '\''routed
+ '[' -f /etc/sdm/history ']'
++ thisdate 
+++ date '+%Y-%m-%d %H:%M:%S'
++ echo '2023-09-15 20:23:48'
'\'''o '2023-09-15 20:23:48 ? Unrecognized Access Point configuration type '\''routed  
'\'''o '? Unrecognized Access Point configuration type '\''routed 
' Unrecognized Access Point configuration type 'routed  
+ exit  
```

I'm still hunting...

gitbls, 2023-09-16:
The problem is that the code is seeing an extra character after the `config=routed` in hotspot.cfg. Is hotspot.cfg by some chance a file that you created on a Windows system?

```
dos2unix -n hotspot.cfg newhotspot.cfg
```
If you don't have dos2unix on your system, you can install it with `sudo apt install`, or you can run the following command on hotspot.cfg:
```
tr -d '\r' < hotspot.cfg > newhotspot.cfg
```
If the "before" and the "after" file are different, then the file had windows line endings.

I haven't used this in quite some time, but it most definitely worked the last time I tested it, which is why I'm suspicious of the file format.

dwatrous, 2023-09-16:
Oh, what a rookie move. I actually checked that for the file sdm-hotspot, but not for my own file. I'm using VSCode and had my settings wrong. After changing it to just LF (\n), it worked. Thanks for your help.

closed 2023-09-16 by dwatrous

gitbls, 2023-09-16:
Pshew! If that wasn't it I was going to be VERY confused. Glad that it was so simple. sdm should probably handle those nasty CRs better. Added to wishlist.