gitbls/sdm #108: Hotspot not working

dwatrous, 2023-09-16:
When I first boot, the following error prints out three times:
"[FAILED] Failed to start Access point and authentication server for Wi-Fi and Et"

I am trying to get the hotspot and WiFi to work at the same time. I previously accomplished this using this process: https://forums.raspberrypi.com/viewtopic.php?t=211542#p1355569 (https://github.com/idev1/rpihotspot). Some of the approach in `sdm` is similar, but I'm not sure exactly what's missing. One difference is that `sdm` references eth0 instead of `wlan0`. I'm not sure if that means it will only work if I connect to a wired network connection.

Some more debugging data:

```
root@mmdwatrous:/etc/sdm# ./0piboot/040-hostapd-enable.sh
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable hostapd
Synchronizing state of dnsmasq.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dnsmasq
```

This doesn't bring up `uap0`

```
root@mmdwatrous:/etc/sdm# ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 28  bytes 3318 (3.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 28  bytes 3318 (3.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.68.131  netmask 255.255.255.0  broadcast 192.168.68.255
        inet6 fe80::4246:82d6:7319:dc0b  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:ca:c3:8a  txqueuelen 1000  (Ethernet)
        RX packets 7909  bytes 1358009 (1.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 721  bytes 131783 (128.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

```
root@mmdwatrous:/etc/sdm# systemctl status hostapd
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2023-09-16 16:35:22 CDT; 2s ago
       Docs: man:hostapd(8)
    Process: 4197 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, statu>
        CPU: 45ms
```

```
root@mmdwatrous:/etc/sdm# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
     Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-09-16 15:29:12 CDT; 1h 6min ago
   Main PID: 397 (dnsmasq)
      Tasks: 1 (limit: 414)
        CPU: 887ms
     CGroup: /system.slice/dnsmasq.service
             └─397 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq>

Sep 16 15:29:09 mmdwatrous dnsmasq[397]: read /etc/hosts - 5 addresses
Sep 16 15:29:11 mmdwatrous dnsmasq[397]: no servers found in /run/dnsmasq/resolv.conf, will retry
Sep 16 15:29:12 mmdwatrous systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
Sep 16 15:29:26 mmdwatrous dnsmasq[397]: reading /run/dnsmasq/resolv.conf
Sep 16 15:29:26 mmdwatrous dnsmasq[397]: using nameserver 75.75.75.75#53
Sep 16 15:29:26 mmdwatrous dnsmasq[397]: using nameserver 75.75.76.76#53
Sep 16 16:14:27 mmdwatrous dnsmasq[397]: no servers found in /run/dnsmasq/resolv.conf, will retry
Sep 16 16:14:34 mmdwatrous dnsmasq[397]: reading /run/dnsmasq/resolv.conf
Sep 16 16:14:34 mmdwatrous dnsmasq[397]: using nameserver 75.75.75.75#53
Sep 16 16:14:34 mmdwatrous dnsmasq[397]: using nameserver 75.75.76.76#53
```

I know you're busy with the current release. When you get around to working on hotspot, I'm ready to test. If I make progress before then, I'll report back on this issue.

gitbls, 2023-09-16:
I think there are much better ways to go at that now than there were 5 years ago when that post was written. 

Once I get the release done I'll convert the existing hotspot to a plugin, then have a look at adding an option for simultaneous hotspot and wifi client.

gitbls, 2023-09-28:
A beta version of the hotspot plugin is available in V9.2, which is branch 'master'. Please give it a try and let me now how goes.

Thx!

PS I don't expect simultaneous hotspot and wifi client to work, one small step at a time. ;)

dwatrous, 2023-09-30:
I attempted to run `sudo sdm --customize --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --hotspot hotspot.cfg --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-l
ite.img`, but the `--hotspot` isn't recognized in version 9. I didn't see any specific mention of it in https://github.com/gitbls/sdm/blob/master/Docs/9Upgrade-Notes.md.

In which plugin is hotspot now?

gitbls, 2023-09-30:
> I attempted to run `sudo sdm --customize --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --hotspot hotspot.cfg --host mmdwatrous [...]

I *assumed* that when I used the phrase "hotspot plugin" that it would be clear that there is now a plugin for hotspot. Sadly, my assumption was based on my vantage point, not yours. Apologies!

See https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#hotspot for details on the hotspot plugin.

dwatrous, 2023-09-30:
I get something similar

```
$ sudo sdm --customize --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin hotspot:"config=hotspot.cfg" --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-l ite.img
? Unrecognized plugin 'hotspot'
```

gitbls, 2023-09-30:
ARRGH...neglected to update EZsdmInstaller for the new hotspot plugin. Please redo the following commands, which will download and rerun the sdm Installer.
```
curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
```
Apologies for the hassle.

dwatrous, 2023-09-30:
Great. I updated the app and was able to build an image which successfully boots.

I may not fully understand the difference between hotspot and captive portal. I expected the RPi to boot and have a WiFi ssid exposed so I could connect my phone to it. There is no published SSID. This is what I see on the Pi.
https://photos.app.goo.gl/u6xqnXDKvpaSvUpu8

gitbls, 2023-09-30:
What was your full customize command line?

dwatrous, 2023-09-30:
`sudo sdm --customize --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin hotspot:"config=hotspot.cfg" --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-lite.img`

`sudo sdm --burnfile sdm.img --expand-root 2023-05-03-raspios-bullseye-armhf-lite.img`

Write image using Raspberry Pi image writer on Windows

gitbls, 2023-09-30:
oops, and what is in hotspot.cfg? Also, can you do the following logged in on the system and post the output?
```sh
sudo systemctl status dhcpcd
sudo systemctl status hostapd
sudo systemctl status dnsmasq
sudo sed -n '/sdm/,$p' /etc/hostapd/hostapd.conf
sudo sed -n '/sdm/,$p' /etc/dhcpcd.conf
sudo grep network /etc/sdm/history
```

Some of these will generate several lines of output. Warm up the google photo paster!

dwatrous, 2023-09-30:
hotspot.cfg

```
# Type of hotspot
#  local: Clients can only access the hotspot IP itself
#  routed: Clients can access the hotspot IP; non-local traffic is routed to the Pi's eth0 network
#  bridged: The Client network is bridged onto the Pi's eth0 network
config=routed
# Channel to use
channel="7"
# WiFi mode: "g" for 2.4Ghz, "a" for 5Ghz
# See https://en.wikipedia.org/wiki/List_of_WLAN_channels for legal channels/modes per country
hwmode="a"
# Country: defaults to --wifi-country setting but can be changed here
country="us"
# Network device to use. Default is "wlan0"
dev="uap0"
# SSID for the hotspot network
ssid="MM_CTRL"
# Passphrase for the hotspot network
passphrase="secret"
# If enable=true, the hotspot will be enabled at system boot
enable="true"
```

dwatrous, 2023-09-30:
hostapd and dnsmasqd don't appear to be setup correctly.

https://photos.app.goo.gl/AFvwkaz8GVkQPn1i8

gitbls, 2023-09-30:
> hostapd and dnsmasqd don't appear to be setup correctly.
> 
> https://photos.app.goo.gl/AFvwkaz8GVkQPn1i8

Thanks. Need to see the /etc/sdm/history file from the IMG you customized. You can either post it here (but be sure to redact passwords) or email it to me as an attachment at gitbls @ outlook.com

A photo (or multiple photos) of the file will NOT be useful. Really need to see the file so that I can sort out what's going on.

What is dev `uap0` in your hotspot config file? I didn't see that in the ifconfig or iwconfig output you provided.

dwatrous, 2023-09-30:
```
mm@mmdwatrous:~ $ sudo cat /etc/sdm/history
2023-09-30 13:11:01 * Start Configuration
2023-09-30 13:11:01 > Command Line: /usr/local/bin/sdm --customize --plugin disables:piwiz --plugin L10n:host
2023-09-30 13:11:01   --plugin user:adduser=mm|password=secret --plugin hotspot:config=hotspot.cfg --host mmdwatrous
2023-09-30 13:11:01   2023-05-03-raspios-bullseye-armhf-lite.img
2023-09-30 13:11:01 * Host Information
2023-09-30 13:11:01    Hostname: DESKTOP-LM9QIPJ
2023-09-30 13:11:01    uname: Linux DESKTOP-LM9QIPJ 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13
2023-09-30 13:11:01   UTC 2023 x86_64 GNU/Linux
2023-09-30 13:11:01    os-release Name: "Debian GNU/Linux"
2023-09-30 13:11:01    os-release Version: "12 (bookworm)"
2023-09-30 13:11:01    sdm Version: V9.2
2023-09-30 13:11:01 * IMG Information
2023-09-30 13:11:01    Name: 2023-05-03-raspios-bullseye-armhf-lite.img
2023-09-30 13:11:01    Date: 2023-05-03
2023-09-30 13:11:01    RasPiOS Version: 11
2023-09-30 13:11:01    os-release Version: "11 (bullseye)"
2023-09-30 13:11:01 % Configure chroot for sdm on your 'x86-64' computer
2023-09-30 13:11:01   * sdm will run more slowly due to qemu emulation
2023-09-30 13:11:01 * Plugins included:
2023-09-30 13:11:01    * disables
2023-09-30 13:11:01        Args: piwiz
2023-09-30 13:11:01    * L10n
2023-09-30 13:11:01        Args: host
2023-09-30 13:11:01    * user
2023-09-30 13:11:01        Args: adduser=mm|password=secret
2023-09-30 13:11:01    * hotspot
2023-09-30 13:11:01        Args: config=hotspot.cfg
2023-09-30 13:11:01 > IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 294300 1K-blocks free
2023-09-30 13:11:01 > Copy sdm to /usr/local/sdm in the IMG
2023-09-30 13:11:01 * Start Phase 0 image customization
2023-09-30 13:11:01 > Run Plugins Phase '0'
2023-09-30 13:11:01 > Run Plugin 'disables' (/mnt/sdm/usr/local/sdm/plugins/disables) Phase 0 with arguments:
2023-09-30 13:11:01   'piwiz'
2023-09-30 13:11:01 * Plugin disables: Start Phase 0
2023-09-30 13:11:01 > Plugin disables: Keys/values found:
2023-09-30 13:11:01    piwiz:
2023-09-30 13:11:01 * Plugin disables: Complete Phase 0
2023-09-30 13:11:01 > Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments: 'host'
2023-09-30 13:11:01 * Plugin L10n: Start Phase 0
2023-09-30 13:11:01 > Plugin L10n: Keys/values found:
2023-09-30 13:11:01    host:
2023-09-30 13:11:01 > Plugin L10n: Read L10n configuration from host system
2023-09-30 13:11:01 % /etc/default/keyboard missing; Keyboard set to 'us'
2023-09-30 13:11:01 > Plugin L10n: Load Localization (L10N) settings from running system
2023-09-30 13:11:01 > Plugin L10n:   Keymap:       us
2023-09-30 13:11:01 > Plugin L10n:   Locale:       en_US.UTF-8
2023-09-30 13:11:01 > Plugin L10n:   Timezone:     America/Chicago
2023-09-30 13:11:01 > Plugin L10n:   WiFi Country:
2023-09-30 13:11:01 > Plugin L10n: Save L10n configuration
2023-09-30 13:11:01 ? Prefix L10n: Unrecognized keymap 'us'
2023-09-30 13:11:01 * Plugin L10n: Complete Phase 0
2023-09-30 13:11:02 > Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments:
2023-09-30 13:11:02   'adduser=mm|password=secret'
2023-09-30 13:11:02 * Plugin user: Start Phase 0
2023-09-30 13:11:02 > Plugin user: Keys/values found:
2023-09-30 13:11:02    adduser: mm
2023-09-30 13:11:02    password: secret
2023-09-30 13:11:02 > Plugin user: Create user 'mm' home directory '/home/mm' so available for other plugins
2023-09-30 13:11:02 * Plugin user: Complete Phase 0
2023-09-30 13:11:02 > Run Plugin 'hotspot' (/mnt/sdm/usr/local/sdm/plugins/hotspot) Phase 0 with arguments:
2023-09-30 13:11:02   'config=hotspot.cfg'
2023-09-30 13:11:02 * Plugin hotspot: Start Phase 0
2023-09-30 13:11:02 > Plugin hotspot: Keys/values found:
2023-09-30 13:11:02    config: hotspot.cfg
2023-09-30 13:11:02 > Plugin hotspot: Copy config file 'hotspot.cfg' to /mnt/sdm/etc/sdm/assets/hotspot/config
2023-09-30 13:11:02 * Plugin hotspot: Complete Phase 0
2023-09-30 13:11:02 * Phase 0 Completed
2023-09-30 19:11:02 * Start Phase 1 image customization
[... 77 lines cut ...]
2023-09-30 13:14:45 > First System Boot Custom Boot Scripts disabled
2023-09-30 13:14:45 > First System Boot automatic restart disabled
2023-09-30 13:14:45 * Burn Completed
2023-05-13 05:37:53 > FirstBoot: Start sdm FirstBoot Service
2023-05-13 05:37:53 > FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
2023-05-13 05:37:54 > FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
2023-05-13 05:37:54 > FirstBoot: Set WiFi Country to 'US'
2023-05-13 05:38:09 > FirstBoot: Waiting 10/120 seconds for system time synchronization
2023-05-13 05:38:21 > FirstBoot: Waiting 20/120 seconds for system time synchronization
2023-05-13 05:38:33 > FirstBoot: Waiting 30/120 seconds for system time synchronization
2023-05-13 05:38:45 > FirstBoot: Waiting 40/120 seconds for system time synchronization
2023-05-13 05:38:58 > FirstBoot: Waiting 50/120 seconds for system time synchronization
2023-05-13 05:39:10 > FirstBoot: Waiting 60/120 seconds for system time synchronization
2023-05-13 05:39:22 > FirstBoot: Waiting 70/120 seconds for system time synchronization
2023-05-13 05:39:34 > FirstBoot: Waiting 80/120 seconds for system time synchronization
2023-05-13 05:39:47 > FirstBoot: Waiting 90/120 seconds for system time synchronization
2023-05-13 05:39:59 > FirstBoot: Waiting 100/120 seconds for system time synchronization
[... cut ...] ystem time synchronization
2023-05-13 05:40:23 > FirstBoot: System time not synchronized after 120 second wait; proceeding to regenerate SSH host keys
2023-05-13 05:40:23 > FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/092-nm-hotspot-bridged.sh'
2023-05-13 05:40:25 > FirstBoot: Skip Custom FirstBoot Scripts
2023-05-13 05:40:25 > FirstBoot: Disable sdm-firstboot service
2023-05-13 05:40:30 > FirstBoot: Re-enable getty@tty1
2023-05-13 05:40:34 > FirstBoot: System startup complete
```

dwatrous, 2023-10-01:
`uap0` was intended to be the interface for hosting the wifi connection. When I have had it working previously (not with `sdm`), wlan0 was connected to the house wifi and uap0 exposed the hotspot.

I'm still not sure I understand the hotspot functionality, but it's what I came to before finding captive portal. I think captive portal might be a closer fit for my scenario, but I'm testing both.

gitbls, 2023-10-01:
Thanks for the logs and the perspective on what you're doing.

Looking at the log and your config file I see a few things going on. I'm tied up the rest of today, but will update the hotspot plugin tomorrow after explicit testing with your config. Will update you here when it's posted, along with the changes you need to make in your config file (less than the changes I'm making to the plugin 😳)

gitbls, 2023-10-01:
A quick update. I've corrected a few issues in the hotspot plugin, and I expect to check it in this afternoon USPT. I'll update this note once it's pushed.

Your hotspot.cfg file has the following errors:
* `config=routed` should be `type=routed`
* `channel=7` is an invalid channel for `hwmode=a`. The referenced wikipedia page might be helpful in sorting this out
* `dev=uap0` is incorrect for two reasons: `dev` should be `device`, and the actual device name must exist. Suggest you try `device=wlan0` on the updated build

gitbls, 2023-10-01:
V9.3 with updated `hotspot` is available.

Please redo the following commands, which will download and rerun the sdm Installer.
```
curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
```
Once you download the latest sdm and update your hotspot.cfg it should work better.

dwatrous, 2023-10-02:
I just ran through this again with v9.3. I don't see an exposed SSID that I can connect to.

hotspot.cfg

```
# Type of hotspot
#  local: Clients can only access the hotspot IP itself
#  routed: Clients can access the hotspot IP; non-local traffic is routed to the Pi's eth0 network
#  bridged: The Client network is bridged onto the Pi's eth0 network
type=routed
# Channel to use
channel="7"
# WiFi mode: "g" for 2.4Ghz, "a" for 5Ghz
# See https://en.wikipedia.org/wiki/List_of_WLAN_channels for legal channels/modes per country
hwmode="g"
# Country: defaults to --wifi-country setting but can be changed here
country="us"
# Network device to use. Default is "wlan0"
device="wlan0"
# SSID for the hotspot network
ssid="MM_CTRL"
# Passphrase for the hotspot network
passphrase="secret"
# If enable=true, the hotspot will be enabled at system boot
enable="true"
```

I'm working on getting `/etc/sdm/history`. Last time I used `raspi-config` to set the SSID and password to connected to WiFi so I could SSH into the device to retrieve the file. This time when I try that, it tells me there is no wireless interface, so I can't setup WiFi.

dwatrous, 2023-10-02:
I know this is messy. I used Google Lens to extract the text from images, but hopefully it gives you enough visibility to make sense of what might be happening.

```
2023-10-02 06:07:13 Start Configuration 2023-10-02 06:07:13 > Command Line: /usr/local/bin/sdn

--customize --plugin disables:piwiz --plugin L10n:host

2023-10-02 06:07:13 -plugin user:adduser=mm Ipassword-secret --plugin hotspot:config-hotspot.cfg --host mndwatrous

S2023-10-02 06:07:13 2023-05-03-raspios-bullseye-armhf-lite. ing 2023-10-02 06:07:13 Host Information 2023-10-02 06:07:13 Hostname: DESKTOP-LM9QIPJ

2023-10-02 06:07:13 2023-10-02 06:07:13

uname: Linux DESKTOP-LM9QIPJ 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13

UTC 2023 x86_64 GNU/Linux 2023-10-02 06:07:13 os-release Name: "Debian GNU/Linux"

2023-10-02 06:07:13 os-release Version: "12 (bookworm)"

2023-10-02 06:07:13 sdn Version: V9.3

2023-10-02 06:07:13 $2023-10-02 06:07:13

IMG Information Name: [2023-05-03-raspios-bullseye-armhf-lite.ing](http://2023-05-03-raspios-bullseye-armhf-lite.ing/)

$2023-10-02 06:07:13 Date: 2023-05-03

2023-10-02 06:07:13 RasPiOS Version: 11

$2023-10-02 06:07:13 os-release Version: "11 (bullseye)" 2023-10-02 06:07:13 Configure chroot for sdn on your

$

'x86-64' computer

$2023-10-02 06:07:13 sdm will run more slowly due

to qenu enulation

2023-10-02 06:07:13 Plugins included:

2023-10-02 06:07:13 *disables

S2023-10-02 06:07:13 2023-10-02 06:07:13

Args: piuiz L10n

2023-10-02 06:07:13

Args: host

2023-10-02 06:07:13

user Args: adduser-nmlpassword-secret

2023-10-02 06:07:13

2023-10-02 06:07:13 * hotspot 2023-10-02 06:07:13

Args: config-hotspot.cfg 2023-10-02 06:07:13 > IMG '2023-05-03-raspios-bullseye-armhf-lite.ing' has 294300 1K-blocks (GB, GiB) free

2023-10-02 06:07:13 > Copy sdn to /usr/local/sdn in the IMG

2023-10-02 06:07:13 Start Phase 0 inage customization

2023-10-02 06:07:13 > Run Plugins Phase '0' 2023-10-02 06:07:13 > Run Plugin 'disables'

(/mnt/sdn/usr/local/sdn/plugins/disables) Phase with arguments:

2023-10-02 06:07:13 'piuiz'
[... 137 lines cut ...]

for user ''

-10-02 06:08:15 Plugin user: Complete

06:08:15 Run Plugin 'hotspot' (/usr/local/sdn/plugins/hotspot) Phase 1 with arguments: 06:08:15 'config-hotspot.cfg" 5:08:15 Plugin hotspot: Start Phase 1

5:08:15 7 in hotspot: Install "hostapd' and install iptables' hotspot: Unrecognized hotspot option 'config' in /etc/sdn/assets/hotspot/config'

:08:15 >

06:08:29 Plugin hotspot: Install dasma

2 06:08:40 Plugin hotspot: Hotspot will be enabled by FirstBoot

: 06:08:40 Plugin hotspot: Complete Phase -02 06:08:40 Phase 1

'apt upgrade" 2 06:08:41 IMG 2023-05-03-raspios-bullseye-armhf-lite.ing' has 211376 1K-blocks (0.268, 0.2618) free at start of 'apt upgrade" 2 06:08:40 Start post-app Installation/configuration

06:10:42 IMG 2023-05-03-raspios-bullseye-armhf-lite. Ing' has 118900 1K-blocks (0.168, 0.1618) free at end of 'apt upgrade" 06:10:42 Start 'apt autorenove" 6:10:42 IMG [2023-05-03-raspios-bullseye-armhf-lite.ing](http://2023-05-03-raspios-bullseye-armhf-lite.ing/) has 118900 1K-blocks (0.16B, 0.1618) free at start of 'apt autorenove

0:45 IMG 2023-05-03-raspios-bullseye-armhf-lite.ing' has 118900 18-blocks (0.168, 0.1618) free at end of 'apt autorenove 10:46 IMG 2023-05-03-raspios-bullseye-arnhf-lite.ing' has 118896 1X-blocks (0.16B, 0.161B) free at end of Phase 1 image customization 0:46 Phase 1 Completed

:10:46 > Run Plugins Phase 'post-install' 'pluizin 'disables' (/usr/local/sdn/plugins/disables) Phase post-install with arguments: 2023-10-02 06:10:46 Plugin disables: Start Phase post-install
```

gitbls, 2023-10-02:
What Pi are you booting this on? Does said Pi have another SD card you can boot that has an OS with a network on it? And, do you have an SD Card reader that you can plug into this Pi to look at the Card you're working on with sdm?

I'm asking the above because I need to see the boot-time logs, which you won't be able to capture with sufficient fidelity using Lens (creative idea, btw). The sdm log you provided looks good.

gitbls, 2023-10-02:
/var/log/syslog

Thx

gitbls, 2023-10-02:
Thx for the syslog. See my previous note about `--restart` on your customize. That will help, but I also noticed:
```
May 13 05:47:18 mmdwatrous hostapd[346]: Configuration file: /etc/hostapd/hostapd.conf
May 13 05:47:18 mmdwatrous hostapd[346]: Line 13: invalid WPA passphrase length 6 (expected 8..63)
May 13 05:47:18 mmdwatrous hostapd[346]: WPA-PSK enabled, but PSK or passphrase is not configured.
May 13 05:47:18 mmdwatrous hostapd[346]: 2 errors found in configuration file '/etc/hostapd/hostapd.conf'
May 13 05:47:18 mmdwatrous hostapd[346]: Failed to set up interface with /etc/hostapd/hostapd.conf
May 13 05:47:18 mmdwatrous hostapd[346]: Failed to initialize interface
```
Your hotspot password is too short. This causes hostapd to fail.

So, fix the length of the hotspot password and add `--restart` to your customize command line, and it should work.

I'm planning to update the hotspot FirstBoot code to actually start the hotspot and dnsmasq if you don't use `--restart`, but that is not in the version you have now.

dwatrous, 2023-10-03:
Increasing the length of the password and adding `--restart` succeeded in getting the hotspot to start and I can connect.

During startup, the following message is displayed twice
`[FAILED] Failed to start Accesspoint...`. When I type `systemctl status hostapd.service`, it shows that the service is running, and I can connect. I'm not sure if it's worth digging deeper into this. Let me know if you want more logs here.

I still don't think I fully understand the difference between hotspot and captive portal. I think hotspot is supposed to share an existing network connection, but based on the current implementation, that would have to be a wired ethernet connection, because it doesn't currently allow the device to concurrently expose a WiFi SSID and connect to one at the same time. Since I'm using a RPi zero 2 w, I don't have a physical connection.

I think the captive portal is intended for one time use and makes it possible to wirelessly provide the device with credentials so it can connect to WiFi.

Is there more to it?

gitbls, 2023-10-03:
> Increasing the length of the password and adding `--restart` succeeded in getting the hotspot to start and I can connect.
> 

Excellent! 

> During startup, the following message is displayed twice `[FAILED] Failed to start Accesspoint...`. When I type `systemctl status hostapd.service`, it shows that the service is running, and I can [...]

A hotspot is typically something that you'd run regularly, and use it to share, for instance, via wifi an eth0-connected internet connection with several wifi client devices, or provide access to an eth0-connected LAN from wireless devices either as part of the LAN address space (bridged), or on a separate wireless network that is routed onto the LAN.

The captive portal (as implemented in sdm's `wificonfig` plugin) is used to configure the Pi's wireless network so it can connect to the LAN, typically as a client device using network services, although there is nothing that would restrict the Pi to that.

So, your understanding is correct, With the Pi02W as a local hotspot, you could use it to provide clients access to shared files on the Pi02W. Or, you could get a physical network connection via use of a USB ethernet adapter to enable clients to access the LAN via bridged or routed mode.

> I think the captive portal is intended for one time use and makes it possible to wirelessly provide the device with credentials so it can connect to WiFi.
> 

Yes, that is exactly correct.

> Is there more to it?

Not in the current implementation in sdm.

Once I've tested my updates to the hotspot plugin I'll post them and update this issue with that information.

gitbls, 2023-10-03:
I've pushed an updated hotspot plugin to master. Just reinstall sdm and you'll pick it up
```sh
curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
```

gitbls, 2023-10-23:
Closing issue since no further activity. Please re-open if you're still having a problem with this.

closed 2023-10-23 by gitbls