gitbls/sdm #109: Captive Portal not working according as described in docs

dwatrous, 2023-09-20:
I was attempting to us https://github.com/gitbls/sdm/blob/master/Docs/Captive-Portal.md, but I get the following errors

`sdm: unrecognized option '--nowpa'`

`? Unrecognized plugin 'wifconfig'`

Should this be working or is it part of the upcoming release?

dwatrous, 2023-09-22:
I read in the [release notes](https://github.com/gitbls/sdm/blob/56017520b915f7d6ee32db3b38e957a617f5d352/CHANGELOG.md#v80) that --nowpa as removed when with the introduction of the [network plugin in version 8](https://github.com/gitbls/sdm/blob/56017520b915f7d6ee32db3b38e957a617f5d352/Docs/Upgrade-Notes.md#upgrading-to-sdm-v8)

I have looked at the [network documentation](https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#network), but I don't see a way to satisfy the Captive Portal requirements.

dwatrous, 2023-09-22:
I also just noticed that I had a typo when referring to the `wificonfig` plugin. I fixed the typo and am trying again.

gitbls, 2023-09-22:
Sorry for the delay in getting back to you. I'm just finishing up sdm V9. I haven't yet revisited the captive portal in V9, but will do that after V9 is complete, and will update if needed to fix any issues.

Please update this with the result of using the `wificonfig` plugin. If you're still having issues, please provide the command line you're using.

dwatrous, 2023-09-22:
I ran this command

`sudo sdm --customize --disable piwiz --L10n --rename-pi mm --password-pi secret --host mmdwatrous --plugin wificonfig:"country=us" 2023-05-03-raspios-bullseye-armhf-lite.img`

relevant output here:

```
> Run Plugin 'wificonfig' (/usr/local/sdm/plugins/wificonfig) Phase 1 with arguments:                                       
  'country=us'                                                                                                              
* Plugin wificonfig: Start Phase 1                                                                                          
* Plugin wificonfig: Complete Phase 1                                                                                       
* Phase 1 post-app installation/configuration                                                                               
> Start 'apt upgrade'                                                                                                       
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 214808 1K-blocks free at start of 'apt upgrade'                      
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 121728 1K-blocks free at end of 'apt upgrade'                        
> Start 'apt autoremove'                                                                                                    
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 121728 1K-blocks free at start of 'apt autoremove'                   
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 121728 1K-blocks free at end of 'apt autoremove'                     
> Disable running piwiz and userconfig on first boot per --disable piwiz                                                    
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 121728 1K-blocks free at end of Phase 1 image customization          
* Phase 1 Completed                                                                                                         
> Run Plugins Phase 'post-install'                                                                                          
> Run Plugin 'wificonfig' (/usr/local/sdm/plugins/wificonfig) Phase post-install with                                       
  arguments: 'country=us'                                                                                                   
* Plugin wificonfig: Start Phase post-install                                                                               
plugin-wificonfig: #!/bin/bash                                                                                              
  logger "FirstBoot: Run plugin-configured Captive Portal"                                                                  
  /usr/local/sdm/sdm-cportal --sdm  --apssid sdm --apip 10.1.1.1 --country us                                               
  >/etc/sdm/wifi-config.log 2>&1                                                                                            
* Plugin wificonfig: Complete Phase post-install                                                                            
> IMG '2023-05-03-raspios-bullseye-armhf-lite.img' has 121716 1K-blocks free at end of image customization                  
* Enter Shell Command Prompt                                                                                                
> 'exit' to exit back to host system                                                                                        
root@DESKTOP-LM9QIPJ:/# exit                                                                                                
exit                                                                                                                        
* Customization complete                                                                                                    
umount: /mnt/sdm/boot unmounted                                                                                             
umount: /mnt/sdm unmounted                                                                                                  
```

On first boot, I see this: https://photos.app.goo.gl/d38DLwCv2QPMK2wQ8

There was no SSID visible that I could connect to

gitbls, 2023-09-24:
This will be corrected in the next release. I'll update this issue so you can test it at that time.

gitbls, 2023-09-28:
V9.2 is now available. Can you please rerun your test using the latest sdm (branch 'master' is V9.2).

thx!

dwatrous, 2023-09-30:
It looks like a lot changed since the last release I was using. Here's what I'm getting with 9.2. I did look at the changelog, but I didn't see anything about all these:

```
$ sudo sdm --version
sdm V9.2
$ sudo sdm --customize --disable piwiz --L10n --rename-pi mm --password-pi secret --host mmdwatrous --plugin wificonfig:"country=us" 2023-05-03-raspios-bullseye-armhf-lite.img
sdm: unrecognized option '--disable'
sdm: unrecognized option '--L10n'
sdm: unrecognized option '--rename-pi'
sdm: unrecognized option '--password-pi'
? /usr/local/bin/sdm: Unable to parse command
```

I'm going to start looking closer to see how these might have changed

gitbls, 2023-09-30:
Yep, a lot has changed. In order to get you jump-started, try this. Not tested, may be typos, but will get you in the right direction quickly. Pretty much everything that used to be done with switches is now done via plugins. It's much more modular and much easier to extend, and features are much easier to add.

Also see https://github.com/gitbls/sdm/blob/master/Docs/9Upgrade-Notes.md
```
$ sudo sdm --customize --plugin disables:piwiz --plugin:L10n:host --plugin user:"adduser=mm|password=secret" --host mmdwatrous --plugin wificonfig:"country=us" 2023-05-03-raspios-bullseye-armhf-lite.img
```

dwatrous, 2023-09-30:
Thanks for the link to the 9 upgrade notes.

I ran the above, but the resulting image wouldn't boot. The screen would only show a flashing cursor in the top left of the screen and would never progress. I removed `--plugin wificonfig:"country=us"` and the resulting image does start to boot. Booting gets stuck at the `resize2fs_once.service`, which eventually fails.

I'll try digging a little deeper later today. Any hints welcome.

gitbls, 2023-09-30:
What command did you use to burn the IMG to an SSD/SD card?

dwatrous, 2023-09-30:
Raspberry Pi imager v1.7.4

gitbls, 2023-09-30:
> Raspberry Pi imager v1.7.4

I don't use that enough to be able to comment on whether that should work or not, but here are some features that you will lose out on by not using sdm to burn the IMG to an SSD/SD Card:
* Auto-disabling of the RasPiOS firstboot service if you specify both `--expand-root` and `--regen-ssh-host-keys`
* Forcing regen-ssh-host-keys to run during first boot AFTER the system time has been set.
* The ability to run plugins during the burn process, which is handy if you want different systems to have different settings, but built from a common IMG
* Password redaction

The first item is the most important from my perspective, but the choice of burn tool is certainly up to you.

If you find that redoing the burn and booting it again results in a hung system as you described, LMK and I can provide some instructions on how to grab the system log from the failed boot disk so you can provide it to me for inspection.

dwatrous, 2023-09-30:
I'm not opposed to using sdm, but I'm running in WSL, which I wasn't sure supported directly writing to the SD card.

gitbls, 2023-09-30:
> I'm not opposed to using sdm, but I'm running in WSL, which I wasn't sure supported directly writing to the SD card.

Agree. I'm pretty frustrated that writing to the SD card on WSL *still* doesn't work. But I'm still hopeful that it gets enabled at some point.

dwatrous, 2023-09-30:
Does this mean I'm out of luck for these features with sdm using WSL? Do I need to get an actual linux host to use these features?

gitbls, 2023-09-30:
The first 2 bullets are definitely tied to using sdm for burning at the moment.

The last 2 are less important, but most definitely tied to using sdm for burning.

Since Pis are more available these days, you could have a Pi "workstation" where you install and run sdm. You can fairly easily set up an NFS server on that pi, and access the IMG over NFS if you want to run sdm there, or in general share files. Of course sdm runs quite well on the Pi itself for both customization and burning.

In other words, a Pi running RasPiOS or Ubuntu makes a great "actual linux host"

dwatrous, 2023-10-03:
With v9.3 I got ran the following

```
sudo sdm --customize --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin wificonfig:"country=us" --restart --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-lite.img
sudo sdm --burnfile sdm.img --expand-root 2023-05-03-raspios-bullseye-armhf-lite.img
```

This booted successfully, started the captive portal, allowed me to set the SSID and password. There was a long delay and then the system restarted. When it came up, it connected to the SSID provided and has internet.

gitbls, 2023-10-03:
Excellent. Based on your comments in https://github.com/gitbls/sdm/issues/108 it sounds like you've fully understood what's there.

As you think through if and how you can use this, please feel free to post any questions/requests/suggestions about either of these plugins or sdm in general.

gitbls, 2024-06-02:
Closing due to lack of activity. @dwatrous please re-open or start another issue if needed.

closed 2024-06-02 by gitbls