gitbls/sdm #111: v9.1.1 - Prompted for keyboard configuration despite using L10n plugin

zachwaite, 2023-09-26:
Since v9 I'm now prompted to enter keyboard layout when the pi first boots, despite customizing with:

`--plugin L10n:"keymap=us|locale=en_US.UTF-8|timezone=America/New_York"

I'm probably missing something, but can't find any other keyboard options in the docs. Any help would be appreciated!

This is the screen I'm seeing:
[image]

gitbls, 2023-09-26:
It's documented, but a bit hidden. See https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#disables. I'll add a note to the L10n plugin mentioning how to avoid this.

Simply add `--plugin disables:piwiz` to your customize or burn command line.

zachwaite, 2023-09-26:
Ah, yes. Thank you.

Also, IMO the v9 API changes do simplify usage.

gitbls, 2023-09-26:
> Ah, yes. Thank you.
> 
> Also, IMO the v9 API changes do simplify usage.

Thanks! I think so too 👍