gitbls/sdm #112: Customizing non raspbian images

wreedamz, 2023-10-05:
Hi. I have been using sdm with raspbian for a while now, and I am interested in using it to customize an ubuntu image. I see [here](https://github.com/gitbls/sdm/blob/master/sdm#L623C1-L623C66) and [here](https://github.com/gitbls/sdm/blob/27da4e020513b69c6f453760d0c43b132d9eb423/Docs/Compatibility.md?plain=1#L5C1-L5C51) that it is not supported. Do you have any pointers about what would need to be changed with sdm in order to support this or if other tools exist for this space? My team would potentially be interested in contributing to expand sdm for this use case

gitbls, 2023-10-05:
There are RasPiOS (Raspbian is no longer the OS name btw) assumptions buried all over the place. Making it work with another OS would be a process of finding all those places, deciding what to do to address it, and then make the appropriate changes without turning the code into unmaintainable spaghetti. Doable, but not trivial.

As far as Ubuntu, every Ubuntu release that I've tried goes through a large configuration process at the start of the first boot. Presumably you'd disable that somehow and make the changes you want?

If you're serious about this, I'd suggest that you disable the OS check in sdm, see what kinds of problems you run into, and then assess what would be needed to support it.  I'd be interested in seeing what you turn up in your evaluation.

wreedamz, 2023-10-05:
Thanks for the suggestion. We are still evaluating if this is necessary for our use case. If it is, I'm sure I will try this and report back.

gitbls, 2023-10-05:
I thought more about this and I think it's not nearly as bad now that most things are implemented as plugins. The _core_ plugins you'll care about for ubuntu are: apps, bootconfig, disables, graphics, network, quietness (if you care), raspiconfig, system, user. 

I have some ideas around how the core sdm would be extended/restructured via distro-specific plugins to more flexibly support additional distros. Having someone to sort out what needs to be done to support another distro (inform the distro plugin design) and then help make it happen (the distro-specific plugin) will make it a lot more interesting for me 👍.

gitbls, 2023-11-06:
Closing this issue due to lack of activity. Please re-open if further discussion warranted.