gitbls/sdm #113: SDM update from 8.6 to 9.x breaks dns config

origonn, 2023-10-09:
Hello, I'd first like to say I'm really enjoying SDM and have been using it since ~April to further update some of my homelab Ansible automation, mainly with a Debian Bookworm image from [https://raspi.debian.net/](https://raspi.debian.net). Unfortunately, with updates from 8.6 to a 9.x branch, a few things that were previously functional have broken.

Before 9.x, it looks like /etc/resolv.conf was taken by default from the host. Now it is not populated, resulting in no updates / app installation possible, unless you copy it in in a phase0 plugin.

### SDM 8.6 customizing Debian12
<details>
<summary>sudo sdm --customize --apt-dist-upgrade --nopassword 20230612_raspi_3_bookworm.img</summary>
<pre>
* Mount IMG '20230612_raspi_3_bookworm.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
chmod: cannot access '/mnt/sdm//usr/local/sdm/local-plugins/*': No such file or directory
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --apt-dist-upgrade --nopassword 
  20230612_raspi_3_bookworm.img
* Host Information
   Hostname: overseer
   uname: Linux overseer 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) 
  x86_64 GNU/Linux
   os-release Name: "Debian GNU/Linux"
   os-release Version: "12 (bookworm)"
   sdm Version: V8.6
* IMG Information
   Date: 
   RasPiOS Version: 12
   os-release Version: "12 (bookworm)"
% No username specified with the --user switch; Using 'pi'
% sdm will try to use systemd-nspawn on this 'x86-64' system
  If this does not work please post a GitHub issue
> IMG '20230612_raspi_3_bookworm.img' has 1074480 1K-blocks free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
*+Phase 0 sdm configuration
*+Phase 0 Network configuration
*+Phase 0 User configuration
*+Phase 0 System configuration
*+Phase 0 Raspberry Pi-specific configuration
> Create user 'pi' home directory /home/pi
* Phase 0 Completed
* Enter image '20230612_raspi_3_bookworm.img' for Phase 1
* Start Phase 1 image customization
> IMG '20230612_raspi_3_bookworm.img' has 1074460 1K-blocks free at start of Phase 1 image customization
*+Phase 1 sdm configuration
> Configure and enable sdm FirstBoot service (sdm-firstboot)
*+Phase 1 Network/WiFi/Bluetooth configuration
> Enable SSH service via /boot/ssh
*+Phase 1 User configuration
> Add user 'pi'
usermod: group 'spi' does not exist
usermod: group 'i2c' does not exist
usermod: group 'gpio' does not exist
> Set /home/pi files ownership to 'pi:pi'
> Add user 'pi' to sudoers.d
/usr/local/sdm/sdm-phase1: line 244: /etc/sudoers.d/010_pi-nopasswd: No such file or directory
chmod: cannot access '/etc/sudoers.d/010_pi-nopasswd': No such file or directory
> Skip password processing per --nopassword
*+Phase 1 System and Boot Configuration
> System journal configuration not modified
> Start 'apt update'
> IMG '20230612_raspi_3_bookworm.img' has 1074436 1K-blocks free at start of 'apt update'
> IMG '20230612_raspi_3_bookworm.img' has 916144 1K-blocks free at end of 'apt update'
* Phase 1 post-app installation/configuration
> Start 'apt dist-upgrade'
> IMG '20230612_raspi_3_bookworm.img' has 916144 1K-blocks free at start of 'apt dist-upgrade'
> IMG '20230612_raspi_3_bookworm.img' has 510040 1K-blocks free at end of 'apt dist-upgrade'
> Start 'apt autoremove'
> IMG '20230612_raspi_3_bookworm.img' has 510040 1K-blocks free at start of 'apt autoremove'
> IMG '20230612_raspi_3_bookworm.img' has 510040 1K-blocks free at end of 'apt autoremove'
> IMG '20230612_raspi_3_bookworm.img' has 510040 1K-blocks free at end of Phase 1 image customization
* Phase 1 Completed
> IMG '20230612_raspi_3_bookworm.img' has 510040 1K-blocks free at end of image customization
* Enter Shell Command Prompt
> 'exit' to exit back to host system
root@sdm:/# cat /etc/resolv.conf 
# Generated by NetworkManager
search lan
nameserver 1.1.1.1
nameserver 10.2.16.1
root@sdm:/# 
exit
* Customization complete
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
</pre>
</details>

### SDM 9.3 customizing Debian12
<details>
<summary>sudo sdm --customize --apt-dist-upgrade 20230612_raspi_3_bookworm.img</summary>
<pre>
* Mount IMG '20230612_raspi_3_bookworm.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --apt-dist-upgrade 
  20230612_raspi_3_bookworm.img
* Host Information
   Hostname: overseer
   uname: Linux overseer 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) 
  x86_64 GNU/Linux
   os-release Name: "Debian GNU/Linux"
   os-release Version: "12 (bookworm)"
   sdm Version: V9.3
* IMG Information
   Name: 20230612_raspi_3_bookworm.img
   Date: 
   RasPiOS Version: 12
   os-release Version: "12 (bookworm)"
% Configure chroot for sdm on your 'x86-64' computer
  * sdm will run more slowly due to qemu emulation
% No plugins are selected
> IMG '20230612_raspi_3_bookworm.img' has 1074524 1K-blocks (1.1GB, 1.0GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> No Plugins specified with --plugin
* Phase 0 Completed
* Enter image '20230612_raspi_3_bookworm.img' for Phase 1
* Start Phase 1 image customization
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1064300 1K-blocks (GB, GiB) free at start of Phase 1 image customization
> Configure and enable sdm FirstBoot service (sdm-firstboot)
> Start 'apt update'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1064296 1K-blocks (GB, GiB) free at start of 'apt update'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at end of 'apt update'
> No Plugins specified with --plugin
* Phase 1 post-app installation/configuration
> Start 'apt dist-upgrade'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at start of 'apt dist-upgrade'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at end of 'apt dist-upgrade'
> Start 'apt autoremove'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at start of 'apt autoremove'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at end of 'apt autoremove'
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at end of Phase 1 image customization
* Phase 1 Completed
> No Plugins specified with --plugin
> Run graphics post-install configuration
> Enable SSH
> Plugin phase1: Enable SSH service via /boot/ssh
//usr/local/sdm/sdm-cparse: line 195: python3: command not found
//usr/local/sdm/sdm-cparse: line 196: python3: command not found
> IMG '20230612_raspi_3_bookworm.img' has 1060912 1K-blocks (GB, GiB) free at end of image customization
* Enter Shell Command Prompt
  'exit' to exit back to host system
root@overseer:/# cat /etc/resolv.conf
cat: /etc/resolv.conf: No such file or directory
root@overseer:/# 
</pre>
</details>

gitbls, 2023-10-09:
I'm not surprised there are issues, since "Debian 12" != "RasPiOS Bookworm". One notable difference from your log: RasPiOS is delivered with Python preinstalled, whereas it appears to not be there on Debian 12. I would expect there to be a few others as well.

That said, with respect to /etc/resolv.conf, sdm never touches that file directly, but I'm mystified enough to have a look at it.

origonn, 2023-10-09:
Thank you for the feedback. While there were a few warnings in L10n related settings, I had no other issues in customizing that Bookworm Debian image for 2-3months. I have noted pre-apt (where I install python3) is in [getgbstr](https://github.com/gitbls/sdm/blob/27da4e020513b69c6f453760d0c43b132d9eb423/sdm-cparse#L195) (and have submitted a PR for your review / feedback). My customization script (once reverted to account for flags -> plugins) still works on 8.6.

To my understanding, the problem lies in the (now) use of qemu instead of systemd-nspawm, but I haven't started digging into it enough to understand why.

<pre>
8.6
% sdm will try to use systemd-nspawn on this 'x86-64' system
  If this does not work please post a GitHub issue
> IMG '20230612_raspi_3_bookworm.img' has 1074480 1K-blocks free 
---
9.3
% Configure chroot for sdm on your 'x86-64' computer
  * sdm will run more slowly due to qemu emulation
% No plugins are selected
> IMG '20230612_raspi_3_bookworm.img' has 1074524 1K-blocks (1.1GB, 1.0GiB) free 
</pre>

gitbls, 2023-10-09:
Issue https://github.com/gitbls/sdm/issues/92 reported that a WSL Ubuntu 22.04 didn't work using systemd-nspawn. There's little win in chasing distro-level problems such as this when re-enabling chroot for all cross-architecture customizations (x86-64 host customizing aarch64) fixes the problem.

So, V9.x re-enables using chroot, which you've noted. It turns out that a bit of `systemd-nspawn` that I had missed is that it (by default) copies (or maybe mount --bind) the host's /etc/resolv.conf into the IMG. I was quite surprised when I did an `sdm --explore` on a virgin IMG and /etc/resolv.conf was exactly what was on the host's /etc/resolv.conf

A quick read of `man systemd-nspawn` reveals the `--resolv-conf` switch that is supposed to disable this behavior when provided an argument of `off`. My quick test before I had to run out indicates that it's a bit more complex than that, since the /etc/resolv.conf was still updated even with `--resolv-conf off`. While I was out on my errand it occurred to me that systemd-nspawn may try to do this unless networking is disabled in some way. I need to have a look into that. I don't want to disable it by default because I find that being able to `scp` from wiithin a systemd-nspawn is pretty handy.

Since there's no /etc/resolv.conf in the `chroot` scenario, it's pretty clear that chroot does not muck with /etc/resolv.conf and I imagine we'll find that NetworkManager updates resolv.conf as part of its boot-time startup. And I'm pretty sure that dhcpcd ran resolvconf at startup to configure it on dhcpcd-powered systems.

So, the resolution of resolv.conf missing in sdm via chroot is "by design". The only bummer is that under WSL, where chroot is used, you can't scp since resolv.conf isn't there. 

I think the best solution is to give you control over whether to use systemd-nspawn or chroot, if you know it works on your WSL host (or maybe sdm can detect it and fall back to chroot?). More investigation required.

Did I miss anything?

origonn, 2023-10-09:
To clarify, the host I am running this on is _also_ a Debian12 (Bookworm) machine, there is no WSL involvement in my case. With no changes other than updating sdm from V8.6 to V9.3, it defaults from using systemd-nspawn to qemu.

> I don't want to disable it by default because I find that being able to scp from wiithin a systemd-nspawn is pretty handy.

I have to say I'm not entirely understanding what you mean by that. I don't mean that the older version (using systemd-nspawn) using host's resolv.conf is an issue (as you naturally need _some_ nameservers to be able to apt update / apt install).

Is it a case that the test/dev environment still defaults to systemd-nspawn (I guess in which case, why did my setup switch to qemu). Or that the test/dev environment has dns resolution (to be able to apt update / apt install) using qemu, with no further steps re: providing nameservers?

gitbls, 2023-10-09:
sdm can't tell whether the host is running in WSL or on bare metal. It CAN tell what the host OS architecture is, and yes, updating sdm changed the default because some older host OSen (either in WSL or bare metal) don't work with systemd-nspawn.

I don't know what you're asking about the test/dev environment. 

I understand (now) that your concern about /etc/resolv.conf is that you want to use systemd-nspawn, and I said "I think the best solution is to give you control over whether to use systemd-nspawn or chroot, if you know it works on your WSL host (or maybe sdm can detect it and fall back to chroot?). More investigation required."

IOW I will look into re-enabling this in some manner for V9.4. More investigation required. Stay tuned.

origonn, 2023-10-09:
>  think the best solution is to give you control over whether to use systemd-nspawn or chroot

Had forgotten to comment on that, sorry, yes that would be great. Though to reiterate, there is **no** WSL here, Just a Debian12 system being used to customize a Debian12 image.

> yes, updating sdm changed the default because some older host OSen (either in WSL or bare metal) don't work with systemd-nspawn.

Thanks for confirming/clarifying. I take this to mean that any system would not be able to run apt update / install in an image, due to qemu now being default.

gitbls, 2023-10-09:
> Thanks for confirming/clarifying. I take this to mean that any system would not be able to run apt update / install in an image, due to qemu now being default?
> 
Yeh. I had tested sdm on an x86/64 but I reference my apt-cacher-ng by IP address so didn't have any problem. More research required.

> Edit: Was able to revert to systemd-nspawn by following the opposite of #92 to disable chroot, to restore dns for updating image. Thank you for pointing that one out. :)

Great. I was going to mention that, glad you sorted that out.

I'll be looking at this over the next couple of days in between some "life stuff". Stay tuned.

gitbls, 2023-10-11:
V9.4 now uses systemd-nspawn by default. If that fails, `--chroot` should make it work. And chroot gets a /etc/resolv.conf so network naming is functional.

closed 2023-10-23 by gitbls