gitbls/sdm #114: bootconfig plugin: values without =, and multiple arguments

origonn, 2023-10-09:
Hello (again). After switching from V8.6 to V9.3 and updating `dtoverlay` and `dtparam` to the `bootconfig` plugin, I have noticed some differences from prior functionality that appear to be bugs.

1. Arguments that are of type `arg=x` (`dtoverlay=disable-bt` / `dtparam=sd_poll_once`) instead of `arg=x=y` (`dtparam=watchdog=on`) are being incorrectly converted to an `arg=x,x`

<details>
<summary><pre>sdm --customize --plugin bootconfig:"dtparam=sd_poll_once" 20230612_raspi_3_bookworm.img</pre></summary>
<pre>
...
* Plugins included:
   * bootconfig
       Args: dtparam=sd_poll_once
...
> Run Plugin 'bootconfig' (/mnt/sdm/usr/local/sdm/plugins/bootconfig) Phase 0 with arguments: 
  'dtparam=sd_poll_once'
* Plugin bootconfig: Start Phase 0
> Plugin bootconfig: Keys/values found:
   dtparam: sd_poll_once
> Plugin bootconfig: Processing keys...
> Plugin bootconfig: Append 'dtparam=sd_poll_once,sd_poll_once' to /mnt/sdm/boot/config.txt
* Plugin bootconfig: Complete Phase 0
...
</pre>
</details>

---  

2. Multiple calls of `dtoverlay` or `dtparam` are accepted but only the last value is used

<details>
<summary><pre>sdm --customize --plugin bootconfig:"dtparam=sd_poll_once|dtparam=watchdog=on" 20230612_raspi_3_bookworm.img</pre></summary>
<pre>
...
* Plugins included:
   * bootconfig
       Args: dtparam=sd_poll_once|dtparam=watchdog=on
...
> Run Plugin 'bootconfig' (/mnt/sdm/usr/local/sdm/plugins/bootconfig) Phase 0 with arguments: 
  'dtparam=sd_poll_once|dtparam=watchdog=on'
* Plugin bootconfig: Start Phase 0
> Plugin bootconfig: Keys/values found:
   dtparam: watchdog=on
   dtparam: watchdog=on
> Plugin bootconfig: Processing keys...
> Plugin bootconfig: Append 'dtparam=watchdog=on' to /mnt/sdm/boot/config.txt
> Plugin bootconfig: Append 'dtparam=watchdog=on' to /mnt/sdm/boot/config.txt
* Plugin bootconfig: Complete Phase 0
...
</pre>
</details>

Currently, a solution is to include all `dtoverlay` / `dtparam` values in a single comma-separated line, but that is sometimes not feasible due to necessary comments in between, and also requires needing at least two arguments if one of them is not an `x=y` type

<details>
<summary><pre>
sdm --customize --plugin bootconfig:"dtoverlay=disable-bt,disable-wifi|dtparam=sd_poll_once,watchdog=on" 20230612_raspi_3_bookworm.img</pre></summary>
<pre>
...
* Plugins included:
   * bootconfig
       Args: dtoverlay=disable-bt,disable-wifi|dtparam=sd_poll_once,watchdog=on
...
> Run Plugin 'bootconfig' (/mnt/sdm/usr/local/sdm/plugins/bootconfig) Phase 0 with arguments: 
  'dtoverlay=disable-bt,disable-wifi|dtparam=sd_poll_once,watchdog=on'
* Plugin bootconfig: Start Phase 0
> Plugin bootconfig: Keys/values found:
   dtoverlay: disable-bt,disable-wifi
   dtparam: sd_poll_once,watchdog=on
> Plugin bootconfig: Processing keys...
> Plugin bootconfig: Append 'dtoverlay=disable-bt,disable-wifi' to /mnt/sdm/boot/config.txt
> Plugin bootconfig: Append 'dtparam=sd_poll_once,watchdog=on' to /mnt/sdm/boot/config.txt
* Plugin bootconfig: Complete Phase 0
...
</pre>
</details>

gitbls, 2023-10-09:
Thanks for your report. Indeed, this is a fallout of the way that arguments are handled. I haven't reviewed the code yet to see if and what can be fixed, but wanted to respond and get your feedback on my comments.

Clearly the first example should work. That's a bug. 

The second example is a known (at least by me 🙄) fallout of the way arguments are parsed, and passed back to the plugin. There's only room for one string per name, so the second one simply overwrites the first one. See below.

I wasn't aware that the last example even worked, so no comment on that atm

In general, my thinking is that if you want to make more than a "touchup" or two, it's more specific to provide a completely new config.txt via the `config=` argument. It's more predictable from our (the user) perspective (since we provide the config.txt). 

Also, you are aware that the bootconfig plugin can be put on the command line multiple times? The invocation of each is independent, so the dtparam/dtoverlay used are also independent, so example 2 goes away.

I would appreciate your feedback on the above.

gitbls, 2023-10-11:
NB This was not addressed in V9.4

gitbls, 2023-11-06:
This should be corrected in V9.6. Please verify.

gitbls, 2023-11-14:
Closing as fixed.

closed 2023-11-14 by gitbls