gitbls/sdm #116: Is it possible to run pip install -r requirements.txt

dwatrous, 2023-10-11:
I would like my image to be ready to run some python scripts. I'm using the copyfile plugin to copy the python files over, but I would like to run `pip install -r requirements.txt` in post install to satisfy all dependencies. Is this possible?

dwatrous, 2023-10-11:
I realize that my copyfile is failing because the directory doesn't exist, so I'm also wondering how to create a directory as part of the sdm customize. Here's my current command, for reference.

`sudo sdm --customize --plugin raspiconfig:"i2c=1" --plugin copyfile:"filelist=sdmfilelist" --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin network:"netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to fate'|wificountry=US|noipv6" --plugin apps:"apps=python3-pip" --restart --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-lite.img`

gitbls, 2023-10-11:
The easiest way to run your command in the post-install phase is to put the code into a plugin. I have a personal plugin that does a bunch of random stuff, and it also runs a few things in post-install.

If you don't already have a plugin, see https://github.com/gitbls/sdm/blob/master/Docs/Programming-Plugins-and-Custom-Phase-Scripts.md#building-plugins. It's really simple, and once you have a plugin you'll find lots of uses for it! 

As far as your command, add `mkdirif` as an argument to the `copyfile` plugin, and it will create the directory if needed. See https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#copyfile for details

dwatrous, 2023-10-11:
I found Custom Phase Script and added a script to create the directory:

`sudo sdm --customize --cscript=sdmprep.sh --plugin raspiconfig:"i2c=1" --plugin copyfile:"filelist=sdmfilelist" --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin user:"deluser=pi" --plugin network:"netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to fate'|wificountry=US|noipv6" --plugin apps:"apps=python3-pip" --restart --host mmdwatrous 2023-05-03-raspios-bullseye-armhf-lite.img`

```
#!/bin/bash
mkdir /home/mm/controller
```

The directory is created, but not soon enough for the copyfile, and it's still failing with:

```
> Plugin copyfile: Add 'cloud.py' to files to copy
? Plugin copyfile: Destination directory '/home/mm/controller' does not exist and mkdirif not
  specified
? Plugin copyfile: This MUST be corrected for proper operation
> Plugin copyfile: Add 'constants.py' to files to copy
? Plugin copyfile: Destination directory '/home/mm/controller' does not exist and mkdirif not
  specified
? Plugin copyfile: This MUST be corrected for proper operation
> Plugin copyfile: Add 'controller.py' to files to copy
```

gitbls, 2023-10-11:
I don't understand. What are you using a custom phase script for? The correct way to create a directory for copyfile is to use the `mkdirif` argument to the copyfile plugin. 

Is there a reason that this doesn't work for your scenario?

dwatrous, 2023-10-11:
I'm not sure how I overlooked that. I added `mkdirif` and it's working great now. Thank you.

For the pip install, you're suggesting I create a custom plugin. Is that right?

gitbls, 2023-10-11:
Correct. Copy the example to a new file, edit in your changes, put it in /usr/local/sdm/local-plugins (need sudo to write there) and then call it `--plugin myplugin`. Super-easy

gitbls, 2023-10-11:
See the `--extend` switch in the documentation.

dwatrous, 2023-10-12:
Custom plugin works perfectly. Thank you.

closed 2023-10-12 by dwatrous