gitbls/sdm #117: Can't enable I2C

dwatrous, 2023-10-12:
I'm using `--plugin raspiconfig:"i2c=1"` in my customize call to enable i2c. After I boot the resulting image, I still have to use `raspi-config` to enable I2C before it works. Is my argument or value incorrect? Any troubleshooting steps?

gitbls, 2023-10-12:
Are you able to access the system? I'd like to see the output from the following commands. 

```sh
sudo journalctl --list-boots
# Find the very first boot (at the top of the list). Note the "-N" at the start of the line, use it on the next line
sudo journalctl -b -N | grep -E "sdm|FirstBoot|i2c"
```

dwatrous, 2023-10-12:
I'm not sure I did it right because there was no output

```
mm@mmdwatrous:~ $ sudo journalctl --list-boots|more
-1 1dc48513d1864e7d97b2f0a350a02e54 Sat 2023-05-13 05:37:06 CDT—Wed 2023-10-11 21:07:15 CDT
 0 2509a4d4969147208036e3acd3654ac7 Wed 2023-10-11 21:07:15 CDT—Wed 2023-10-11 21:47:45 CDT
mm@mmdwatrous:~ $ sudo journalctl -b -N | grep -E "sdm|1dc48513d1864e7d97b2f0a350a02e54|i2c"
mm@mmdwatrous:~ $

```

dwatrous, 2023-10-12:
I get it. I thought you wanted me to use the first boot as input to the second command.

```
mm@mmdwatrous:~ $ sudo journalctl -b -1 | grep -E "sdm|FirstBoot|i2c"
May 13 05:37:51 mmdwatrous systemd[1]: Starting sdm System FirstBoot Configuration...
May 13 05:37:52 mmdwatrous systemd[1]: Started sdm System FirstBoot Configuration.
May 13 05:37:52 mmdwatrous root[511]: FirstBoot: Start sdm FirstBoot Service
May 13 05:37:53 mmdwatrous root[517]: FirstBoot: Run Generated FirstBoot Script '/etc/sdm/xpiboot/005-nm-config.sh'
May 13 05:37:53 mmdwatrous root[521]: FirstBoot: Configure Network Manager WiFi connection ''wishy washy'' with SSID 'Wishy Washy'
May 13 05:37:57 mmdwatrous sdm-firstboot[522]: Connection 'wishy washy' (a645a073-8255-43f6-908f-2878836b0ad5) successfully added.
May 13 05:38:00 mmdwatrous sdm-firstboot[564]: Error: unknown connection 'Wired connection 1'.
May 13 05:38:00 mmdwatrous root[568]: FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
May 13 05:38:00 mmdwatrous root[573]: FirstBoot: Set i2c to '1'
May 13 05:38:02 mmdwatrous kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
May 13 05:38:02 mmdwatrous kernel: i2c_dev: i2c /dev entries driver
May 13 05:38:03 mmdwatrous root[610]: FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
May 13 05:38:03 mmdwatrous root[623]: FirstBoot: Set WiFi Country to 'US'
May 13 05:38:16 mmdwatrous root[762]: FirstBoot: Waiting 10/120 seconds for system time synchronization
Oct 11 21:06:23 mmdwatrous root[801]: FirstBoot: System time synchronization achieved
Oct 11 21:06:24 mmdwatrous root[810]: FirstBoot: Skip Custom FirstBoot Scripts
Oct 11 21:06:31 mmdwatrous root[838]: FirstBoot: set boot behaviour to B1
Oct 11 21:06:40 mmdwatrous root[874]: FirstBoot: Re-enable getty@tty1
Oct 11 21:06:44 mmdwatrous root[894]: FirstBoot: System will restart in 20 seconds
Oct 11 21:07:08 mmdwatrous root[916]: FirstBoot: System restarting now
Oct 11 21:07:10 mmdwatrous systemd[1]: sdm-firstboot.service: Succeeded.
Oct 11 21:07:10 mmdwatrous systemd[1]: Stopped sdm System FirstBoot Configuration.
Oct 11 21:07:10 mmdwatrous systemd[1]: sdm-firstboot.service: Consumed 7.150s CPU time.
```

gitbls, 2023-10-12:
Indeed. At the beginning of the line there is `-1 1dc48513d1864e7d97b2f0a350a02e54 `. I wanted you to pick up the -1, when I said " Note the "-N" at the start of the line". I'm sorry that wasn't clear. What should I have said to make it clearer?

As far as the issue at hand, I see 
```
May 13 05:38:00 mmdwatrous root[573]: FirstBoot: Set i2c to '1'
```
That message indicates that the following command was done:
```
# run this command non-interactively
raspi-config do_i2c 1 nonint
```
So, I believe the right things are happening. I don't know anything about using i2c, so you'll have to sort this out. 

I'd suggest doing the following:
* Burn a new disk from the IMG you have. Unless you used `--restart` on your customize command line? In that case, re-customize without that
* Boot the disk
* When you can login, verify whether i2c is enabled or not (how do you check that?)
* If it's not enabled, try the above raspi-config command, and then check i2c. 
* LMK what you find

dwatrous, 2023-10-12:
The logic for i2c in raspi-config is [opposite what I expected](https://pi3g.com/enabling-and-checking-i2c-on-the-raspberry-pi-using-the-command-line-for-your-own-scripts/#:~:text=Again%2C%20note%20that%200%20means%20true%20/,I2C%20port:%20sudo%20raspi%2Dconfig%20nonint%20do_i2c%201.).

It works when set to 0

closed 2023-10-12 by dwatrous

gitbls, 2023-10-12:
Yea, I've never been clear on whether "1" is "on" or "off", although sdm's 1piboot/1piboot.conf nicely notes that i2c=0 enables i2c (but is commented out as it should be).

Anyhow, glad that it's solved.