gitbls/sdm #119: PARTUUIDs written w/o

tinker2much, 2023-10-15:
If sdm creates a PARTUUID which is only seven digits long, it writes them into conf files as is, without the leading (eighth) zero digit, and things won't boot. Example: a line selected from an sdm log: 

`> Set new Disk ID 'cf7e848' in /etc/fstab and /boot/cmdline.txt`

In that case, fstab and cmdlin.txt get "PARTUUID=cf7e848-0x" instead of "PARTUUID=0cf7e848-0x", and the pi won't boot. If I HAND CHANGE those values to add the zero, things work as expected:

```
# original values in /etc/fstab:
PARTUUID=0cf7e848-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=0cf7e848-02  /               ext4    defaults,noatime  0       1
# and in /boot/cmdline.txt:
console=serial0,115200 console=tty1 root=PARTUUID=0cf7e848-02 rootfstype=ext4 fsck.repair=yes rootwait plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
```

I have used sdm on a couple of dozen pi's, and a little quick awk on whichever sdm logs I'd saved shows that this is not common - look for a length of 7, one of which is this example machine, pi2: 
```
pi@pi8:~/Documents/bin $ awk '/Set new Disk ID/ {print FILENAME, $6, length($6)-2}' burn*.log 
burn_amy2.log 'a0a55473' 8
burn_pi02.log '6958a45e' 8
burn_pi03.log '704979ad' 8
burn_pi04.log 'fcd28b6d' 8
burn_pi0.log 'afb2186e' 8
burn_pi0w2.log 'a3a03b29' 8
burn_pi0w3.log 'b854149d' 8
burn_pi0w4.log '7b5fe637' 8
burn_pi0w5.log '8af1d2da' 8
burn_pi0w6.log 'd8a80af2' 8
burn_pi0w7.log '688c86ba' 8
burn_pi0w.log '7143f748' 8
burn_pi2.log 'cf7e848' 7
burn_pi3a.log 'cc8338ba' 8
burn_pi3.log 'e9a20f96' 8
burn_pi42.log '2e1a22e1' 8
burn_pi5.log '91aafa9' 7
burn_pi8.log '72f07280' 8
burn_piano.log 'c5699e7b' 8
burn_pib.log '894b130b' 8
burn_pimb.log 'c904b185' 8
burn_piz2.log 'af774cf9' 8
burn_piz.log 'bb3eca04' 8
```
I had previously had a handful of pi's fail, though I hadn't realized what the problem was; when I went back to the pictures I had taken of the screens, they were all the 7-digit case: 
[image]
[image]
[image]
[image]

Since I didn't know what was wrong when that handful of pi's wouldn't boot, and assumed (guessed) that the cards were messed up, I just reburned them and they magically seemed to work - they must have gotten 8-digit IDs the second time, and ben fine.

gitbls, 2023-10-15:
Excellent detective work! This is corrected in V9.5.

gitbls, 2023-10-23:
Fixed, and no further activity here, so closing. Please re-open/update if you find any more zeros gone astray. Thx!

closed 2023-10-23 by gitbls