gitbls/sdm #122: Unable to use file name "apps" with apps plugin

wreedamz, 2023-10-23:
SDM 9.5

Running `sudo sdm --customize --plugin apps:"apps=@apps" foo.img` produces
```
? Plugin ./apps is not executable
```
I'm not sure what's going on under the hood in this case but I expected to be able to use an app file with the name `apps`. Eventually I realized I need to change the name of my app file to something else, but this seems like a bug or that it should be noted in the documentation that another name must be used

gitbls, 2023-10-23:
Wow, that's a good one! Sorry that you had to spend time soring that out. Will fix (if possible), or update documentation (if not). First order of business of course is to understand what's happening, will do that when I'm back at my desk later this week.

gitbls, 2023-10-27:
The issue you're seeing is not due to the name "@apps", it's due to the fact that the *file* `apps` is in your default directory. 

When sdm looks for a plugin name, it first treats the plugin name that is specified as a /path/to/plugin. In your example, it finds the plugin name of "apps", which has a directory of ".". And voila! There is a file named `apps` in your default directory.

IOW if you move the file "apps" from your default directory to literally anywhere else on your system and used `--plugin apps:"apps=@/path/to/apps" this error will not occur.

The Plugins documentation (Docs/Plugins.md) will be updated to reflect this.

gitbls, 2023-11-13:
This is corrected in V9.6 Please test and verify. Thx

closed 2023-11-29 by gitbls