gitbls/sdm #124: --expand-root option for size of partition

stb68, 2023-10-30:
Hi, I'm looking to start using SDM, looks like a great piece of work.

One thing I like to do is to control the size of the expanded filesystem, e.g. to leave some space so it will fit on any card of a given size if you want to do a simple dd copy. 

Is it possible to add the option to specify the size to expand the root filesystem to while preventing the standard boot from expanding to the whole card?

(Or even just a separate option to prevent the standard auto expand from running so you can just do it manually.)

Thank you

gitbls, 2023-11-01:
I've thought about doing something around this, since I've seen are occasional mentions from folks wanting some sort of control over this.

Without checking the code, if you use `--expand-root` when burning a disk it will 
* Disable the "standard boot" firstboot service, which expands the root partition to fill the rest of the disk
* Expand the root partition to fill the rest of the disk

Seems like there are two options:
* Disable "standard boot" firstboot service and don't expand the root partition at all, controlled by a switch
* Enable the burn command "expand root but leave XXXX free" or "make the root partition XXXX"

The first option is pretty simple. Is that sufficient? It would require that you do the additional partitioning after burn before you boot, or otherwise ensure that the root partition has sufficient free space to boot.

stb68, 2023-11-01:
Thanks for the reply

The first option (disable the standard boot expansion) would be OK - it's easier to make the partition larger manually than it is to reduce one.  (Presumably I could even just do that after the burn on the machine I did the burn on, so no trouble at all.)

gitbls, 2023-11-02:
OK. Will get it into the next release, likely sometime next week.

stb68, 2023-11-03:
Thank you, that is brilliant.

gitbls, 2023-11-06:
`--no-expand-root` is in V9.6. Please give it a try and LMK how it goes.

stb68, 2023-11-07:
Thank you
I'm having trouble burning anything at the moment. I can expand an img, customize it, ls the file, but I can't burn it...

```
sudo sdm --burn /dev/mmcblk0 --hostname GatewayPi --plugin user:"adduser=gw|prompt" \
--no-expand-root \
--plugin apps:"apps=fail2ban,ufw" \
...ridiculously long list of args...
--plugin copyfile:"from=/home/stb/Desktop/Pi/Shared Scripts etc/bin/aprestart|to=/home/gw/bin|chown=gw:gw|chmod=755|mkdirif"    2023-10-10-raspios-bookworm-armhf-lite.img 
```

and dd complains that it can't open the img file:
```
* Burn '2023-10-10-raspios-bookworm-armhf-lite.img' (4.7GB, 4.3GiB) to '/dev/mmcblk0'...
dd if=2023-10-10-raspios-bookworm-armhf-lite.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
dd: failed to open '2023-10-10-raspios-bookworm-armhf-lite.img': Invalid argument
? Exiting due to dd error
```
If the filename was wrong then it couldn't report the size and I've used sudo, so currently no idea what is wrong!

gitbls, 2023-11-07:
Well, that's kind of interesting. Nothing changed in that area, and I don't have any good ideas at the moment. But, it's lunch time here, so will look at this again after lunch.

In the meantime, what is "Shared Scripts etc/bin..." Do you have some directory names with spaces in them? Is the IMG perhaps in one of those directories-with spaces? There has been ZERO testing of sdm with spaces in directories or filenames....

Here's a test you could do that would be helpful:
* sudoedit /usr/local/sdm/sdm-cmdsubs (or sudo edit with your favorite editor)
* Find the line **        $ddcmd &**
* Insert immediately before it
```
pwd
ls -l $dimg
```
* exit the editor, rerun the failing command and tell me what those two  lines of output are.

stb68, 2023-11-07:
Thanks for the ideas

I'm suspecting it will be something with my end rather than any changes, I haven't actually burnt with SDM yet, this is my first time.  I'm expecting bits to not work (e.g. I'm copying some files into /etc/skel and hoping they'll take effect for the user creation) and need me to change things.

Yes, that "Shared Scripts etc" directory is where I have previously put some scripts and things I use on various Pis.  I did wonder if that would be one thing I'd need to change.

I'm running the sdm command in the directory with the img file, but there aren't and spaces in the path to that anyway.  I amended the file and re-ran....

```
* Burn '2023-10-10-raspios-bookworm-armhf-lite.img' (4.7GB, 4.3GiB) to '/dev/mmcblk0'...
dd if=2023-10-10-raspios-bookworm-armhf-lite.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
/home/stb/Desktop/Pi/Images
-rw-rw-r-- 1 stb stb 4668260352 Nov  6 22:48 2023-10-10-raspios-bookworm-armhf-lite.img
dd: failed to open '2023-10-10-raspios-bookworm-armhf-lite.img': Invalid argument
? Exiting due to dd error
```
I get the same if I just run the dd (which makes sense):

```
$ sudo dd if=2023-10-10-raspios-bookworm-armhf-lite.img of=/dev/mmcblk0 status=progress bs=16M iflag=direct
dd: failed to open '2023-10-10-raspios-bookworm-armhf-lite.img': Invalid argument
```

I also tried copying the file to have a short name, but that didn't help.  The file shouldn't have been locked in any way as I rebooted since doing the customize last night and trying one run and getting the error.

I did do a quick search for possible causes and didn't see anything promising, I'll have to do some more investigation.

OK, let's just try without the 'iflag=direct'  ...

`sudo dd if=short.img of=/dev/mmcblk0 status=progress bs=16M`

and it works!  I'm running on Linux Mint 20.3.

stb68, 2023-11-07:
So, I amended /usr/local/sdm/sdm to remove the iflag argument and SDM seems to have completed the burn operation.

I wonder, and I haven't checked this at all; I'm using disk encryption and home directory encryption on the Mint machine, maybe that's the issue with the 'direct' option.

It's getting late here, so I'll have to check out the image tomorrow.

Thank you so much for the new option and the help.

gitbls, 2023-11-07:
Ah! If you had said "Mint" in the first place, I would have pointed you at https://github.com/gitbls/sdm/discussions/34, which was this exact problem, on Mint 20.3. 

You can work around this by using the `--ddsw` switch. The defaults that sdm uses are `bs=16M iflag=direct` so try `--ddsw bs=16M`

stb68, 2023-11-08:
I'm not sure if it is a Mint thing in general (unless it's particularly Mint badly handling 'direct' being used when it shouldn't).  I formatted a USB connected SSD as Ext4 and copied the image there before running the burn again and this time it worked (I undid my mod to the sdm file first).  So I do wonder if it's an encrypted disk or encrypted home thing that might happen on any Linux.

FYI: Spaces in the copyfile path don't work, even if they are escaped with a backslash.  There's a tool I found recently, called shellcheck, that checks shell scripts and gives hints to make them work better including pointing out where quoting might be needed to stop splitting.

Anyway, thanks for the help and I'll carry on.

gitbls, 2023-11-24:
> FYI: Spaces in the copyfile path don't work, even if they are escaped with a backslash. There's a tool I found recently, called shellcheck, that checks shell scripts and gives hints to make them work better including pointing out where quoting might be needed to stop splitting.

My observation is that very few people use spaces in paths and file names on Linux, largely because of the extreme PITA in supporting them.

I do not plan to update sdm to support these but will add a note to the documentation about it.

stb68, 2023-11-24:
Yeah I wasn't pointing out the shellcheck tool for that, I've just found it useful a couple of times since I found it when writing shell scripts to find issues.  Thought you might potentially find it useful.

I've renamed that directory anyway, to a more sensible permanent name - it was originally a temporary dump for copying some files around - I just tried the backslash for info.

gitbls, 2023-11-28:
I was aware of shellcheck, but haven't brought it into the toolset yet. Curious as to whether it will support my coding style...

gitbls, 2023-12-05:
In chasing down another issue it appears that systemd-nspawn doesn't handle spaces in quote-enclosed arguments well. The result is that a single argument with an embedded space passes into the called script (inside systemd-nspawn) as 2 arguments. Definitely not what was desired.

There are clearly ways to fix this (passing stuff via /tmp), but I'm going to put that on the "maybe sometime in the future" list for now.